lumapu/ahoy #530: last_success liefert falsche Zeiten?

MarenB, 2022-12-30:
Version 0.5.41 an HM-1500 und HM-1200

Tag zusammen,

ich habe versucht, mit Hilfe des Zeitstempels "last_success" eine Ablaufzeit für die MQTT-Daten umzusetzen, damit z.B. bei einer längeren Verbindungsstörung die aktuelle PV-Leistung auf "0" gesetzt werden kann, da bei mir Verbraucher abhängig davon gesteuert werden.

Ich bekomme aber z.B. 1657582856 geliefert, was einer Unix-Zeit irgendwann am 12. Juli 2022 entspricht - tatsächlich ist der Eintrag aber vom 30.12.2022, 12:31 Uhr. 
Da der Stempel zum Teile innerhalb weniger 30-Sekunden Intervalle um einige Tage springt, vermute ich einen Umrechnungsfehler.

DukeWolle, 2022-12-31:
Ich denke das dieses issue geschlossen werden kann, da seit einiger Zeit in den development Versionen keine Probleme mehr aufgetaucht sind. Spätestens nach dem neuen Release eh hinfällig.

closed 2023-01-01 by lumapu

MarenB, 2023-01-02:
Danke für's Feedback, dann springe ich mal direkt auf die aktuelle Beta!