lumapu/ahoy #532: HM 600 - Leistungswerte bei 100% werden zu niedrig angezeigt.

hajulied, 2022-12-30:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ? Wemos Mini____
Retailer URL: __??____

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5_.41__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

hajulied, 2022-12-30:
Moin,
der AHOYDTU zeigt bei 100% Leistung zu geringe Wert an. Shelly und der Batteriemonitor zeigen die korrekten Werte. Bei einer schrittweisen Reduzierung um je 10% stimmt AHOY mit Shelly und  BattMon überein. 
Der Log zeigt manchmal "Missing frame..."
[image]
[image]
[image]
[image]
[image]

lumapu, 2022-12-30:
Ja das wurde vor längerer Zeit schon mal besprochen. Der Wechselrichter gibt 2% mehr aus, da wohl davon ausgegangen wird, dass bis zur Messstelle Leitungsverluste auftreten.
Evtl. nochmal in Disord nach dem Thema suchen.

Gerri1, 2022-12-30:
@hajulied 
Der Shelly Leistungswert ist doch Wechselstromseitig und der Batteriemonitor und Ahoy-DTU (untere Anzeige) Gleichstrom, oder liege ich da falsch! Der MPPT bei der DC Spannung ist, denke ich, nicht optimal!

hajulied, 2023-01-01:
Moin,
Die Daten des Batterie Monitors sind korrekt (12,7 A, 26,3 Volt),nachgemessen mit Zangenamperemeter, d. H. der WR zieht etwa 336 Watt. Der Wert von Shelly 329 passt dazu. Ahoy zeigt aber nur 242 Watt an. Das bleibt so bei 100,90,80% so. Erst ungefähr ab 70%passen die Werte des Monitors, von shelly und ahoy zusammen.

Gerri1, 2023-01-01:
Den Shelly als Leistungsmesser würde ich nicht machen (sehr ungenau)! Auch ein Vergleich der Leistung von Gleich-und Wechselstrom ist stark vom momentanen Wirkungsgrad abhängig. Wie schon oben geschrieben, der MPPT Regler hat nur einen bestimmten Bereich bei dem er richtig arbeitet, schau mal ins HM- Datenblatt.
Auch gibt dir Ahoy nur die Messwerte wieder, was dein WR ausgibt!

Gerri1, 2023-01-01:
[image]

hajulied, 2023-01-01:
Moin,
Es kommt mir nicht auf die totale Genauigkeit an. Ich möchte nur wissen, warum der HM bzw. Ahoy etwa ab 70 % aufwärts immer die gleichen Werte zeigt, obwohl tatsächlich mehr eingespeist wird und ob ich etwas dagegen machen kann.

Gerri1, 2023-01-01:
Erstmal würde ich mir für die AC-Messung einen Sonoff POW mit Tasmota zulegen und den dann richtig Kalibrieren! 😁
Dann wäre es evtl. besser einen anderen WR zu nehmen, der für die Batteriespannung passt!

Gerri1, 2023-01-01:
Hat dein Batteriemonitor eine seperate Spannungsversorgung (Netzstecker) oder wird der aus der Batterie versorgt?
Wenn das zweite zutregffend ist, dann hast du ja noch einen Verbraucher in der Schleife!

hajulied, 2023-01-01:
Immer noch moin,
vielen Dank für die Antworten, aber sie haben nichts mit der eigentlichen Frage zu tun. (Akku und HM passen sehr gut zusammen, er startet nach wenigen Sekunden und läuft fehlerfrei, Genauigkeit und Kalibrierung brauche ich nicht, der Eigenverbrauch des Monitors liegt im mA -Bereich. )
[image]
[image]
[image]
[image]
Habe den HM jetzt auf 65% gestellt. Bei jeder Einstellung unter 70% passen die Werte zusammen und sind plausibel, siehe Screenshots.

Gerri1, 2023-01-01:
[image]

hajulied, 2023-01-01:
... Danke für Eure Antworten . 
Ich gebe es auf :-((

Gerri1, 2023-01-05:
Bei meinem Aufbau passen die Leistungen eigentlich sehr gut zusammen!
Sonoff POW und Ahoy-DTU

[image]

stefan123t, 2023-01-05:
> Immer noch moin, vielen Dank für die Antworten, aber sie haben nichts mit der eigentlichen Frage zu tun. (Akku und HM passen sehr gut zusammen, er startet nach wenigen Sekunden und läuft fehlerfrei, [...]
[image] Habe den HM jetzt auf 65% gestellt. Bei jeder Einstellung unter 70% passen die Werte zusammen und sind plausibel, siehe Screenshots.

Man sieht die Ursache des Problems doch bereits in Deinen AhoyDTU Screenshots:
* Du hast zwei MPPT in Deinem HM-600 die beide zur Stromerzeugung herangezogen werden sollen.
* Du stellst das Limit auf 100% bzw. darunter auf 90%, 80% und 70% ein.

Die Batt1 liefert ca. 68,9% der zu erwartenden Leistung des Kanals / MPPTs

* bei 100% sollte er den ersten MPPT an dem Batt1 hängt mit 50% seiner Nennleistung ausreizen.
* bei 90% bzw. 80%  verteilt er diese mit je 45%/40% auf beide Batt1 und Batt2, da aber Batt2 leer ist (U_DC = 0,7V) liefert der zweite MPPT keine Leistung.
* bei 70% oder irgendwann darunter kommst Du in den Bereich den Dein erster MPPT mit der Leistung der Batt1 auch abdecken kann und liefert damit auch die von Dir erwartete Leistung.

Für eine Batterieeinspeisung sind die Modelle mit nur einem MPPT m.E. besser geeignet, da hier auch wirklich x% der am Ausgang erwarteten Leistung entsprechen. Bei allen Modellen mit zwei / vier Eingängen und immer zwei MPPT teilt sich diese Leistung auf beide MPPTs auf. Die Einstellung ist dabei nicht mehr intuitiv richtig sondern maximal 50%!

dermeb, 2023-07-01:
Hallo zusammen, hat das Phänomen sonst niemand beobachten können?

Beste Grüße
Michael

lumapu, 2023-07-03:
ich kann hier leider nichts beitragen. Ahoy gibt nur die von Wechselrichter gemeldeten Werte weiter.

lumapu, 2023-07-23:
ist und bleibt wohl ein Bug des Wechselrichters, hier kann man in Ahoy höchstens versuchen zu rechnen, das wage ich aber lieber nicht.

closed 2023-07-23 by lumapu