lumapu/ahoy #534: v0.5.65

Gerri1, 2022-12-30:
NTP timeserver unreachable und nrf24l01+ funkt fleißig weiter obwohl disable night communication! 😮

lumapu, 2022-12-30:
Ihr seit ja schneller als das Licht - das korrigierte Release baut seit 6 Minuten

knickohr, 2022-12-30:
Guckst Du 66 😅

Gerri1, 2022-12-30:
😪

knickohr, 2022-12-30:
Es ist ein Junge, ähhh Stable 🥳

Gerri1, 2022-12-30:
Hurraaaa, ... da ist es ja!! 👍😊

beegee3, 2022-12-31:
@knickohr @Gerri1 ihr seid ja hammer: live dabei 🙇‍♂️ 

> nrf24l01+ funkt fleißig weiter obwohl disable night communication!

@lumapu: in `tickIVCommunication` passt der Trigger nicht
Abhilfe schafft `nxtTrig = mSunset + mConfig->sun.offsetSec + 1;`
(alternativ: `if (mTimestamp >= (mSunset + mConfig->sun.offsetSec)) {` )

knickohr, 2022-12-31:
@beegee3 

Wer was will muß auch was dafür tun. Da ich zwar des klingonisch mächtig bin aber nicht dieser kryptischen Programmiersprache kann ich wenigsten so meinen Beitrag dazu leisten und jede Menge Issues auf machen 😂

Wir müssen mal Gagh essen gehen 😉

Qapla‘

closed 2023-03-27 by lumapu