lumapu/ahoy #536: Hinweis zu app::loop

beegee3, 2022-12-31:
@lumapu: kleine Anpassung zur Entlastung der `app::loop`
neue Variable `bool mRxRdy` mit Default `mRxRdy = false;`
und in der loop
```
if (ah::checkTicker(&mRxTicker, 5)) {
        ...
        if (rxRdy && !mRxRdy)
            mPayload.process(true, mConfig->nrf.maxRetransPerPyld, &mStat);
        mRxRdy = rxRdy;
}
```
dann wird `mPayload.process` nur noch bei Bedarf aufgerufen, nicht mehr alle 5ms !

lumapu, 2023-01-01:
Ich glaube das funktioniert nicht und halbiert eher nur den Takt. Besser wäre es evtl. direkt in der `payload.h` einzubauen, zB. bei `add` ein `mNewData = true` und bei `process` eine Prüfung `if(!mNewData) return`.
Aber auch das funktioniert nicht ohne weiteres, da in die Payload noch zu viel 'reingestrickt' ist. Hier muss ich zuerst aufräumen und die ganzen Control Commands auslagern, zB. Richtung `sendTicker` in der `app.cpp`

beegee3, 2023-01-02:
Du hast ja jetzt angefangen, den ganzen `payload` Komplex zu überarbeiten, scheint aber im Moment zu viel des Guten zu sein. Bekomme keine Verbindung zum Inverter, erhalte im serial log für alle sendTimePacket x (x = 1, 5, b) und Chyy (yy = 03, 23, 40, 61, 75) nur:
W: while retrieving data: last frame missing: Request Complete Retransmit
I: (#0) sendTimePacket
I: sendTimePacket x
I: TX 27B Chyy | ...

beegee3, 2023-01-02:
Oh sorry, war noch auf v.05.67, gibt ja schon v.05.68, ich schau nochmal ...

stefan123t, 2023-01-05:
@lumapu habe auch die Tage mal im payload Komplex herum gestochert. Leider noch basierend auf der 0.5.64:

<details>
<summary>git diff on top of 0.5.64</summary>
<p>

```c
$ git diff
diff --git a/src/app.cpp b/src/app.cpp
index 29df1e6..49eec17 100644
--- a/src/app.cpp
+++ b/src/app.cpp
@@ -217,15 +217,15 @@ void app::tickSend(void) {
 
                 if (iv->devControlRequest) {
                     if (mConfig->serial.debug)
-                        DPRINTLN(DBG_INFO, F("(#") + String(iv->id) + F(") Devcontrol request ") + String(iv->devControlCmd) + F(" power limit ") + String(iv->powerLimit[0]));
-                    mSys->Radio.sendControlPacket(iv->radioId.u64, iv->devControlCmd, iv->powerLimit);
+                        DPRINTLN(DBG_INFO, F("(#") + String(iv->id) + F(") requestDevControlPacket ") + String(iv->devControlCmd) + F(" power limit ") + String(iv->powerLimit[0]));
+                    mSys->Radio.requestDevControlPacket(iv->radioId.u64, iv->devControlCmd, iv->powerLimit);
                     mPayload.setTxCmd(iv, iv->devControlCmd);
                     iv->clearCmdQueue();
                     iv->enqueCommand<InfoCommand>(SystemConfigPara); // read back power limit
                 } else {
                     uint8_t cmd = iv->getQueuedCmd();
-                    DPRINTLN(DBG_INFO, F("(#") + String(iv->id) + F(") sendTimePacket"));
-                    mSys->Radio.sendTimePacket(iv->radioId.u64, cmd, mPayload.getTs(iv), iv->alarmMesIndex);
+                    DPRINTLN(DBG_INFO, F("(#") + String(iv->id) + F(") requestDevInfoPacket"));
+                    mSys->Radio.requestDevInfoPacket(iv->radioId.u64, cmd, mPayload.getTs(iv), iv->alarmMesIndex);
                     mPayload.setTxCmd(iv, cmd);
                     mRxTicker = 0;
                 }
@@ -242,7 +242,6 @@ void app::tickSend(void) {
 
 //-----------------------------------------------------------------------------
 void app::handleIntr(void) {
-    DPRINTLN(DBG_VERBOSE, F("app::handleIntr"));
     mSys->Radio.handleIntr();
 }
 
diff --git a/src/hm/hmRadio.h b/src/hm/hmRadio.h
index 13508a4..f64c702 100644
--- a/src/hm/hmRadio.h
+++ b/src/hm/hmRadio.h
@@ -32,7 +32,7 @@
 #define RF_CHANNELS             5
 #define RF_LOOP_CNT             300
 
-#define TX_REQ_INFO         0x15
+#define TX_REQ_DEVINFO      0x15
 #define TX_REQ_DEVCONTROL   0x51
 #define ALL_FRAMES          0x80
 #define SINGLE_FRAME        0x81
@@ -194,9 +194,9 @@ class HmRadio {
             return mRfChLst[mTxChIdx];
         }
 
-        void sendControlPacket(uint64_t invId, uint8_t cmd, uint16_t *data) {
-            DPRINTLN(DBG_INFO, F("sendControlPacket cmd: ") + String(cmd));
-            sendCmdPacket(invId, TX_REQ_DEVCONTROL, SINGLE_FRAME, false);
+        void requestDevControlPacket(uint64_t invId, uint8_t cmd, uint16_t *data) {
+            DPRINTLN(DBG_INFO, F("requestDevControlPacket cmd: ") + String(cmd));
+            requestMainCmdPacket(invId, TX_REQ_DEVCONTROL, SINGLE_FRAME, false);
             uint8_t cnt = 0;
             mTxBuf[10 + cnt++] = cmd; // cmd -> 0 on, 1 off, 2 restart, 11 active power, 12 reactive power, 13 power factor
             mTxBuf[10 + cnt++] = 0x00;
@@ -218,9 +218,9 @@ class HmRadio {
             sendPacket(invId, mTxBuf, 10 + cnt + 1, true);
         }
 
[... 114 lines cut ...]
                     // process payload from web request corresponding to the cmd
                     iv->enqueCommand<InfoCommand>(cmd);
                 }
diff --git a/tools/rpi/hoymiles/__init__.py b/tools/rpi/hoymiles/__init__.py
index c02a0c7..667c4b7 100644
--- a/tools/rpi/hoymiles/__init__.py
+++ b/tools/rpi/hoymiles/__init__.py
@@ -483,7 +483,7 @@ def compose_send_time_payload(cmdId, alarm_id=0):
     """
     timestamp = int(time.time())
 
-    # indices from esp8266 hmRadio.h / sendTimePacket()
+    # indices from esp8266 hmRadio.h / requestDevInfoPacket()
     payload = struct.pack('>B', cmdId)                # 10
     payload = payload + b'\x00'                       # 11
     payload = payload + struct.pack('>L', timestamp)  # 12..15 big-endian: msb at low address
```
</p>
</details>

Speziell habe ich die Kommando / Methoden Namen refaktorisiert damit es für mich (TM) besser lesbar wird.
Mal sehen ob ich den u.a. Patch auch noch auf das aktuelle 0.5.68 development release angwendet bekomme ...
Cmd müsste man evtl. noch teilweise in SubCmd umbenennen um beim Hoymiles Lingo zu bleiben.

Hoffentlich bekomme ich irgendwann mal einen ordentlichen Workflow für Patches / PR mit PlatformIO/Git hin.

closed 2023-02-25 by beegee3