lumapu/ahoy #537: Nokia 5110

Peertje62, 2022-12-31:
Hi Got the new firmware up and running
How do I connect the nokia 5110 LCD Diplay
Kind regards
Peertje

## Hardware
  * [x ] ESP8266
  
### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x ] external antenna

### Power Stabilization:
* [X ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

## Software
* [x ] AhoyDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [x ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Gerri1, 2022-12-31:
... gugst du! https://github.com/lumapu/ahoy/tree/main/tools/pcb-nokia5110

knickohr, 2022-12-31:
https://github.com/lumapu/ahoy/pull/401

Peertje62, 2022-12-31:
Hi thank you that is what I have tried
connected D1 & D2
221230_ahoy_0.5.66_esp8266_nokia5110_f8fe044.bin
no go

knickohr, 2022-12-31:
For the Nokia you have Top connect more than the 2 I2C Pins, see schematic in PR #401

closed 2023-02-07 by lumapu