lumapu/ahoy #538: Binaries

billy0xff, 2022-12-31:
Bitte um Info wo man für die Binaries der Releases die Unterschiede findet, danke.
(ein kleines !Info.txt im Archivfile wäre schon sehr nützlich)
* ca. 1MB und ca.500kB Filesize
* ssd1306 und ohne ssd1306 ?
(Plattform und Typ sind ja klar)

Gerri1, 2022-12-31:
[image]

Gerri1, 2022-12-31:
@billy0xff 
Ich finde da sind eigentlich alle Informationen vorhanden!

billy0xff, 2022-12-31:
aja, danke, das muss ich übersehen haben, was bleibt wäre der Unterschied von 

221230_ahoy_0.5.66_esp8266_f8fe044.bin
zu
221230_ahoy_0.5.66_esp8266_ssd1306_f8fe044.bin

dann noch ein paar Beobachtungen nach Updateprozess 8266 via Web-Update auf das aktuelle Release 221230_ahoy_0.5.66_esp8266_f8fe044.bin

* Browser: Zugriff auf 192.168.4.1 (im AP-Modus) war nicht möglich mit Brave-Browser (ERR_CONNECTION_TIMED_OUT), mit Opera schon.
* Netzwerkeinstellungen (WLAN client) gespeichert, dann muss power-cycle gemacht werden, nach einfachem Reboot bleibt er immer im AP-Modus
* LAN-Modus: hier 192.168.1.140: auch der Brave-Browser funktioniert hier wieder wie gehabt einwandfrei.

Gerri1, 2022-12-31:
Das 221230_ahoy_0.5.66_esp8266_ssd1306_f8fe044.bin ist für das SSD1306 Grafikdisplay.
Nach dem Flashen den ESP einmal Stromlos machen, spart oft Ärger! 😉

billy0xff, 2022-12-31:
Danke, alles klar jetzt!

stefan123t, 2023-01-05:
@lumapu eventuell sollten wir den githash direkt vor / nach der Versionsnummer in die Dateinamen einbauen, dann irritiert das nicht so sehr das Erscheinungsbild ?
Eventuell noch ein zusätzliches _dev_ / _rel_ vor der Versionsnummer und den Dateinamen des Archivs ahoydtu sowie der ahoy Bin Dateien angleichen ?

Archive:  230102_ahoydtu_dev_0.5.68_27b3a9c.zip
```
firmware/
install.html
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp32.bin
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp32_nokia5110.bin
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp32_ssd1306.bin
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp8266.bin
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp8266_nokia5110.bin
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp8266_ssd1306.bin
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp8285.bin.gz
firmware/230102_ahoydtu_dev_0.5.68_27b3a9c_esp8285.bin
firmware/bootloader.bin
firmware/manifest.json
firmware/ota.bin
firmware/partitions.bin
14 files
```

lumapu, 2023-01-07:
kommt in der nächsten Verison

stefan123t, 2023-01-12:
@lumapu kann man auch beim ZIP der Development Builds noch die Versionsnummer mit einbauen ?

Also die Namen der Dateien innerhalb des ZIPs passen ja jetzt,
aber der Build Lauf generiert unter Actions immer ein ahoydtu_dev.zip
besser wäre ein ahoy_dev_v0.5.66.zip analog zu den Releases :grinning:

lumapu, 2023-01-16:
ja wäre gut, aber dann muss man immer manuell die ahoyDtu.de Seite nachziehen. Deswegen habe ich einen generischen Namen gewählt. Im Zip sind dann die aktuellen Hashes und Versionen, daher würde ich es gerne so lassen.

closed 2023-01-16 by lumapu