lumapu/ahoy #539: Reste Yield Day Werte um Mitternacht

ArminArmout, 2023-01-01:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 

https://github.com/lumapu/ahoy/commits/dec333f

### Build & Flash Method:
* [ ] Arduino
* [x ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ x] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

## Issue:
Guten Morgen,

ich werte für eine Statistik den Gesamtertrag je Tag aus. Dabei ist mir aufgefallen das dieser Wert erst wieder zurückgesetzte wird wenn am neuen Tag neue Werte empfangen werden. 
Dies ist leider negativ wenn man den max Wert für einen Tag in einer Datenbank sucht. 
gibt es eine Möglichkeit diesen Werte um Mitternacht auf 0 zurückzusetzen?

Grüße und Danke im Voraus.

PS:
Frohes neues Jahr euch noch allen

DukeWolle, 2023-01-01:
@ArminArmout 

Das ist leider so ... Wurde auch schon viel darüber Diskutiert. Gerade hier gibt es ein paar Issues dazu. Einige die mit Datenbanken arbeiten erledigen das einfach Automatisch .. Ich zum Beispiel werte den Wert 'available' aus. Bei '0' setze ich erst mal ch1 und ch2 des Hoymiles auf 0 ..  YieldDay wird dann um Mitternacht genullt. 

Ich arbeite allerdings mit Homeassistant und dem Mosquitto Broker  

Gutes Neues !

knickohr, 2023-01-01:
Und täglich grüßt das Murmeltier „Werte nullen“ 😉 Endless story, endless discussion. Summa Summarum, es selbst machen wie DukeWolle schon geschrieben hat.

stefan123t, 2023-01-05:
Ähnlich der Sunrise / Sunset Communication Stop Funktion sollte es auch eine Dayend / Midnight YieldDay auf 0 setzen Option geben. Einfach einen Trigger etwa kurz vor Mitternacht der das YieldDay Topic einmal auf 0 setzt und gut ist es ?
Wenn man die Disable Night Communication aktiv hat sollte das doch m.W. bereits nach Sunset gemacht werden.
Aber für die Nutzer ohne diese Option aktiv wird es m.E. noch nicht aktiv zurückgesetzt und da das Feld `retained` ist wird es bis zum Kommunikationsbeginn am nächsten Morgen gehalten.

knickohr, 2023-01-10:
@lumapu 

Abschließende Frage :  Welche Werte werden jetzt (außer Yield Day) noch zurück gesetzt bei „comm Stop“ not available“ ?

lumapu, 2023-01-11:
@knickohr 

alle, außer die hier
```
                    switch(fld) { 
                         case FLD_YD: 
                         case FLD_YT: 
                         case FLD_FW_VERSION: 
                         case FLD_FW_BUILD_YEAR: 
                         case FLD_FW_BUILD_MONTH_DAY: 
                         case FLD_FW_BUILD_HOUR_MINUTE: 
                         case FLD_HW_ID: 
                         case FLD_ACT_ACTIVE_PWR_LIMIT: 
                             continue; 
                             break; 
                     }

```

closed 2023-03-27 by lumapu

ottelo9, 2023-04-15:
Es gibt nun die Option: Reset values and YieldDay at midnight

Welche Werte werden denn zurück gesetzt? Die die @lumapu gepostet hat?