lumapu/ahoy #540: OTA nicht möglich 0.5.66

MubiTec, 2023-01-01:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: wemos d1 mini
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: Git SHA: dec333f :: 0.5.41

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich wünsche ein frohes neues Jahr! 
Ich konnte auf die aktuell installierte Version Git SHA: dec333f :: 0.5.41 per OTA updaten. Nun will ich per OTA auf die 0.5.66 updaten, aber das Update schlägt immer fehl. 
Hab mal ausprobieren auf die Git SHA: dec333f :: 0.5.41 noch mal zu updaten, aber auch da schlägt es fehl. 
Irgendwelche Ideen? Oder hilft da nur ein Flash per serial? 
Danke schon mal!

DukeWolle, 2023-01-01:
@MubiTec 

Zwischen 41 und 66 liegt ein gutes Stück Arbeit. Ich würde dir empfehlen die 66 über den Webinstaller zu installieren. Ein erase Flash wäre sicher auch von Vorteil. Danach hast Du wirklich wieder ein sauberes System laufen. 

Gutes Neues Jahr !!!

MubiTec, 2023-01-01:
Ok, danke! Dann mach ich das so!

closed 2023-01-01 by MubiTec