lumapu/ahoy #541: Feature request: Display On/Off an Sun Einstellungen koppeln

TB-79, 2023-01-01:
Wäre es denkbar, daß man Display On und Off an die Sunrise/Sunset Einstellungen koppelt?
Wenn die DTU eh nicht kommuniziert,kann das Display über Nacht ja auch ruhig aus?

## Hardware
  * [ ] ESP8266
  * [X ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X ] external antenna

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-01-01:
Wenn das so umgesetzt wird, dann bitte ich um eine Einstellung die es ermöglicht dieses Feature zu aktivieren/deaktivieren.

Denn ich hätte gerne das ich auch Nachts wenn ich heim komme meinen Tagesertrag auf dem Display noch ansehen kann ohne meinen PC/Handy anschalten muß um an diese Daten zu gelangen. Denn ansonsten brauche ich kein Display mehr 😉

stefan123t, 2023-01-05:
@knickohr man kann ja auch einfach die aktuellen Daten in der Nacht ausblenden / auslassen. 
Oder gibt es auf dem Display aktuell eh nur eine Seite mit den globalen Angaben zum Yield Day ?

knickohr, 2023-01-06:
Es gibt nur eine Seite. Wobei der untere Teil durchrollest, tagsüber finden sich dort die Leistungswerte der WRs und die Taotal-Werte abwechselnd. Im oberen Bereich ist die aktuelle eingespeiste Leistung zu finden, nachts „offline“.
[image]

stefan123t, 2023-01-07:
Ich habe irgendwo noch eine Bargraph Darstellung für LCD Displays. Die arbeitet mit acht verschiedenen benutzerdefinierten Zeichen.
Falls das von Interesse ist ?
[image]

tqma1, 2023-01-23:
Könnte man nicht einfach Zeiten  einstellenden zu denen dann das Display on bzw. off ist?
Z.Bsp. Bei mir könnte es von  22:00 bis 06:00 aus sein.

lumapu, 2023-01-27:
@tqma1 das genau ist schon vorgesehen, aber noch nicht vollständig implementiert - wird sicher demnächst kommen. @knickohr hatte dies schon mal vorgeschlagen.

lumapu, 2023-03-29:
das der issue noch offen ist, werden wir an den Thema noch arbeiten

knickohr, 2023-03-31:
Was wollt ihr hier eigentlich ?

einen Schalter/Taster um das Display (zetgesteuert oder besser gesagt für ein paar Sekunden) einzuschalten ?  Eine PWM um das Display (Nokia) in der Hintergrundhelligkeit) zu regeln ?  Oder eine Uhrzeitgesteuerte Regelung ?

Alles gleichzeitig wird wohl nicht funktionieren, also muß eine Entscheidung von den 3 beschriebenen Fällen her. Die ersten beiden Punkte benötigen auch Pins, die eventuell nicht  vorhanden sind, Fall 2 sogar zwei.

tqma1, 2023-03-31:
Ich bin, wie bereits oben erwähnt, für eine Uhrzeit gesteuerte Regelung .

lumapu, 2023-03-31:
das war auch meine Idee, klar wäre ein Taster cool, aber Pins sind Mangelware.
Daher erst mal nur per fixer Zeit an und aus

csch25, 2023-03-31:
Taster ist wieder eine Komponente die Hardwareanpassung benötigt.

Ich wäre dankbar, wenn man eine Uhrzeit einstellen könnte wann sich das Display abends ausschaltet.
Einschalten wie gehabt.

stefan123t, 2024-10-27:
@lumapu ist das mit der Tageszeit schon so implementiert bzw. können wir den Feature Request schließen ?
Alternativ ein Taster oder Bewegungsmelder (IR) wäre von einigen Nutzern sicher gern gesehen, da nicht alle die ganze Zeit das Display anhaben wollen (Einbrenngefahr). Bei OpenDTU gab es Probleme als ein Nutzer das Display einfach vom Strom VCC getrennt hat, dann blieb das Programm hängen, da es nicht erwartet wurde, daß jemand das Display einfach ausschaltet.