lumapu/ahoy #542: MqTT Push Intervall zu lange in Verbindung mit Shelly3EM für Kompletten Haus Verbrauch

BambuLap, 2023-01-01:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: f8fe044 :: 0.5.66

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x ] Platform IO

### Desktop OS:
* [x ] Linux
* [x ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich komme von openDTU, hier kann man den 

**MqTT
Publish Interval: 5 seconds** angeben. 

Ich erfasse meine Daten alles über mehrere Shelly3EM in verschieden Unterverteilungen. 
Diese sind auf 10sec Push programmiert.
Shelly pusht 10mal wenn man glück hat auch die Ahoy 1mal. 
Daher passiert es ständig das Daten nicht kommen und somit
meine Berechnung nicht mehr stimmt und lückenhaft ist.

Ist es möglich MqTT Push Interval auf Niveau von openDTU anzupassen?

DennisXK, 2023-01-02:
Ja, vor dem Problem werde ich auch bald stehen. man bräuchte eine Synchronisation der einzelen Werten. Vorausetzung sollte aber erstmal ein gleiches Interval sein.

knickohr, 2023-01-02:
Der Shelly sendet nicht iin regelmäßigen Intervallen, sondern immer wenn sich Werte ändern. Also ist dieser Issue eigentlich sinnbefreit. Man müßte um die Werte alle korrekt für eine Berechnung zu bekommen das erst mal synchronisieren.

lumapu, 2023-01-02:
@DennisXK @BambuLap 

Schaut euch doch bitte die aktuelle Development Version an, die sendet das Limit jetzt sofort nach Empfang zum Wechselrichter. Version 0.5.68

BambuLap, 2023-01-02:
> Der Shelly sendet nicht iin regelmäßigen Intervallen, sondern immer wenn sich Werte ändern. Also ist dieser Issue eigentlich sinnbefreit. Man müßte um die Werte alle korrekt für eine Berechnung zu bekommen das erst mal synchronisieren.

Ja, aber in Verbindung mit Calculate nicht mehr sinnbefreit. Wenn Ahoy aber in dieser Zeit nichts pusht passen die Werte nicht mehr.

Gerri1, 2023-01-02:
@BambuLap
Wenn du es so genau haben willst, dann bist du aber mit der Übertragung per Netzwerk auch sehr dünn unterwegs! 😮

knickohr, 2023-01-02:
Die Werte passen eigentlich nie !  Dann muß schon alles synchron in der selben Sekunde passieren. Und nicht mal das ist streng genommen korrekt 😉 Es gibt immer einen Versatz, wir wollen jetzt aber nicht päpstlicher als der Papst sein. Mir reichen 30s Intervalle.

Gerri1, 2023-01-02:
Ich mach's alle 60s! 😁

BambuLap, 2023-01-06:
> Die Werte passen eigentlich nie ! Dann muß schon alles synchron in der selben Sekunde passieren. Und nicht mal das ist streng genommen korrekt 😉 Es gibt immer einen Versatz, wir wollen jetzt aber nicht päpstlicher als der Papst sein. Mir reichen 30s Intervalle.

Meine Werte passen mit openDTU vielleicht mit 2,5W unterschied damit kann ich aber leben vergleich zur DTSU.

Hab mit dem Inverter Interval gespielt. Gut bei 5Sec läuft es bei Producing bis jetzt ganz gut. 

Problem ist wenn HM auf not Producing geht kommen nur alle 60sec Mqtt an. Dies kann ich nicht beeinflussen.
@lumapu Wäre schön wann man dies auch im WebIF ändern könnte so das die Daten weiter gesendet werden.
 

Und somit wäre das Problem gelöst.

stefan123t, 2023-01-07:
> Ja, aber in Verbindung mit Calculate nicht mehr sinnbefreit. Wenn Ahoy aber in dieser Zeit nichts pusht passen die Werte nicht mehr.

Dann darfst Du eben nur wenn Du vom Ahoy aktuelle Werte bekommem hast die Berechnung durchführen. 😀 
Du kannst Dich dann entscheiden ob Du einen Mittelwert der Werte / den letzten Wert vom Shelly verwendest um Deine Berechnung durchzuführen.

Ahoy leitet jetzt ja alle Werte weiter sobald es diese vom WR bekommen hat. Ist das damit für Dich gelöst oder was hättest Du hier gerne anders bzw prinzipiell erreichen ?

Eine Aktualisierung unter 15 Sekunden hatten wir in der Vergangenheit idR nie empfohlen da dies zu instabilem Verhalten des ESP8266 führte. Das sollte mit dem AsyncWebServer zwar weitestgehend behoben sein...

Auch hängt es stark davon ab wieviele WR die AhoyDTU hintereinander abfragen muss. Ich glaube ca 5s war Anfangs das unterste Limit für einen einzelnen Inverter und da waren noch keine weiteren Kommandos zur PowerLimit Anpassung etc im Code vorgesehen.

Wenn die Funkverbindung zwischen AhyoDTU und WR nicht stabil / schnell genug ist, hat das mM nichts mit MQTT und dessen Intervall zu tun. Dh Werte per MQTT als aktuelle Werte auszugeben obwohl Ahoy nicht weiss, was der WR gerade macht ist wie Knickohr schrieb tatsächlich sinnfrei und vermutlich auch nicht was Du willst.

Bitte ggf ein neues Issue aufmachen falls die Funkverbindung nicht stabil ist bzw. ggf mal hier in #523 nachschauen ob das bei Dir auch zutrifft mit dem Parallelbetrieb mehrerer DTUs ?

BambuLap, 2023-01-08:
@stefan123t 

`Ahoy leitet jetzt ja alle Werte weiter sobald es diese vom WR bekommen hat. Ist das damit für Dich gelöst oder was hättest Du hier gerne anders bzw prinzipiell erreichen ?`

Genau hier ist das große Problem. @lumapu #542 hat die Option in MQTT Settings Interval [s] mit eingebaut. Angenommen MQTT Settings 5sec Interval sollte bei available 0 eines HMs (ich hab 4HMs) Daten gepusht werden und das immer. Man braucht die available 0 für die Errechnung mit Shellys für Gesamt verbrauch. 

Leider klappt dies noch nicht mit der #542
Im Moment ist es noch so HM OFF kein Push von available 0

Die Funkverbindung ist es auf keine Fall.

Ich komme von openDTU hier gibt es keine Problem lief 3 Monate problemlos kein Rechenfehler der Leistung oder
Funkaussetzer.

Hier Screen Short weil @knickohr das nicht glaubt bei openDTU mit 5sec.

[image]
[image]
[image]

knickohr, 2023-01-08:
Ja sorry, was nützt es mir denn wen ich alle 5 Sekunden die gleichen Werte vorgehalten bekomme. Die Wechselrichter können nun mal nicht so schnell Werte ausgeben wie die Shellies.

Ein Öfteren Ausgeben alter Werte ist in meinen Augen Verschwendung und korrigiert auch keinerlei Berechnungen. Ich schätze mal Du spielst auf die Lücken in deiner Grafik an.

nur so, ich betreibe auch 4 Shellies mit Ahoy und habe keine Lücken in den Kurven 😉

[image]
(Der kurze „Nullpeak“ um ca. 16:30 kommt von einem Restart des Flows)

BambuLap, 2023-01-11:
@lumapu 

Big Thanks jetzt passt es mit den 5sec läuft sehr stabil mit der #d8e255ddc2c9c5b09ee14e739a744ed0597c980c. 
Nur eine frage, vielleicht um Daten zu sparen.
HM 1500 & 800 nicht verfügbar. Hier wäre nur available=0 interessant zu pushen.
Die Ch0 bis Ch2 könnte man hier deaktivieren. Ist dies möglich?

lumapu, 2023-01-13:
klar kann man das machen, muss prüfen wie viel Code dafür nötig ist und was der letztendliche nutzen daraus ist

BambuLap, 2023-01-14:
> klar kann man das machen, muss prüfen wie viel Code dafür nötig ist und was der letztendliche nutzen daraus ist

Super, interessant ist nur des jeweiligen Inverters das er alle 5sec nur available=0 sendet der Rest an Daten ändert sich nicht bis er wieder available=2 bringt. Also kann dies ignoriert werden um Daten zu sparen. 
Wenn sich dies umsetzen lässt wäre genial und die Daten halten sich in grenzen.

closed 2023-03-27 by lumapu