lumapu/ahoy #543: Nach Flash auf 0.5.67/68 kein Zugriff mehr

Ollipop030, 2023-01-02:
Hi, habe heute versucht die 0.5.67 zu flashen, danach war die DTU nicht mehr erreichbar. Neustarts helfen nicht, es gibt auch keinen AP. Bin dann wieder auf die 0.5.66 stable, und zufälligerweise gab es dann direkt die 0.5.68. Also nochmal damit probiert, gleiches Problem.

Geflasht habe ich über die Weboberfläche, so wie immer. Mir wurde auch Update success angezeigt, danach gab es aber nur einen Timeout im Browser.

Ollipop030, 2023-01-02:
Ok. Hab grad nochmal probiert: Softreboot, direkt danach das Update auf 0.5.68 übers Webinterface. Diesmal hat es geklappt. Das Verhalten ist aber neu. Habe von der 0.5.41 bis zur 66 fast alle Dev Builds auf diese Weise installiert, gab nie Probleme.

Gerri1, 2023-01-02:
Bei mir verlief das Update von 0.5.66 auf 67 und 68 ohne Auffälligkeiten! 
(Wemos D1 mini)

knickohr, 2023-01-02:
68 hat jetzt funktioniert 👍

Hat zwar gefühlt Minuten gedauert bis die Inverter wieder da waren, ich bilde mir ein bei den vorherigen Versionen (66 und darunter) ging es schneller, aber egal, es geht.

Warum gibt es eigentlich 2 Changelog auf der Index ?  Ich vermute mal die obere ist für die Stable und die darunter für die DEVs ?   Sollte man deutlicher machen, bspw. Release-Changelog und Developer-Changelog.
[image]

lumapu, 2023-01-02:
ok, ja das sind doch Peanuts 🥜

knickohr, 2023-01-02:
Ich mag keine Erdnüsse 😂

closed 2023-02-27 by Ollipop030