lumapu/ahoy #544: Limit per Mqtt ändern/ übertragen

raspihomegit, 2023-01-02:
Ich benutze den iobroker mit Mqtt, wo ich die ahoydtu auslesen kann. 
Ist es möglich, im iobroker das Limit zu setzen und über Mqtt an die ahoydtu zu übertragen?
Damit könnte ich eine Nulleinspeisung realisieren, wenn das Limit per Mqtt änderbar ist.

Gerri1, 2023-01-02:
@raspihomegit 
... gugst du hier! https://github.com/lumapu/ahoy/blob/development03/User_Manual.md
[image]

lumapu, 2023-01-02:
sogar im Main Branch:

https://github.com/lumapu/ahoy/blob/main/User_Manual.md

stefan123t, 2023-01-05:
@raspihomegit ist das Problem damit gelöst oder deutet Dein Username daraufhin, dass Du gar nicht die ESP sondern die RPi Variante verwendest ?

raspihomegit, 2023-01-06:
> @raspihomegit ist das Problem damit gelöst oder deutet Dein Username daraufhin, dass Du gar nicht die ESP sondern die RPi Variante verwendest ?

Ich denke, die Antworten zeigen, dass es funktioniert. Leider konnte ich er noch nicht testen, weil ich noch alles aufbaue. 
Ich nutze einen raspi für iobroker mit Mqtt, der die Werte empfangen soll und an die dtu ein Limit senden soll. Ich habe alles schon miteinander verbunden und hatte einen limitwert im Mqtt gesehen. Aber ich habe noch nicht versucht, diesen Wert zu ändern. Denke aber, dass es klappt, wenn der schon angezeigt wird.

raspihomegit, 2023-01-06:
> Ich denke, die Antworten zeigen, dass es funktioniert. Leider konnte ich er noch nicht testen, weil ich noch alles aufbaue.
> Ich nutze einen raspi für iobroker mit Mqtt, der die Werte empfangen soll [...]

Ich konnte nun die Mqtt Anbindung testen. Im ch0 gibt es die variable active_powerlimit, die ich mit einem Wert beschrieben habe. Aber in der dtu kommt es nicht an. 

Wahrscheinlich muss es doch mit einem Skript realisiert werden.

jimknopf63, 2023-01-06:
> Ich konnte nun die Mqtt Anbindung testen. Im ch0 gibt es die variable active_powerlimit, die ich mit einem Wert beschrieben habe. Aber in der dtu kommt es nicht an.
>
> Wahrscheinlich muss es doch mit [...]

Wenn Du den Wert von Hand änderst tut sich da auch nichts, Active_Power_Limit ist meines Wissen nach nur lesend. Ich habe es mit Blockly gelöst und es funktioniert einwandfrei

jimknopf63, 2023-01-06:
[image]

So wird das per Blockly gesteuert. Hoymiles ist der Pfad in MQTT (Standart=inverter)

raspihomegit, 2023-01-06:
> [image]
> 
> 
> 
> So wird das per Blockly gesteuert. Hoymiles ist der Pfad in MQTT (Standart=inverter)

Wow fantastisch. Absolut einfach und genial. Danke dafür. Dann kann ich meine Nulleinspeisung ganz einfach aufbauen.

jimknopf63, 2023-01-06:
Falls es Dir nützt, ich habe im Anhang mein Blockly exportiert. HM1500 ab 100 Watt in 30 Watt Schritten. Läuft seit ca. 2 Monaten bei mir und in der Familie.
[Ahoy Nulleinspeisung.zip](https://github.com/lumapu/ahoy/files/10360011/Ahoy.Nulleinspeisung.zip)

raspihomegit, 2023-01-06:
> Falls es Dir nützt, ich habe im Anhang mein Blockly exportiert. HM1500 ab 100 Watt in 30 Watt Schritten. Läuft seit ca. 2 Monaten bei mir und in der Familie. ZIP runterladen und als ZIP in Blockly importieren. Musst natürlich deinen aktuellen Verbrauch anpassen. Die Variable Leistung habe ich genommen damit der WR nicht dauernd mit neuen Werten befeuert wird. [Ahoy Nulleinspeisung.zip](https://github.com/lumapu/ahoy/files/10360011/Ahoy.Nulleinspeisung.zip)

Das kann ich leider nicht importieren. Wenn ich meine Skripte exportiere, sind das json Dateien. in deiner zip ist eine rtf Datei. Ist das korrekt?

jimknopf63, 2023-01-06:
> Das kann ich leider nicht importieren. Wenn ich meine Skripte exportiere, sind das json Dateien. in deiner zip ist eine rtf Datei. Ist das korrekt?

Ja, habe es eben selbst getestet. In Skripte open links auf die 3 Punkte. Da Import auswählen und in dem öffnenden Fenster das ZIP (also nicht entpackt) rein ziehen. Danach bestätigen kurz warten bis OK kommt und Du hast das Script Namens Ahoy DTU nonpersistent bei deine Scripten. Habe es extra vorher mal umbenannt um zu testen ob das auf dem Weg funktioniert.

Gerri1, 2023-01-06:
... so siehr Grafisch meine Regelung aus! 
Gelb - ungedämpft
Grün - gedämpft
[image]
[image]

raspihomegit, 2023-01-06:
> Ja, habe es eben selbst getestet. In Skripte open links auf die 3 Punkte. Da Import auswählen und in dem öffnenden Fenster das ZIP (also nicht entpackt) rein ziehen. Danach bestätigen kurz warten [...]

Genau so habe ich es gemacht. Skript sagt auch fertig, aber zeigt nichts an. Nutzt du auch den Javaskript Adapter?

raspihomegit, 2023-01-06:
> ... so sieht Grafisch meine Regelung aus, Gelb - ungedämpft Grün - gedämpft an einem wilden Tag!
> 
> [image] [image]

Das sieht auch gut aus. Überlege nämlich auch, wie man die Regelung träger machen kann, damit der Wechselrichter nicht dauern befeuert wird.

Gerri1, 2023-01-06:
Ich lese den Zeitstempel der letzten Regelung mit aus und lasse den erst wieder denen Befehl zukommen wenn die eingestellte Zeit rum ist. Auch eine Hysterese ist mit drin, damit der nicht bei jedem Watt da regeln will!

Bei mir funktioniert das Blockty von @jimknopf63!

jimknopf63, 2023-01-06:
> Genau so habe ich es gemacht. Skript sagt auch fertig, aber zeigt nichts an. Nutzt du auch den Javaskript Adapter?

Das Script sollte was anzeigen ist ziemlich lang und träge (bei mir). Hast Du mal im Script Fenster gescrollt. Nach unten und links. Bei mir zeigt er auch erst ein leeres Fenster an, nach dem scrollen werden dann die Einträge sichtbar

raspihomegit, 2023-01-06:
> Das Script sollte was anzeigen ist ziemlich lang und träge (bei mir). Hast Du mal im Script Fenster gescrollt. Nach unten und links. Bei mir zeigt er auch erst ein leeres Fenster an, nach dem [...]

Es erscheint ja noch nicht mal ein Skript in dem ich scrollen könnte. :( Habe gerade einen anderen Browser verwendet, aber gleicher Effekt. Und wenn ich ein neues BlocklySkrip erstelle und dann Blöcke importiere, kommen diverse Fehler.

jimknopf63, 2023-01-06:
Verstehe ich jetzt nicht, bei Gerri1 funktioniert das Script

raspihomegit, 2023-01-06:
> Verstehe ich jetzt nicht, bei Gerri1 funktioniert das Script

Ja, das wundert mich auch, wieso das bei mir nicht geht. Wir nutzen aber alle den Javascript Adapter oder?

jimknopf63, 2023-01-06:
> > Verstehe ich jetzt nicht, bei Gerri1 funktioniert das Script
> 
> Ja, das wundert mich auch, wieso das bei mir nicht geht. Wir nutzen aber alle den Javascript Adapter oder?

Ja, den nutze ich. Könnte mir nur vorstellen das deiner nicht auf dem aktuellsten Stand (6.14) ist. Sonst wüsste ich jetzt keine Erklärung.

raspihomegit, 2023-01-06:
> Ja, den nutze ich. Könnte mir nur vorstellen das deiner nicht auf dem aktuellsten Stand (6.14) ist. Sonst wüsste ich jetzt keine Erklärung.

Ich habe den 6.1.4

raspihomegit, 2023-01-06:
@jimknopf63  Ich habe deine Blöcke importiert. Aber das sieht auch unvollständig aus.

[image]

raspihomegit, 2023-01-06:
@jimknopf63  Wieso ist das eigentlich eine rtf Datei in der Zip und keine json Datei? Wenn ich meine Skripte exportiere, sind die in json.

Gerri1, 2023-01-06:
Ich hatte nur den Inhalt der *.rtf Datei kopiert und dann in Blockly importiert!

jimknopf63, 2023-01-06:
> @jimknopf63 Ich habe deine Blöcke importiert. Aber das sieht auch unvollständig aus.
> 
> [image]

Das passt schon, klicke mal mit rechts ins freie Feld und dann alle Blöcke entfalten. Oder die Träger jeweils unter den gelben Markierungen mit rechts anklicken und Bausteine entfalten. Dann wird's sichtbar

jimknopf63, 2023-01-06:
> @jimknopf63 Wieso ist das eigentlich eine rtf Datei in der Zip und keine json Datei? Wenn ich meine Skripte exportiere, sind die in json.

weil ich die x.json Datei nicht importieren konnte, kam immer eine Fehlermeldung bei mir. Da habe ich sie in Apple TextEdit kopiert und abgespeichert. Ausserdem konnte ich die so nicht hochladen weil es ein unerlaubtes Format war.

Gerri1, 2023-01-06:
als *.txt geht das bei mir immer!

jimknopf63, 2023-01-06:
> Es erscheint ja noch nicht mal ein Skript in dem ich scrollen könnte. :( Habe gerade einen anderen Browser verwendet, aber gleicher Effekt. Und wenn ich ein neues BlocklySkrip erstelle und dann [...]

Dann öffne mal die entpackte Textdatei, kopiere den ganzen Text und importiere den mal in einem neu erstellten Script. Danach evtl. scrollen. Für alles zu sehen rechts im freien Bereich klicken und alle Bausteine entfalten wählen.

Gerri1, 2023-01-06:
... die gute alte EDIT! 😂

raspihomegit, 2023-01-06:
> Dann öffne mal die entpackte Textdatei, kopiere den ganzen Text und importiere den mal in einem neu erstellten Script. Danach evtl. scrollen. Für alles zu sehen rechts im freien Bereich klicken und [...]

Jaaaaa. Es hat geklappt. Nun sehe ich dein riesiges Programm :) Danke für eure Hilfe. Echt klasse. Nun muss ich es nur noch verstehen. Aber der Raspi läuft dabei auf Hochtouren :)

stefan123t, 2023-01-07:
@raspihomegit können wir das Issue jetzt eigentlich schliesse. Oder sollen wir das Blockly noch in das Repo aufnehmen ?

Danke @jimknopf63 für Dein Blockly Script zur Nulleinspeisung.

raspihomegit, 2023-01-07:
> 
> @raspihomegit können wir das Issue jetzt eigentlich schliessen oder sollen wir das Blockly noch in das Repo aufnehmen ?
> 
> Danke @jimknopf63 für Dein Blockly Script zur Nulleinspeisung.

Ja, wir können das Issue schließen. Es funktioniert alles wie gewünscht. Es wäre zwar schön, wenn man direkt mit der variable im iobroker schreiben könnte, aber mit dem Blockly funktioniert es auch super. Danke an alle.

jimknopf63, 2023-01-07:
Das kann hier ruhig geschlossen werden.

closed 2023-01-08 by lumapu

raspihomegit, 2023-01-08:
@jimknopf63 Grüße, ich habe dein Skript angepasst. Am Anfang habe ich Variablen gesetzt, wo man die ObjectID Pfade und aktuelle Verbrauchsdatenpunkte einrichten kann. Somit muss man nicht das ganze Skript anpassen. Des Weiteren habe ich eine Funktion für das Rücksetzen der Variablen eingebaut. Dadurch wird der Code ziemlich verschlankt und läuft flüssiger in Blockly. Vielleicht möchtest du es dir anschauen.
Danke nochmal für deine Hilfe.

[Nulleinspeisung_Fkt_Var0.txt](https://github.com/lumapu/ahoy/files/10368759/Nulleinspeisung_Fkt_Var0.txt)

jimknopf63, 2023-01-08:
> @jimknopf63 Grüße, ich habe dein Skript angepasst. Am Anfang habe ich Variablen gesetzt, wo man die ObjectID Pfade und aktuelle Verbrauchsdatenpunkte einrichten kann. Somit muss man nicht das ganze [...]

Vielen Dank, das sieht auf den ersten Blick ja schon mal ganz gut aus. Ich habe wenig Blockly Erfahrung und muss immer wieder nachlesen wie man was erreicht. Mit den Variablen ist super, da hatte ich auch schon dran gedacht. Mir fehlte es aber bisher an Ideen wie man das umsetzt. Danke für deine Arbeit und das verschlanken des Scripts.

raspihomegit, 2023-01-08:
> 

> Vielen Dank, das sieht auf den ersten Blick ja schon mal ganz gut aus. Ich habe wenig Blockly Erfahrung und muss immer wieder nachlesen wie man was erreicht. Mit den Variablen ist super, da hatte [...]

Ich bin noch am Überlegen, ob man die Abfrage der Zeit herauszieht und diese als Trigger verwendet. Allerdings würde das Skript dann nur noch jede Minute Abfragen und regeln. Wäre das zu träge für die Regelung? 
Mit deinem Skript wird ja sicherlich immer etwas mehr produziert als verbraucht, oder? Demzufolge würde man am Zähler irgendwann eine exportierte Leistung sehen. Ich bin mir nämlich noch nicht sicher, ob ich meine Anlage anmelden. Und wenn, dann vielleicht nur als 600W Anlage.

jimknopf63, 2023-01-08:
> Ich bin noch am Überlegen, ob man die Abfrage der Zeit herauszieht und diese als Trigger verwendet. Allerdings würde das Skript dann nur noch jede Minute Abfragen und regeln. Wäre das zu träge für [...]

Als 600 Watt Anlage kannst Du doch einiges Rückspeisen. Bei mir läufts seit 14.09. und ich habe am Anfang nur auf 600 Watt limitiert und nachher erst mit der Regelung. Bei der Limitierung auf 600 Watt habe ich mehr verschenkt als mit Regelung. Und Ahoy braucht ja auch ca. 10-15 Sec. um eine Änderung umzusetzen.

raspihomegit, 2023-01-08:
> Als 600 Watt Anlage kannst Du doch einiges Rückspeisen. Bei mir läufts seit 14.09. und ich habe am Anfang nur auf 600 Watt limitiert und nachher erst mit der Regelung. Bei der Limitierung auf 600 [...]

Ja das stimmt. Da kann man einiges einspeisen, wenn man anmeldet. 
Hast du an deinem Wechselrichter noch etwas eingestellt? In der Bedienungsanleitung von Hoymiles HM-1500 steht was über verschiedene Betriebsmodies. Unter anderem Nulleinspeisung. Oder weiß der Wechselrichter automatisch Bescheid?
Also Meinste eine Abfrage jede Minute wäre nicht zu träge? Denn dann bräuchte man ca. 1 Minute und 15s für eine Änderung. Hast du mal geprüft wie oft dein Wechselrichter mit Änderungen beschossen wird?

jimknopf63, 2023-01-08:
> Ja das stimmt. Da kann man einiges einspeisen, wenn man anmeldet. Hast du an deinem Wechselrichter noch etwas eingestellt? In der Bedienungsanleitung von Hoymiles HM-1500 steht was über verschiedene [...]

Ich habe am HM nichts eingestellt. Habe eine Originale DTU hier liegen und da kannst Du für den HM die Nulleinspeisung einstellen. Das geht aber nur in Verbindung mit einem Chint Zwischenzähler der per Kabel an die DTU angeschlossen wird. Und das musst Du für den HM in der DTU aktivieren. ... Ich habe nicht geprüft wie oft der HM die Änderung macht, aber doch schon ziemlich regelmäßig. Am Anfang hatte ich ein Blockly wo bei Änderung des Bezugs der neue Wert als Variable an den HM geschickt wurde. Funktionierte auch aber nach 3-4 Std. war Schluss. Dann musste ich den HM vom Netz nehmen und neu starten lassen. Darauf dann das komplizierte Blockly was bis jetzt keine Probleme macht. Bis vor kurzem war morgens ein Neustart des Ahoy nötig, seit 0.5.66 hat sich das auch erledigt

jimknopf63, 2023-01-08:
> > > > > > @jimknopf63 Grüße, ich habe dein Skript angepasst. Am Anfang habe ich Variablen gesetzt, wo man die ObjectID Pfade und aktuelle Verbrauchsdatenpunkte einrichten kann. Somit muss man nicht das ganze Skript anpassen. Des Weiteren habe ich eine Funktion für das Rücksetzen der Variablen eingebaut. Dadurch wird der Code ziemlich verschlankt und läuft flüssiger in Blockly. Vielleicht möchtest du es dir anschauen. Danke nochmal für deine Hilfe.
> > > 
> > > 
> > > > > > [Nulleinspeisung_Fkt_Var0.txt](https://github.com/lumapu/ahoy/files/10368759/Nulleinspeisung_Fkt_Var0.txt)
> > > 
> > > 
> > > > >
> > > 
> > > 
> > > > >
> > > 
> > > 
> > > > > Vielen Dank, das sieht auf den ersten Blick ja schon mal ganz gut aus. Ich habe wenig Blockly Erfahrung und muss immer wieder nachlesen wie man was erreicht. Mit den Variablen ist super, da hatte ich auch schon dran gedacht. Mir fehlte es aber bisher an Ideen wie man das umsetzt. Danke für deine Arbeit und das verschlanken des Scripts.
> > > 
> > > 
> > > > 
> > > 
> > > 
> > > > Ich bin noch am Überlegen, ob man die Abfrage der Zeit herauszieht und diese als Trigger verwendet. Allerdings würde das Skript dann nur noch jede Minute Abfragen und regeln. Wäre das zu träge für die Regelung? Mit deinem Skript wird ja sicherlich immer etwas mehr produziert als verbraucht, oder? Demzufolge würde man am Zähler irgendwann eine exportierte Leistung sehen. Ich bin mir nämlich noch nicht sicher, ob ich meine Anlage anmelden. Und wenn, dann vielleicht nur als 600W Anlage.
> > > 
> > > 
> > > Als 600 Watt Anlage kannst Du doch einiges Rückspeisen. Bei mir läufts seit 14.09. und ich habe am Anfang nur auf 600 Watt limitiert und nachher erst mit der Regelung. Bei der Limitierung auf 600 Watt habe ich mehr verschenkt als mit Regelung. Und Ahoy braucht ja auch ca. 10-15 Sec. um eine Änderung umzusetzen.
> > 
> > 
> > Ja das stimmt. Da kann man einiges einspeisen, wenn man anmeldet. Hast du an deinem Wechselrichter noch etwas eingestellt? In der Bedienungsanleitung von Hoymiles HM-1500 steht was über verschiedene Betriebsmodies. Unter anderem Nulleinspeisung. Oder weiß der Wechselrichter automatisch Bescheid? Also Meinste eine Abfrage jede Minute wäre nicht zu träge? Denn dann bräuchte man ca. 1 Minute und 15s für eine Änderung. Hast du mal geprüft wie oft dein Wechselrichter mit Änderungen beschossen wird?
> 
> Das war mein Ursprungsscript was den HM aber nach ein paar Std. hat einfrieren lassen. Wurde wohl zu oft geändert.
Ausserdem hat er bei Rückspeisung nicht mehr geregelt. 

[image]

raspihomegit, 2023-01-08:
> Ausserdem hat er bei Rückspeisung nicht mehr geregelt.
>
>
>
> [image]

Ja bei dem Skript ist natürlich Dauerfeuer angesagt. :) da ist dein neues Skript schon viel besser. 
Hast du eigentlich auch eine Auswertungsmethode der Daten? Im iobroker kann ich im echart nur die Leistung aufzeichnen. Ich sicher allerdings noch nach einer Möglichkeit die erzeugte Arbeit pro Tage, Wochen und Monate auszuwerten? Ich wollte zusätzlich ein Shelly Plus 1PM verwenden, weil man da einfach Diagramme auswerten kann. Allerdings misst der zu ungenau gegenüber der Ahoydtu.

jimknopf63, 2023-01-08:
> Ja bei dem Skript ist natürlich Dauerfeuer angesagt. :) da ist dein neues Skript schon viel besser. Hast du eigentlich auch eine Auswertungsmethode der Daten? Im iobroker kann ich im echart nur die [...]

Ich logge Yields Total, 1.8.0 vom Zähler mit dem Scource Analytics Adapter. Zusätzlich einen Datenpunkt wenn der Strombezug mit - voran steht mit SourceAnalytics als Überschuss. Das ganze sieht dann auf Info Display so aus
[image]

raspihomegit, 2023-01-08:
> Ich logge Yields Total, 1.8.0 vom Zähler mit dem Scource Analytics Adapter. Zusätzlich einen Datenpunkt wenn der Strombezug mit - voran steht mit SourceAnalytics als Überschuss. Das ganze sieht dann [...]

Das sieht echt genial aus. Super aufbereitet. Ich muss mal sehen ob ich den sourceanalytix Adapter noch bekomme. Der ist glaube ich raus aus der Auswahl beim iobroker.

jimknopf63, 2023-01-08:
> Das sieht echt genial aus. Super aufbereitet. Ich muss mal sehen ob ich den sourceanalytix Adapter noch bekomme. Der ist glaube ich raus aus der Auswahl beim iobroker.

Für den Adapter musst du kurz von Stable auf Beta Adapter gehen, da versteckt der sich

jimknopf63, 2023-01-09:
> @jimknopf63 Grüße, ich habe dein Skript angepasst. Am Anfang habe ich Variablen gesetzt, wo man die ObjectID Pfade und aktuelle Verbrauchsdatenpunkte einrichten kann. Somit muss man nicht das ganze [...]

Du hast in deinem überarbeitetem Script in den Variablen den Zähler 1.8.0 angegeben. Dann steuert der HM aber zu niedrig, ich habe die Werte auf aktuellen Verbrauch (also Netzbezug + Solarertrag) gestellt. Dann funktioniert es bisher (wegen Wetter nur bis 200 Watt)

raspihomegit, 2023-01-09:
Ich glaube mein Skript kann so gar nicht funktionieren. Die Variable "aktueller Hausverbrauch" wird ja nur zu Beginn einmal gesetzt und danach nicht mehr verändert oder? :) 
Aber du hast auch Recht. Um das Limit zu setzen, muss man die benötigte Leistung + erzeugte Leistung rechnen.

jimknopf63, 2023-01-09:
> @jimknopf63 Ich glaube mein Skript kann so gar nicht funktionieren. Die Variable "aktueller Hausverbrauch" wird ja nur zu Beginn einmal gesetzt und danach nicht mehr verändert oder? :)
> Aber du hast auch Recht. Um das Limit zu setzen, muss man die benötigte Leistung + erzeugte Leistung rechnen.

Ich denke wenn sich der aktuelle Verbrauch ändert wird die Variable neu geschrieben. Jedenfalls hat nach meiner Anpassung der HM öfter die Leistung gewechselt am heutigen Tag. Schau morgen noch mal drauf

raspihomegit, 2023-01-09:
> Ich denke wenn sich der aktuelle Verbrauch ändert wird die Variable neu geschrieben. Jedenfalls hat nach meiner Anpassung der HM öfter die Leistung gewechselt am heutigen Tag. Schau morgen noch mal [...]

OK, 
Ich kann es leider noch nicht testen, da ich meine Anlage erst 1-2 Wochen montieren kann. Aktuell habe ich nur eine Platte im Wintergarten stehen, um überhaupt ein bisschen testen zu können. :) Im Wintergarten erzeuge ich grad mal eine Leistung von 5-20W. Da muss noch nichts regeln. :D

SourceAnalytix konnte ich erfolgreich installieren. Damit werde ich mich dann auch mal beschäftigen. Danke für deinen Tipp.

jimknopf63, 2023-01-10:
> @jimknopf63 Ich glaube mein Skript kann so gar nicht funktionieren. Die Variable "aktueller Hausverbrauch" wird ja nur zu Beginn einmal gesetzt und danach nicht mehr verändert oder? :)
> Aber du hast auch Recht. Um das Limit zu setzen, muss man die benötigte Leistung + erzeugte Leistung rechnen.

Tut mir Leid aber dein Script funktioniert doch nicht. Es bleibt auf 100% Leistung, wo heute mal richtig Sonne ist kann man das gut sehen. Habe auch schon mal die Variablen (Verbrauch) mit Träger auf Änderung gesetzt. Hat aber noch nicht funktioniert

raspihomegit, 2023-01-10:
Das habe ich mir schon gedacht. Ich muss noch die Variable "aktueller Hausverbrauch" triggern, damit neue Werte in die Variable geschrieben werden. Werde mich heut Abend noch ran setzen.

Gerri1, 2023-01-10:
Du musst dir jedenfalls eine Hysterese in der Leistungsregelung und ein Zeitlimit mit reinbasteln, sonst regelt der die ganze zeit!

raspihomegit, 2023-01-10:
> Du musst dir jedenfalls eine Hysterese in der Leistungsregelung und ein Zeitlimit mit reinbasteln, sonst regelt der die ganze zeit! Auch habe ich den Baustein zum Senden des Powerlimits an MQTT in ein seperates Modul gesetzt. Das hat für mich den Vorteil, das ich auch mal selber durch eingabe eines Wertes das einfach schicken kann.

In dem Programm vom @jimknopf63 ist die Hysterese quasi diese 30 W Schritte beim Limit. Dadurch wird nur ein Wert gesendet, wenn ein neues Limit (also außerhalb der 30W Spanne ) gesetzt werden muss.
Ich möchte das Programm noch erweitern, indem ich alle 30s einen neuen Leistungswert in die Variable des Hausverbrauches schreiben möchte. Und das Skript soll nur von Sonnenaufgang bis Untergang durchlaufen werden.

Wenn du Lust hast, könntest du uns gern dein Programm zur Verfügung stellen, damit wir noch andere Eindrücke und Ideen bekommen. :)

Gerri1, 2023-01-10:
So sieht mein Sende-Modul aus.

[image]

[Leistungsreduzierung.txt](https://github.com/lumapu/ahoy/files/10385585/Leistungsreduzierung.txt)

Bitte nicht lachen!

raspihomegit, 2023-01-10:
> So sieht mein Sende-Modul aus.
> 
> [image]
> 
> [Leistungsreduzierung.txt](https://github.com/lumapu/ahoy/files/10385585/Leistungsreduzierung.txt)
> 
> Bitte nicht lachen!

Wow, Hut ab sich so ein Skript auszudenken. Aber sieht auf dem ersten Blick komplizierter aus. Hast du bei deiner Hysterese einen Festwert drin? 
Aktualisierst du deinen Wechselrichter aller 5min?

Gerri1, 2023-01-10:
Ich glaube ich habe momentan 3 min Wartezeit drin und die Hysterese glaube ich waren 20W. Die beiden Werte habe ich in den 0_userdata drin. Ich lese ja den Zeitstempel der letzten Aktualisierung aus und lasse erst nach der angegebenen Zeit wieder eine Aktualisierung zu. Das Ganze habe ich so geplant, dass das auch keine zu großen Sprünge wieder zurück werden.

jimknopf63, 2023-01-10:
> Wow, Hut ab sich so ein Skript auszudenken. Aber sieht auf dem ersten Blick komplizierter aus. Hast du bei deiner Hysterese einen Festwert drin? Aktualisierst du deinen Wechselrichter aller 5min?

Das sieht interessant aus, das werde ich morgen mal ein wenig testen. Ich würde deinem Script nochhinzufügen das direkt beim ersten Aufruf die Datenpunkte in userdata.0 automatisch erstellt werden.

Gerri1, 2023-01-11:
@jimknopf63 ... hast du mal meins getestet?

Gerri1, 2023-01-11:
@jimknopf63 Ich habe mir deins mal angeschaut. Es ist zwar sehr schlank, aber die Regelung findet ständig statt. 
Das sieht dann wie in meiner Grafik oben aus, gelb - ungedämpft.

Gerri1, 2023-01-11:
Heute z.B. gelb - ungedämpft und grün - gedämpft. Der schickt gar keinen einzigen Wert nach Ahoy.

[image]
[image]

raspihomegit, 2023-01-11:
Habt Ihr eure Balkonkraftwerke südlich oder ost-west ausgerichtet?

[83 more replies]