lumapu/ahoy #547: MQTT Format Fehler

Waldo56, 2023-01-03:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Hi,
habe einige Fehler bei MQTT Nachrichten mit MQTTFx dokumentert.

1. inverter/dis_night_comm -> wieso ein String und nicht 0 und 1?
[image]

2. inverter/mqtt -> Fehler beim parsen
[image]

3. inverter/device -> Fehler bei parsen
[image]

LG

lumapu, 2023-01-03:
danke fürs Feedback! Ich prüfe was sich machen lässt

stefan123t, 2023-01-05:
@lumapu, @knickohr & @dukewolle was ist denn jetzt aus der Idee geworden per MQTT JSON Payloads zu publizieren.

Einerseits sind natürlich numerische Status codes wie `0`, `1`, `2`, etc. einfach und klar. Wohingegen `true` / `false` entsprechend größere Strings und damit ausführlicher sind. 
Aber sobald wir ganze Payloads als JSON formatiert übergeben, wäre doch `true` / `false` wiederum besser wenn es sich tatsächlich um `boolean` Werte handelt ?

Ich habe mir u.a. die Auto-Discovery Funktionen von HomeAssistant noch nicht wirklich angesehen, aber sowohl OpenDTU als auch AI-on-the-edge verwenden das und publizieren hierüber die notwendigen Endpoints die z.B. zum Ein / Ausschalten des WR aus HomeAssistant verwendet werden können:

```c
solar/dtu/rssi	-84
solar/dtu/hostname	OpenDTU
solar/dtu/ip	192.168.1.5X
solar/dtu/uptime	228257
solar/dtu/status	online

solar/114185212345/status/last_update	1672068980
solar/114185212345/status/producing	0
solar/114185212345/status/reachable	0
solar/114185212345/status/limit_absolute	600.00
solar/114185212345/status/limit_relative	100.00
solar/114185212345/device/hwversion	01.00
solar/114185212345/device/hwpartnumber	269553683
solar/114185212345/device/fwbuilddatetime	2020-07-10 12:58:00
solar/114185212345/device/fwbuildversion	10008
solar/114185212345/device/bootloaderversion	104
solar/114185212345/name	HM-600
solar/114185212345/0/yieldday	231.00
solar/114185212345/0/powerdc	0.80
solar/114185212345/0/yieldtotal	488.46
solar/114185212345/0/voltage	229.80
solar/114185212345/0/current	0.00
solar/114185212345/0/power	0.00
solar/114185212345/0/frequency	50.01
solar/114185212345/0/temperature	10.50
solar/114185212345/0/powerfactor	0.00
solar/114185212345/0/efficiency	0.00
solar/114185212345/0/reactivepower	0.00
solar/114185212345/1/name	links
solar/114185212345/1/voltage	16.20
solar/114185212345/1/current	0.02
solar/114185212345/1/power	0.30
solar/114185212345/1/yieldday	116.00
solar/114185212345/1/yieldtotal	246.50
solar/114185212345/1/irradiation	0.08
solar/114185212345/2/name	rechts
solar/114185212345/2/voltage	16.20
solar/114185212345/2/current	0.03
solar/114185212345/2/power	0.50
solar/114185212345/2/yieldday	115.00
solar/114185212345/2/yieldtotal	241.96
solar/114185212345/2/irradiation	0.13

homeassistant/button/dtu_114185212345/turn_inverter_off/config	{"name":"HM-600 Turn Inverter Off","uniq_id":"114185212345_turn_inverter_off","ic":"mdi:power-plug-off","ent_cat":"config","cmd_t":"solar/114185212345/cmd/power","payload_press":"0","dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"}}
homeassistant/button/dtu_114185212345/turn_inverter_on/config	{"name":"HM-600 Turn Inverter On","uniq_id":"114185212345_turn_inverter_on","ic":"mdi:power-plug","ent_cat":"config","cmd_t":"solar/114185212345/cmd/power","payload_press":"1","dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"}}
homeassistant/button/dtu_114185212345/restart_inverter/config	{"name":"HM-600 Restart Inverter","uniq_id":"114185212345_restart_inverter","dev_cla":"restart","ent_cat":"config","cmd_t":"solar/114185212345/cmd/restart","payload_press":"1","dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"}}
homeassistant/number/dtu_114185212345/limit_nonpersistent_relative/config	{"name":"HM-600 Limit NonPersistent Relative","uniq_id":"114185212345_limit_nonpersistent_relative","ic":"mdi:speedometer","ent_cat":"config","cmd_t":"solar/114185212345/cmd/limit_nonpersistent_relative","stat_t":"solar/114185212345/status/limit_relative","unit_of_meas":"%","min":1,"max":100,"dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"}}
homeassistant/number/dtu_114185212345/limit_persistent_relative/config	{"name":"HM-600 Limit Persistent Relative","uniq_id":"114185212345_limit_persistent_relative","ic":"mdi:speedometer","ent_cat":"config","cmd_t":"solar/114185212345/cmd/limit_persistent_relative","stat_t":"solar/114185212345/status/limit_relative","unit_of_meas":"%","min":1,"max":100,"dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"}}
homeassistant/number/dtu_114185212345/limit_nonpersistent_absolute/config	{"name":"HM-600 Limit NonPersistent Absolute","uniq_id":"114185212345_limit_nonpersistent_absolute","ic":"mdi:speedometer","ent_cat":"config","cmd_t":"solar/114185212345/cmd/limit_nonpersistent_absolute","stat_t":"solar/114185212345/status/limit_absolute","unit_of_meas":"W","min":10,"max":1500,"dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"}}
homeassistant/number/dtu_114185212345/limit_persistent_absolute/config	{"name":"HM-600 Limit Persistent Absolute","uniq_id":"114185212345_limit_persistent_absolute","ic":"mdi:speedometer","ent_cat":"config","cmd_t":"solar/114185212345/cmd/limit_persistent_absolute","stat_t":"solar/114185212345/status/limit_absolute","unit_of_meas":"W","min":10,"max":1500,"dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"}}
[... cut ...] 2.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"},"exp_aft":15}
homeassistant/sensor/dtu_114185212345/ch0_ReactivePower/config	{"name":"HM-600 ReactivePower","stat_t":"solar/114185212345/0/reactivepower","unit_of_meas":"var","uniq_id":"114185212345_ch0_ReactivePower","dev":{"name":"HM-600","ids":"114185212345","cu":"http://192.168.178.52","mf":"OpenDTU","mdl":"HM-600, HM-700, HM-800","sw":"g40203f7"},"exp_aft":15,"dev_cla":"reactive_power","stat_cla":"measurement"}
```

knickohr, 2023-01-06:
Meinst Du das hier ?  Das kommt doch rüber @stefan123t 

allerdings habe ich dem noch nie Beachtung geschenkt, ich klaube mir meine Werte von Hand zusammen 😅

[image]

Waldo56, 2023-01-06:
JSON Payloads würde ich sehr begrüßen. Natürlich ist true/false besser als 1/0. 
Bleiben noch die Fehler bei "mqtt" und "device".

knickohr, 2023-01-06:
mqtt ist offenbar eine Last Will Meldung. Das donnert er noch raus bevor er sich verabschiedet. Geht glaub nur als String, bzw. zu großer Aufwand das noch zu ändern.

lumapu, 2023-01-06:
1. `true` und `false` finde ich hier richtig, da hierdurch der Status schnell zu erfassen ist. `1` und `0` werden immer unterschiedlich interpretiert - was heißt zB. `1` ...
2. kann ich @knickohr nur bestätigen, das ist die `last-will` Message, die ist statisch im System hinterlegt und wird schon zu beginn übertragen - ich glaube der Broker zeigt es aber erst als gültig an, wenn die Client die Verbindung verliert / schließt
3. bei mosquitto kann ich unter `<TOPIC>/device` das richtige finden: `AHOY-DTU`

closed 2023-03-08 by lumapu