lumapu/ahoy #549: Wird in der FRITZ!Box nicht gefunden

herrvonbalou, 2023-01-04:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x ] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Gerri1, 2023-01-04:
@herrvonbalou 

- Welche FW- Version?
- z.B. mit Handy WLAN enschalten und im Browser die 192.168.4.1 eingeben!

herrvonbalou, 2023-01-04:
Hallo,

schön, dass du mir helfen möchtest

ich habe die aktuelle aufgespielt und komme auch mit dem iPad in die Einstellung
Heute Morgen wurde es auch im Netzwerk angezeigt ich konnte es mit der Adresse aber nicht anpingen
Das Teil macht mich noch kirre

Viele Grüße

Bernd

Sent with Spark

Gerri1, 2023-01-04:
Das nRF24L01+ Modul ist richtig angeschlossen?
Evtl. plus und minus zum nRF24L01 mal trennen und einen Reset machen, dann nochmal über den AP Modus probieren.

herrvonbalou, 2023-01-04:
Darf ich dir morgen ein Foto vom Anschluss zur Kontrolle schicken?

Sent with Spark

Gerri1, 2023-01-04:
> ich habe die aktuelle aufgespielt und komme auch mit dem iPad in die Einstellung

unter >Settings/Network die SSID und Passwort vom WLAN richtig eingegeben?
Static IP Einstellungen erstmal leer lassen, falls da was eingetragen war!
Speichern und Reboot, danach solltest du den ESP im Router wiederfinden.

herrvonbalou, 2023-01-05:
Guten Morgen,
es geht weiter
Ganz blöder Fehler wlan Schlüssel und Fritzbox Kennwort ist halt ein Unterschied
Komme jetzt auf den logger und habe den Schlüssel vom hm1500 eingetragen
Bekomme aber noch keine Daten
Beim serial/control geht etwas habe ein Foto angefügt
Vielleicht hast du eine Idee

Gruß Bernd
Sent with Spark

lumapu, 2023-01-05:
@herrvonbalou leider ist das Bild nicht angekommen

Gerri1, 2023-01-05:
@herrvonbalou Bilder bitte direkt in GitHub hochladen (ins Nachrichtenfenster ziehen)! 😁

herrvonbalou, 2023-01-05:
[image]

Gerri1, 2023-01-05:
Welche Pins für CE und IRQ hast du genommen, evtl. musst du da andere nehmen!

herrvonbalou, 2023-01-05:
D4 und D3

herrvonbalou, 2023-01-05:
Muss ich dann aber auch in der konfig so konfigurieren oder?

Gerri1, 2023-01-05:
... mimm mal D1 und D2 (HW und Software)

Gerri1, 2023-01-05:
[image]

Gerri1, 2023-01-05:
... evtl. auch mal die Leistung des nRF mal auf MIN stellen, wenn dass das mit der externen Antenne ist!

herrvonbalou, 2023-01-05:
Kann machen was ich will immer der gleiche Fehler! 
Am Wechselrichter selbst kann es nicht liegen?

Gerri1, 2023-01-05:
Stimmt die Seriennummer unter Settings/Inverter und ist der Haken drin bei >Communication Enable< ?

Gerri1, 2023-01-05:
Da musst du das morgen noch mal probieren, jetzt geht eh nix mehr!

Gerri1, 2023-01-05:
Mache Bitte morgen mal einen Screenshot von der Systemseite!

stefan123t, 2023-01-06:
Und bitte auch mal ein Foto von Deinem Aufbau hier hochladen, damit man die Module und deren Verkabelung mal gesehen hat.
Meist reicht auch ein Tausch von D3 & D4 für IRQ und CE.
Du hast keinen 10-100uF Elko direkt am NRF24 Modul (Pin 1&2 GND / +3.3V) angebracht / angegeben, 
das gibt so leider kein Fleißbienchen :bee: für Dein Projekt! 
Auch solltest Du evtl. mal die Stromversorgung bzw. das USB Kabel des ESP Moduls durch ein anderes, qualitativ hochwertiges tauschen.

herrvonbalou, 2023-01-06:
Na das mit dem Fleißbienchen könnte ich ja verschmerzen 
Ich werd mit dem GitHub noch nicht so richtig warm !

herrvonbalou, 2023-01-06:
[image]
[image]
[image]
[image]

herrvonbalou, 2023-01-06:
Kondensatoren hab ich bestellt 
Spannung habe ich 3.3 Volt gemessen

Gerri1, 2023-01-06:
@herrvonbalou 
Wie ist denn die Distanz zum Wechselrichter?
Evtl. wenn möglich, mal die zwei Teile nacheinander Tauschen!
Ich habe schon des öfreren gelesen, das die nRF's nicht funktioniert haben!

herrvonbalou, 2023-01-06:
Hatte ihn schon nah (5meter) und weiter weg 
nRF hab ich schon getauscht 
Morgen kommt noch ein Kondensator und dann gehen mir langsam die Ideen aus

Gerri1, 2023-01-06:
... und einen anderen ESP?

herrvonbalou, 2023-01-06:
noch nicht, da sind andere pins dran. versuche ich nachher noch umzulöten

Gerri1, 2023-01-06:
... sehr, sehr komisch! 😯
Mit den D1 minis hatte ich ja auch Probleme mit CE und IRQ, aber seitdem die an D1 und D2 dran sind, funktionieren die schon super!

dtuuser, 2023-01-06:
Bei mir #535 jetzt alles perfekt.

Mit D1 und D2, Kondensatoren 100er oder 220er und mit Power Level LOW. Sogar ein Parallelbetrieb von zwei "Sticks" mit Version 66 und Beta 68 möglich.

Mit einem 10er Kondensator klappte es nicht

Perfekte Alternative für die HM DTU Hardware.
Vielen Dank für das Projekt

herrvonbalou, 2023-01-06:
Wie merke ich welcher KONDENSATOR MORGEN RAN MUSS?

Gerri1, 2023-01-06:
Ich würde es jedenfalls mal mit einem 100µF Kondensator probieren!
Aber auch mal mit einem anderen ESP probieren, nicht das der einen Treffer hat!

herrvonbalou, 2023-01-06:
Hat so ein Kondensator eine Einbaurichtung?
Hatte jetzt 2 Miniwemos bestellt Einer geht gar nicht! Der andere mit dem gezeigten Problem
Na ja, hab dafür im Lotto kein Geld

Viele Grüße aus Albstadt sendet

Bernd Güttinger

Sent with Spark

herrvonbalou, 2023-01-06:
Sollte Glück heißen

Viele Grüße aus Albstadt sendet

Bernd Güttinger

Sent with Spark

Gerri1, 2023-01-06:
... und die Verkabelung CE -> D2 (GPIO04) und IRQ -> D1(GPIO05) sind auf alle Fälle richtig, auch in den Einstellungen?

Gerri1, 2023-01-06:
> Hat so ein Kondensator eine Einbaurichtung?
[image]

lumapu, 2023-01-07:
nein muss nicht da sein, kann zu einer beliebigen Zeit verdrahtet werden.
Du kannst auch schon alle Einstellungen machen, wenn du dann das NRF richtig verbindest geht es direkt los.

herrvonbalou, 2023-01-13:
So ist der momentane Stand:
Es funktioniert! 

Noch einmal ein Original Lolin Wemos bestellt alles gleich angeschlossen und programmiert und nun läuft es absolut problemlos 

Also waren 2 neu bestellte Wemos Clone defekt so ein Mist 

Soweit so gut
Da ich nun sehe was vom Dach kommt ist mir etwas komisches aufgefallen 

Ich habe 4 Module an einem HM1500 und es wird auch an allen 4 Spannung angezeigt aber nur an 3 Strom und das schon seit ich es vor 5 Monaten montiert habe

An was kann es liegen wenn ein Modul zwar Spannung aber keinen Strom anzeigt?

Das ganze wäre mir ohne das AHOY DTU vermutlich nie aufgefallen

herrvonbalou, 2023-01-13:
Weiß jemand welcher Anschluss am HM zu welchem Ausgabefeld an der DTU gehört?

lumapu, 2023-01-13:
@herrvonbalou ja, von links nach rechts in Ahoy ist 1 -4, auf dem Wechselrichter stehen diese Nummern hinten drauf.

Bin gespannt was der Fehler ist. Kann nur ein riesiger Widerstand sein, oder ein Kanal nicht angeschlossen, da die Spannung beim HM1500 immer für zwei Eingänge zusammen erfasst wird. Also Kanal 1 und 2 haben eine Spannung und 3 und 4.

herrvonbalou, 2023-01-14:
Hallo, komme gerade nicht weiter 
Der Fehler wandert mit.
Wenn ich umstecke ist ein anderer Kanal ohne Leistung 
Also ist es das Modul!
Wenn ich an dessen Leitungen messe liegen ungefähr 40 Volt Gleichstrom an
Hat noch jemand eine Idee?

lumapu, 2023-01-14:
@herrvonbalou ja den Kurzschlussstrom messen, der sollte je nach Modul ca. 8 bis 14A sein. Steht auch hinten drauf.

Finde toll, das Ahoy dazu beigetragen hat einen Fehler der Anlage zu offenbaren. Hoffentlich kannst du den Fehler beheben und hast dann 25% mehr Ertrag.

closed 2023-03-27 by lumapu