lumapu/ahoy #550: PowerLimit MQTT iobroker

MubiTec, 2023-01-05:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen,

kann mir jemand kurz auf die Sprünge helfen? Ich würde gerne das PowerLimit non-persistent in absolut anpassen. Ich habe dazu diesen Datenpunkt erstellt:
mqtt.0.Ahoy.ctrl.limit_nonpersistent_absolute.0
Setze ich einen Wert ein als payload, so erscheint in der DTU, dass er ein MQTT Befehl erhalten hat, aber man sieht keine Payload und es passiert leider nicht.

Danke schon mal!

Matze7779, 2023-01-05:
Hallo,

versuche das auch gerade hinzubekommen.
Wo siehst Du in der DTU das der MQTT Befehl angekommen ist?

Matze7779, 2023-01-05:
Nachtrag. In einer älteren Version hat es noch "devcontrol" hieß hatte ich es laufen.
Keine Ahnung was jetzt anders ist.

Gerri1, 2023-01-05:
inverter/ctrl/limit_nonpersistent_relative/0   70    (70%)
Einen Datenpunkt habe ich nit erstellt!

Matze7779, 2023-01-05:
> Einen Datenpunkt habe ich nicht erstellt!

Ahoy hat den bei dir angelegt in IOBroker?

Gerri1, 2023-01-05:
Den Datenpunkt gibt es bei mir nicht!
Ein Skrript erstellen mit dem Inhalt (Werte anpassen) >sendTo('mqtt.0', 'sendMessage2Client', {topic: 'AHOY-DTU/devcontrol/0/11/1', message: '100'});<

MubiTec, 2023-01-05:
> Hallo,
> 
> versuche das auch gerade hinzubekommen. Wo siehst Du in der DTU das der MQTT Befehl angekommen ist? Nachtrag. In einer älteren Version hat es noch "devcontrol" hieß hatte ich es laufen. Keine Ahnung was jetzt anders ist.
> 
> [image]

genauso wie du ist es auch bei mir. Dass was angekommen ist sehe im "Serial / Control"

jimknopf63, 2023-01-05:
Topic inverter/ctrl/limit persistent absolute/0 und dann als Message 600 z.B. Das sind die gültigen Werte ab 0.5.50
[image]

Gerri1, 2023-01-05:
https://github.com/lumapu/ahoy/blob/main/User_Manual.md

[image]

MubiTec, 2023-01-05:
ich verstehe es nicht, habe jetzt im Skript diesen Befehl:
sendTo('mqtt.0', 'sendMessage2Client', {topic: 'Ahoy/ctrl/limit_nonpersistent_absolute/0', message: '551'})
im Serial sieht man, dass eine MQTT Nachricht ankommt, aber wieder passiert nix. Ich verstehe es nicht :-(

Gerri1, 2023-01-05:
das kann manchmal ein paar minuten dauern, bis das in der Live - Ansicht zu sehen ist!

Gerri1, 2023-01-05:
so sieht mein Blockly aus. ... nicht lachen! 😬

[image]

MubiTec, 2023-01-05:
nun gehts, sogar mit der Datenpunkt-Methode. Hab in der System-Config das Interval von den Standarf 5s auf 30s hochgesetzt. Weiß nicht, ob das jetzt Zufall war, aber nun gehts :-)

closed 2023-01-05 by MubiTec

Matze7779, 2023-01-05:
> Hab in der System-Config das Interval von den Standarf 5s auf 30s hochgesetzt.

Tatsache. Geht.

Matze7779, 2023-01-05:
Und wieder auf 5s umgestellt. Geht immer noch ?!?!?!?

MubiTec, 2023-01-05:
Manchmal verrückt. Aber wenns jetzt geht um so besser 😁

Matze7779, 2023-01-05:
Mal für die Nachwelt:

Habe kein Blockly Skript dazu laufen.
Nur diesen Datenpunkt erstellt. Anscheinend muss man mit dem Interval etwas spielen. Evtl. ein Bug.

[image]