lumapu/ahoy #551: JSON Settings upload?

DennisXK, 2023-01-05:
Wie bzw. wo kann man die Settings JSON wieder hochladen? Denn es ist ziemlich nervig nach jedem update immer die ganzen Einstellungen wieder vorzunmen.

TB-79, 2023-01-05:
Bei meinem letzten Update sind die Einstellungen erhalten geblieben. Wenn man über Update in ahoy geht .
Oder flasht du jedesmal über das Tool neu?

mroenne2022, 2023-01-05:
@DennisXK  @TB-79 
Ich hab kürzlich bei 2 DTU`n ein Update (auf 0.5.66) direkt im Menü von der Ahoy durchgeführt. Die Einstellungen zum WLAN und Inverterdaten blieben ebenfalls erhalten ! Beide waren anschliessend wieder über die IP in meinem WLAN erreichbar.

knickohr, 2023-01-05:
Die Einstellungen bleiben eigentlich schon ab der DEV 3x erhalten wenn man OTA flasht. Gehen nur verloren wenn irgendwas beim Update in die Hose geht. Bei mir war es meist das Wurstfingersyndrom, indem ich öfters auf den Update-Button gekommen bin. Aber das ist auch schon gefixed und kann nicht mehr vorkommen.

Natürlich wäre der Import des JSON wünschenswert wenn doch mal wieder was schief geht oder man auf andere Hardware umschwenkt 😉

stefan123t, 2023-01-06:
@lumapu das sollte doch eigentlich mit einem `POST` an `/save` und der alten `settings.json` als `Content-Type: application/json` funktionieren, oder ?
Der `/save` endpoint ist an `showSave()` bzw. über die `app` und das `appInterface` and `saveSettings()` in der `settings.h` gebunden ist. 
Dort wird dann der JSON Payload entsprechend ins FileSystem / die Config Datei geschrieben.

Am besten könnte man das Upload Form direkt auf der Settings Seite neben dem Download Button platzieren, der den Endpoint `/get_setup` für den `onDwnldSetup()` Handler für die `getSetup()` Methode aufruft.

lumapu, 2023-01-06:
nein, aktuell akzeptiert der endpoint kein JSON, ich denke aber dass das nicht viel Arbeit ist. Man kann das kein empfangen, evtl. paar checks durchführen und dann ins Filesystem schreiben.

DennisXK, 2023-01-06:
@mroenne2022 bei mir klappt das updaten meistens nie, deswegen habe ich gefragt!

lumapu, 2023-01-06:
wird eingebaut 😊

mroenne2022, 2023-01-06:
@DennisXK : Bei etwas älteren Versionen hatte ich das auch schon das Einstellungen verloren waren. Bei meinen letzten Updates allerdings nicht, da war alles vorhanden. Lediglich Communication enable noch aktivieren, das wars. Warum es bei dir selten klappt weiß ich dann nicht.

DennisXK, 2023-01-06:
Wollte ebene von der 0.5.41 auf die 0.5.66 updaten...da kam dann das das update fehlgeschlagen ist und er startet in 20 sec neu. Somit muste ich wieder neu flashen 🙈🤷‍♂️

MrLee99, 2023-02-07:
Moin! 

teste gerade mehrere ESPs...da wäre ein upload der config sehr hilfreich. Leider ist der upload button ohne Funktion (mouseover geht, keine reaktion beim drücken)
Bin ich mit dem Problem alleine?

stefan123t, 2023-02-08:
@MrLee99 sollte ab 0.5.70 behoben sein. 
Die aktuelle Version aus dem Development03 Branch ist 0.5.83.
Welche Version hast Du denn ?

closed 2023-03-27 by lumapu