lumapu/ahoy #553: Feature Request: Rest-Api call to get result of adjust power level

modem-man-gmx, 2023-01-05:
## Hardware
  * [X] ESP8266
  * [X] ESP32
  * [ ] Raspberry Pi

Modelname: Any
Retailer URL: n.a.

### nRF24L01+ Module
n.a.

## Software
* [X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: any next version

**From Discord discussion:**
Mo had a question about the statement
"http://192.168.0.142/api/ctrl"
and there the payload 
`{"id": "0", "cmd": "limit_persistent_absolute", "val": "30"}` 
must be POSTed as content type `application/json`:

Q (Mo): Is there an immediate return code from which one can see whether the HM understood and implemented the command?

A (lumapu): Yes, it is read back and the value is then also transmitted via MQTT, but it feels like it takes about a minute. I don't know the exact MQTT path at the moment

Q (Mo): It would be nice to also have a rest-api call for this, then you can process the order and the check via the same interface (and maybe don't have to wait up to 1 minute?).

stefan123t, 2023-01-05:
IMHO the two call(s) should remain async as the REST API would otherwise be blocked not only until the [`0x0B ActivePowerContr`](https://github.com/lumapu/ahoy/wiki/Protocol/#wie-funktioniert-das-active-power-limit-0x0b-sub-kommando) / Limit Command has been accepted by the Inverter. But the ask here to wait until the [`0x05 SystemConfigPara`](https://github.com/lumapu/ahoy/wiki/Protocol/#systemconfigpara--0x05) where the inverter updates the DTU on the current Power Limit set, may return a valid result only some ~1 minute after the successful 0x0B command has been sent and received / acknowledged.

lumapu, 2023-01-06:
1. We could introduce another Mqtt publish to show the acknowledge of new power limit setpoint.
2. After approx. one minute the new limit will be published of the Readback (0x05) value. The second point is already implemented.

modem-man-gmx, 2023-01-07:
Will check this soon, thank you!

lumapu, 2023-01-07:
for me the acknowledge works as expected, the field `ack_pwr_limit` was set to `true` at `07:50:58`, the field `active_PowerLimit` (read-back value) was set to `100` at `07:51:30`

closed 2023-03-27 by lumapu