lumapu/ahoy #556: Die aktuelle IP-Adresse bitte mit über MQTT publishen

matthias882, 2023-01-06:
## Hardware
  * [x ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [x ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x ] Windows
* [ ] Mac OS

### Debugging:
* [x ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo,

ich würde darum bitten die aktuell vergebene IP-Adresse der AHOY-DTU mit über MQTT bereitzustellen. Dies erleichtert den Weg um bei DHCP-Systemen auf die DTU zu gelangen. Auch wären somit dynamische Links aus Visualisierungsoberflächen möglich. Der Umweg über die DHCP-Tabelle des DHCP-Servers würde somit entfallen.

MfG
Matthias

knickohr, 2023-01-06:
Hast Du schon mal den Send-Befehl im MQTT Settings ausprobiert ?  Meines Wissens wird da die IP übermittelt :

[image]

Gerri1, 2023-01-06:
@knickohr  Die IP-Adresse ist nicht mit dabei!
Bei OpenDTU schon!

Gerri1, 2023-01-06:
In den ioBroker wird nix mit übertragen! ☹

Gerri1, 2023-01-06:
free_heap ist mit drin, da wäre die IP-Adresse informativer.

knickohr, 2023-01-06:
Da gebe ich Dir Recht, auch die Version könnte nur einmal übertragen werden.

Je öfter ich darüber nachdenke, halte ich den Vorschlag das Ganze als JSON zu verschicken für immer mehr sinnvoller 😲

Gerri1, 2023-01-06:
Ich finde das so schon gut, im JSON muss ich wieder einen ganz schönen Aufwand betreiben, die Daten da raus zubekommen! 😒

knickohr, 2023-01-06:
Ja, das ist es was mich auch abschreckt. Prinzipiell machen wir das aber sowieso irgendwie, und sei es nur wenn wir den Topic zerpflücken. Aber da gehen die Mein7ngen auseinander. Ich möchte jetzt ungern meine Flows neu machen nur weil es jetzt JSON ist.

a
Außerdem würde der ursprüngliche Sinn und Zweck von MQTT untergraben werden, so viel Information wie möglich in so wenig Daten wie möglich zu übermitteln, vor allem beim Subscribe.

Gerri1, 2023-01-06:
Eigentlich bin ich mit der MQTT - Daten - Übermittlung, bis auf ein paar kleinigkeiten, sehr zufrieden!

knickohr, 2023-01-06:
Natürlich !  Ist Meckern auf sehr hohem Niveau, aber alkes kann man besser machen 😉

knickohr, 2023-01-06:
… und bitte nur einmal ausgeben nach Boot oder Änderung 😉 @lumapu 

Vielleicht sollten wir das Thema MQTT mal komplett neu aufrollen und überdenken. Ich sehe aber leider schon wieder den Shitstorm 😪

JSON kann man sich doch per REST-api abholen wenn ich das richtig verstanden habe im Discord.

matthias882, 2023-01-06:
Wow, was für eine riesige Diskussion habe ich da denn losgetreten? ;)
Bitte als einzelne Subscriptions lassen, das ist äußerst praktisch um direkt auf die 2 oder 3 Werte zugreifen zu können die man dauerhaft wirklich braucht und auch loggen will. Nur die IP-Adresse eben noch dazu.

Und übrigens ein ganz großes Dankeschön für diese tolle Arbeit. Ich lese das ganze Projekt schon seit den ersten Tagen auf Microcontroller.net mit und dachte mir die Tage ich könnte es ja endlich mal ausprobieren. Und was soll ich sagen, ich bin begeistert wie schnell und reibungslos es beim ersten Versuch in Betrieb ging und wie klasse die Bedienung und Parametrierung gemacht ist. Da können sich meine eigen WLAN-Sensoren ne Scheibe abschneiden ;)

Gerri1, 2023-01-06:
> JSON kann man sich doch per REST-api abholen wenn ich das richtig verstanden habe im Discord.

... sollte so sein!

knickohr, 2023-01-06:
@matthias882 

MQTT ist **immer** ein Reizthema, habe mir da auch schon blutige Nasen geholt 😅

lumapu, 2023-01-06:
@knickohr ich glaube, wenn ich mir die letzten Diskussionen so anschaue, dass wir das nicht 'einfach so' definieren können. Es ergeben sich einfach über die Zeit neue Anforderungen und wir bessern Stück für Stück nach und testen dann wieder. Wenn du konkrete Ideen hast und die Diskussion nicht scheust, dann raus damit :stuck_out_tongue_winking_eye:

knickohr, 2023-01-06:
@lumapu 

Das nennt sich Feigheit vor dem Feind. Ich laufe nicht weg und verkrieche mich 😂

stefan123t, 2023-01-07:
@lumapu das sollten wir mE schon einmal grundlegend auf die Beine stellen. Vielleicht mit einem kleinen Konzept (Spreadsheet?) in dem wir u.a. auch die MQTT Topics von AhoyDTU und OpenDTU abgleichen. 

@knickohr Du könntest doch mal einen Vergleich der beiden Topic Bäume erstellen und in einem neuen Issue als Anhang posten. Vielleicht kann dan @DukeWolle auch mit drüber schauen ob es aus seiner Sicht auch passt und evtl. kommt dann sogar @Ichirou dazu und schaut sich das im Vergleich mit den ModBus485 Werten an ?

closed 2023-03-27 by lumapu