lumapu/ahoy #557: Web Interface tot aber MQTT funkt Daten

Skusi, 2023-01-07:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Morgens nach Tagesstart ist die Web Oberfläche zwar erreichbar, aber es wird nur die Hauptseite ohne Daten angezeigt.
Bedienung nicht möglich.
Anderere Browser oder reload der Seite hilft nicht.
MQTT sendet weiter korrekte Daten.
Nur eine reboot per [http//](http://[IP]
[image]
/reboot hilft.

lumapu, 2023-01-07:
Das schreit nach Kondensator. Habe auch ein Wemos Board, das auch ohne NRF genau das gleiche Verhalten zeigt. Mit Elko habe ich es nicht mehr beobachten können.

Da gibt's im Hintergrund eine Exception (der ESP bootet neu), deswegen fehlt zB das Menü (wird asynchron geladen). Die Exception sieht man nur per USB serial debug

Skusi, 2023-01-07:
Ok, allerdings habe ich einen 100uF und 10pF an der 3,3V Leitung von Down-stepper  hängen !?
Ich weiß, hab ich nicht richtig oben angegeben - sorry. Ich war etwas überfordert mit dem Ausfüllen der Daten.

Neu bootet die DTU sowieso öfters. Meine Uptime erreicht selten mehr als 20 Std. Aber das Interface ist dann immer wieder normal da.

lumapu, 2023-01-07:
Ok, dann evtl. ein anderes USB Netzteil verwenden. Für mich ist das ein Powerproblem, da ich selbst gleiche Probleme hatte und die Version `stable` ist

maikt, 2023-01-08:
Nach der Installation der "221230_ahoy_0.5.66_esp32_ssd1306_f8fe044.bin" Firmware (also mit Display Support) hatte das gleiche Problem. Das Gerät startet, das WebInterface funktioniert, MQTT funktioniert  und am Display werden auch aktuelle Daten angezeigt. Nach ein paar Minuten funktioniert das WebInterface dann aber nicht mehr, über MQTT werden aber noch Daten übertragen und auch am Display angezeigt.
Den 100µF Kondensator ist bei meinem Gerät verbaut.
Abhilfe schaffte die Installation der Firmware ohne Display Support. Seit dem läuft alles.

tqma1, 2023-01-09:
Ich hatte die Version mit Nokia Display OHNE aktiviertes MQTT problemlos am laufen,
seit der Aktivierung von MQTT tritt das Problem bei mir auch auf.
Deckt sich somit aus meiner Sicht mit der Aussage von maikt …

lumapu, 2023-01-11:
ok, das sind gute Hinweise nach einem Fehler zu suchen. Was ich noch nicht verstehe ist der Zusammenhang mit dem Webinterface.

Matze7779, 2023-01-12:
Das Problem hatte ich auch vor einer Woche.
Update installiert und alles ok.
Nächsten Tag war die Weboberfläche tot.
Tags drauf ging wieder alles bis heute.

Seltsam.

lumapu, 2023-02-07:
ist wieder aktuell siehe #660. Ich mache hier zu, wir diskutieren dort weiter (wegen Issue auch in neuester Verison)

closed 2023-02-07 by lumapu