lumapu/ahoy #562: Wo finde ich die Datei(en) zum OTA Update

MatthiasFrenzel, 2023-01-08:
Hallo,
die erste Installation per flash tool habe ich vor einigen Wochen erfolgreich geschafft. Nun scheitere ich am OTA Update, weil ich nicht weiß wo ich die richige Datei finde. Das müsste doch eine "bin" sein, oder? Ich bin aber auch komplett neu hier...
Zum Download finde ich nur die zip files. 
Danke
Matthias

MubiTec, 2023-01-08:
Die zip files musst du entpacken und dann die bin oza flashen

MatthiasFrenzel, 2023-01-08:
dann habe ich wohl eine falsche Datei. Da ist nämlich keine bin drin.. Über einen andere Seite habe ich einen Link auf eine zip gefunden, die diverse bin enthält. Das ist aber 230107_ahoy_0.5.69_...bin und nicht die latest stable version?!

MubiTec, 2023-01-08:
https://github.com/lumapu/ahoy/releases/tag/ahoy_v0.5.66Hier liegt dir aktuelle latest--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

MatthiasFrenzel, 2023-01-08:
Danke, die habe ich auch gefunden. Aber ich sehe darin keine .bin Datei. Oder zeigt mein PC das falsch an???

MubiTec, 2023-01-08:
Also bei mir sind da bins drin 😁

MatthiasFrenzel, 2023-01-08:
Bei mir jetzt auch... confused
Ich habe die alle 3 schon mal angesehen.

danke für die Hilfe!

lumapu, 2023-01-09:
sehr gut, dass es gelöst werden konnte. Kann man hier zu machen?

Tjossul, 2023-01-16:
Hallo,

ich möchte mich auch nochmal kurz hier einklinken...

Ich wollte eine OTA von der Version v0.5.41 zur 0.5.66 machen.

Ich habe also Zip-Datei entpackt und wollte die ota.bin per Web-Update hochladen. Dann kam aber ein "Update failed". So habe ich dann andere "bin"-Dateien versucht. Schließlich habe ich mir alles zerstört, so daß ich nochmal von vorn anfangen mußte (mit NodeMCU flashen usw.).

Ich wollte nur nochmal ganz sicher gehen - welche Datei im Release-Ordner muß genau ausgewählt werden? Ist es die ota.bin?

Danke & Gruß!
T.

lumapu, 2023-01-16:
@Tjossul 
nein, es ist immer zB. eine `221230_ahoy_0.5.66_esp8266_f8fe044.bin` mit `esp8266` im Namen oder für ESP32 eine Firmware mit `esp32` im Namen richtig.

Tjossul, 2023-01-17:
@lumapu 
OK, dann muß ich beim nächsten Update nochmal genau drauf achten - ich könnte schwören, auch von der Reihenfolge her, daß ich die richtige zuerst versucht habe. Ich bin mir aber auch nicht zu 100% sicher. Danke nochmal für Deine Antwort.

lumapu, 2023-01-17:
@Tjossul es gibt keine Reihenfolge bei OTA. Es muss _nur_ die Firmware geladen werden.

Toto-72, 2023-01-19:
Ich hatte 0.5.41 OTA geflashed. Mit der 0.5.66 bekomme ich auch immer die Meldung "update failed". Habe ESP8266.

lumapu, 2023-01-21:
@Tjossul probiere doch mal 
https://ahoydtu.de/web_install/

closed 2023-03-08 by lumapu