lumapu/ahoy #564: HW Watchdog und Exceptions

Waldo56, 2023-01-08:
Hi,
sehe eigentlich schon seit der Installation von 0.5.66 immer wieder eine reboot_reason HW Watchdog oder Exception (siehe Screenshots). Das passiert eigentlich täglich. 
Ich habe MQTT enabled. Ein HM800 ist ca. 40m von der DTU entfernt. Das führt immer wieder zu Wiederholungen beim Datentransfer. Die Daten beider HM800 scheinen aber trotzdem nahezu vollständig und plausibel. Leider weiss ich nicht wie ich bei der Ursachenfindung behilflich sein könnte. Habt Ihr dieses Verhalten auch schon beobachtet?
[image]
[image]

knickohr, 2023-01-08:
Da Du ein recht kräftiges wifi_rssi hat, kann es daran nicht liegen. Aber mit dem Power Level high könnte das erklärbar sein. 40m ist schon ein Stück, geht auch weniger Leistung ?  Kondensator verbaut ?

dtuuser, 2023-01-08:
Bei mir seit Version 0.5.66 habe ich auch fast täglich einen Reboot wegen dem Hardware Watchdog. Habe irgendwo gelesen, dass es bei den AZ Minis an der Nutzung von Port D1 oder am Stromverbrauch oder fehlendem bzw. zu kleinem Kondensator liegen kann.

knickohr, 2023-01-08:
Das würde erklären das ich nur ca. Alle 2 Tage mal einen Reboot habe. Bei mir sind dicke Elkos zusätzlich mit verbaut 😉

was ist mit D1 ?  Hast Du nähere Infos ?

Gerri1, 2023-01-08:
Schaut bei mir so aus!
[image]
[image]

knickohr, 2023-01-08:
Joo, Du bist auch nicht up to Date, wir sind schon bei 69 😅

Bei dem sieht es momentan so aus :

[image]

Wenn Lukas nicht wieder vorher eine neue DEV hervorzaubert 😂

Gerri1, 2023-01-08:
Ich kann erst wieder morgen Abend updaten! 😩

dtuuser, 2023-01-08:
Ich suche mal meinen Verlauf ab. Sobald ich die Infos zu Pin D1 gefunden habe, poste ich.

Gerri1, 2023-01-08:
Am D1 hängt bei mir jetzt das Display mit SCL dran. Mit IRQ an D1 und CE an D2 st bei meiner NodeMCU V3 nichts mehr gegangen! 😭

knickohr, 2023-01-08:
Ähhh, ich weiß gar nicht wie ich verdrahtet habe. Halt Standard Ahoy und eben Standard Nokia, bzw OLED

Waldo56, 2023-01-08:
Kondensator ist verbaut.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

Gerri1, 2023-01-08:
Bei der NodeMCU ist auch wieder alles Standard. Beim Wemos D1 mini muss ich mal schauen, wie da jetzt die Verdrahtung ist!
Aber Hauptsache es funktioniert. 😁

dtuuser, 2023-01-08:
Zu dünnes bzw. falsch geschirmtes USB Kabel könnte auch ein Problem sein. 
Oder wie hier beschrieben.....
https://www.mikrocontroller.net/topic/408026           ist zwar schon älter
.... äußere Störeinflüsse über nicht genutzte Pins bzw. Leiterbahnen.
Habe bei mir unbewusst gestern noch ein dickeres USB Kabel genommen, da ich die Hardware in eine Abzweigdose gebaut habe. Seit gestern 12 Uhr kein Reset mehr.

Waldo56, 2023-01-08:
Das ist meine Konfiguration.
[image]

Gerri1, 2023-01-08:
Die Kabel zum nRF wirken auch wie eine Antenne und sammeln auch was mit ein!

Waldo56, 2023-01-08:
100uF sind direkt an den Pins vom NRF.

knickohr, 2023-01-10:
Bei der DEV70 knurrt die Bulldogge auch wieder öfters 😪

Keine Nacht ohne Reboot, unter Tags aber offenbar alles OK.

Waldo56, 2023-01-10:
Kann es sein, dass auch ein gesteuerter Reboot manchmal zu einer reboot_reason = Exception führen kann? Nachdem ich in der Vergangenheit Probleme mit der Erreichbarkeit der DTU hatte, habe ich eine Automatisierung eingebaut, die immer um Mitternacht einen reboot auslöst. Heute hatte ich Exception als reason. Komisch.

knickohr, 2023-01-10:
Ja, beim OTA-Update habe ich auch hin und wieder Exception.

lumapu, 2023-01-11:
ja man sieht beim reboot auf der Seriellen Konsole eine Exception, der ich nicht weiter nachgegangen bin

stefan123t, 2023-01-11:
@Waldo56 Du hast den NRF24 auf Power Level HIGH, geht er mit LOW / MIN nicht auch genau so gut ?

Waldo56, 2023-01-12:
Habe ich nicht probiert bei ca. 40m Entfernung.

knickohr, 2023-01-17:
Ich habe da mal ein bißchen rum gespielt und kann den Pitbull richtig oft bellen lassen 😂

Je niedriger ich mit dem Abfrageintervall von den Invertern gehe, umso öfter knurrt der Wachhund. Ich frage mich nur warum ?  Was ist das für ein Unterschied ob ich alle Minute meine Inverter abfrage oder alle 20s ?  Ja klar, irgendwann wird die Zeit zu kur6 bevor eine Antwort kommt, aber 20s sollte drin sein. Zumal er nicht in 20s alle Inverter abfragt, sondern nur einen nach dem anderen. Bei 6 oder gar 8 Invertern würde ich dann bei der voreingestellten Zeit für einen kompletten Abfragezyklus 3, respektiv 4 Minuten brauchen 😲

Was veranlaßt den Watchdog hier zu resetten ?   Was hat der Watchdog denn für eine Zeit, ist sie vielleicht zu kurz ?

Waldo56, 2023-01-17:
Ich habe das Intervall auf 60s bei 5 retries und 2 Wechselrichter. Habe den Watchdog trotzdem. 
Wie funktioniert eigentlich das WD Konzept bei der Ahoy? Könnte schon sein, dass die Überwachungszeit zu kurz gewählt ist.

knickohr, 2023-01-17:
Ich weiß auch nicht wie der Hund bei Ahoy gefüttert wird. Meines Wissen beißt der spätestens nach 8 Sekunde. Könnte mir schon vorstellen das da irgendwelche Abfrageroutinen zum Wechselrichter länger dauern. Vielleicht muß man ihn einfach nur öfters mal füttern ? 🤔

lumapu, 2023-01-18:
@knickohr wir füttern den Hund schon sehr viel, jedes `yield()` macht genau das. Evtl. ist er nach kurzer Zeit überfressen 🤣

stefan123t, 2023-01-18:
Ich vermute wenn er (aufgrund IRQ) nicht mehr dazu kommt seine normalen Aufgaben abzuarbeiten und der Stapel an wartenden RX-Paketen, neu zu sendenen Commands, zu sendenen MQTT Nachrichten und zu beantwortenden HTTP UI/API Anfragen zu groß wird, dann schnappt der Waldi halt irgendwann auch mal zu.
Wenn man nun nicht wie empfohlen alle 15-30 Sekunden eine Anfrage an den/die WR schickt, sondern gleich alle 5 Sekunden oder gar noch öfter, dann passiert das halt relativ schnell und der Hund wird dann einfach bissig. 
Es ist also m.E. nicht eine Frage des Hundes sondern es kommt tatsächlich auch hier auf den Halter an.

knickohr, 2023-01-18:
15 Sekunden funktioniert bei 2 WRs nich mehr, da die Zeit zum pollen, zumindest wenn die WRs offline sind, schon zu lange dauert. Bin deshalb auf 3 Retry runter, das reicht dann noch, zumindest sind dann bei mir noch ca. 3s Pause.

Wollte eigentlich erreichen das ich zumindest alle 30s von allen WRs eine Antwort bekomme. Scheint wohl nicht zu funktionieren. Leider sind das halt bei 6 WRs im Endausbau, vielleicht sogar 8 😲, schon mal 4 Minuten bis alle Daten von allen WRs up to date sind.

knickohr, 2023-01-21:
Ich hab’s mal ein paar Tage beobachtet. 90% der Waldi-Reboots passieren bei mir in der Zeit zwischen dem „communication start“, bzw. „communication Stop“ und dem „online“ bzw. „Offline“. Also in der Zeit wo die DTU ins leere pollt.

lumapu, 2023-01-22:
@knickohr danke für die Beobachtung. Das heißt in Umkehrschluss, dass man am besten sunrise und Sunset als Start und Stop nehmen sollte.
Der älteste offene issue so genau an der Kommunikation Verbesserungen bringen.

knickohr, 2023-01-22:
@lumapu 

Nein, das heißt es nicht, da ansonsten wieder Daten von den WRs abgewürgt werden. Nicht zurück entwickeln, so lassen wie es ist. Viel mehr sollte man sich Gedanken machen warum es beim „ins Leere“ pollen zu den HW-Watchdog Reboots kommt. Muß doch irgendwas passieren, wenn keine Antwort mehr kommt. Meine Vermutung ist, das hier irgendwas klemmt und der Waldi dann zuschlägt. Ich beobachte das auch hin und wieder am Morgen beim Start, aber längst nicht so oft wie Abends.

knickohr, 2023-01-22:
Heute ist natürlich nichts passiert als ich mitgehört habe 🤪

Aber 2 Dinge sind mir aufgefallen :
Nach dem 2. Retransmitt legt er eine kurze Pause für 2 bis 3 Sekunden ein 😲 Danach folgen die restlichen 3 in gleichem Zeitsbstand.

Bei Inverter #0 sieht das erste TX irgendwie anders aus wie bei #1 🤔
[image]
[image]

 Vielleicht hilfts irgendwie weiter. Ich werde jetzt erst mal auf 2 Retransmits umstellen. Mein Verdacht ist einfach immer noch das der Waldi irgendwann zubeißt weil die Zeit nicht mehr ausreicht bis er wieder gefüttert wird. Ich vermute auch das es mit noch mehr WRs noch schlimmer wird.

stefan123t, 2023-01-22:
Nach Sonnenuntergang bzw während dem Communication Disable wird zwar per TX gesendet, aber es kommt idR. keine Antwort mehr vom WR zurück. 
D.h. es wird auch kein IRQ ausgelöst und somit dürfte auch die (m.E. wie ein delay wirkende) RX Routine nicht aufgerufen werden.
@lumapu vielleicht hat @beegee3 hier noch eine Idee wie wir das in der Original DTU verbaute "Delay" hier ein- / nachbauen können.
Er muß ja bevor er das nächste Paket sendet zumindest die 300-500ms warten, bis der WR genug Zeit hatte auf das erste Paket zu antworten.

knickohr, 2023-01-23:
Waldi hat wieder gebissen. Zur üblichen Zeit 😢

Klar, ich habe ja heute wieder bichtsufgrpaßt 😡
Meine Einstellung brachte also rein garnix 😩

Gerri1, 2023-01-23:
Zwei Wechselrichter HM-400 mit 30s Intervall

[image]

roku133, 2023-01-23:
@Gerri1 - Das kann ich bestätigen: bei meinem HM-400 und ESP8266 in Standardkonfiguration (Intervall 30 s, Retries 5) läuft v0.5.70 seit mehr als 13 Tagen unterbrechungsfrei.
Offensichtlich sind die vorhandenen Abfragezyklen für ein Solarpanel passend.

Gerri1, 2023-01-23:
Bin mal gespannt, wenn ich meine anderen zwei 400er WR noch mit einbinde, was dann passiert!

beegee3, 2023-01-23:
@stefan123t danke, zieh mich da mit rein, hab' keinen ESP8266 und der Teil ist so viel Dinosaurier Code 🦕
Spaß beiseite, hab' zuerst gedacht, da laufen irgendwelche Buffer über, also rein in den Code und die checken. Überall Debug Infos gesetzt: nix gefunden, alles save! Hier nochmal nachgelesen ... 🤕 na klar, hab an den falschen Stellen gesucht; Watchdog ist doch eine Timer Überwachung; wird hier eigentlich die ganze Zeit gesagt, das Programm scheint in einer Schleife zu hängen.
Nur warum kein Waldi Problem bei mir? Also mal andere Settings ausprobiert und siehe da, bei eingeschaltetem MQTT 'Reset Values at Communication stop' hängt das Teil. Zwar kein reset, aber kein Update der Uptime, bei MQTT mit timeout rausgeflogen ohne reconnect usw. Also den Teil debugged. @lumapu: `zeroValues` in `pubMqtt.h` ist die Todesschleife. Das `continue;` sorgt dafür, dass `fld` auf der Stelle tritt und die `while` Schleife nicht mehr verlassen wird. Ein `fld++;` vor `continue;` schafft Abhilfe. Dann noch aufräumen: es reicht einmal `mSendList.push(RealTimeRunData_Debug);`, also raus damit aus `zeroValues` und rein in `zeroAllInverters` vor `sendIvData();` 😀

Warum die lange Vorrede? 1. um zu sagen, dass der 🦕 Code wohl keine Macken hat, 2. Waldi ein guter Hund ist, nur beim ESP32 wohl eine längere Leine hat und nicht so bissig ist

Hoffe, die reboots sind damit Geschichte!

lumapu, 2023-01-23:
@beegee3 danke für's Gassi gehen! 🐶
Die zerovalues Funktion ist noch recht neu, das reboot Problem besteht seit Anfang an, da gab's einige der heutigen Funktionen noch garnicht.
Ich vermute weiterhin das Problem in der nrf24 Lib. Schau doch bitte auch mal im ältesten offenen issue nach wenn du Lust hast 😊

stefan123t, 2023-01-23:
@beegee3 ja danke fürs Gassi gehen mit unserem 🦖 Code an der besonders langen Leine 🕵️ das hat der sicher gebraucht, jetzt hat er hoffentlich den Waldi gleich mit verspeist.

Ja issue #83 ist einer unserer Langläufer den wir auch schon upstream bei RF24 nachgehakt haben. Nur haben sich weder Lukas noch ich mit deren Antwort ausreichend beschäftigt. Da liegt also vermutlich auch noch der eine oder andere 🐩🦴 begraben...

Ggf. kann ich auch den Timeout Calculation Code aus der Hoymiles DTU Pro raussuchen, damit Du Dir das mal ansehen kannst ?

beegee3, 2023-01-24:
@lumapu zerovalues ist definitiv ein Waldi Problem und ja, ich hab' auch einiges in #83 gelesen. Aber du hast auch einiges im hm Code gemacht und damit evtl. dort beschriebene Probleme eliminiert. Hab mir auch die benutzten nrf24 lib Funktionen angesehen. und glaube dass in `hmRadio` weder Schreiben (`sendPacket`) noch Lesen (`loop`) zeittechnisch problematisch sind. Auch wenn ich nicht alles verstehe:

- wieso setzt man zunächst 'AutoAck' auf `true` und öffnet dann eine (merkwürdige) `DUMMY_RADIO_ID` Pipe, um 'AutoAck' auf `false` zu setzen?
- braucht man `mNrf24.whatHappened(tx_ok, tx_fail, rx_ready)` nicht, wenn man nur sendet und nichts zurückkommt (der oben beschriebene WR offline Fall) ?

beegee3, 2023-01-24:
@stefan123t 

> Ggf. kann ich auch den Timeout Calculation Code aus der Hoymiles DTU Pro raussuchen, damit Du Dir das mal ansehen kannst ?

Kann ja nicht Schaden und jeder andere, der hier mitliest, kann ja auch draufschauen!

lumapu, 2023-01-24:
> Auch wenn ich nicht alles verstehe:

das kann ich dir auch nicht erklären und stammt noch von der ersten Version. Irgendwann kamen Daten und man hat es dann nicht mehr abgefasst.

Bzgl. Zerovalues bin ich bei dir, ich wollte nur sagen, dass es noch was im Code gibt, das Exceptions verursacht. und das muss schon immer drin sein, da wir den Fehler seit Anfang an haben.

knickohr, 2023-01-24:
Ähhh Leute 😲

Ich habe keine Zero Values angeklickert. Doch, hatte schon, aber nachdem ich gemerkt habe das er mir um Mitternacht nicht nullt, habe ich es wieder deaktiviert. Entweder macht er es wirklich nicht, oder er gibt es über MQTT nicht aus. Anyway, andere Geschichte, werde vielleicht dazu noch einen extra Issue öffnen. Ist anscheinend niemand aufgefallen, sooo wichtig war das damals mit dem Nullen 🤪

Zurück zum Thema, wie gesagt, da ist bei mir nichts aktiviert und trotzdem mag mich Waldi nicht. In der Zeit wo Waldi beißt, passiert eigentlich gar nichts, außer das die DTU zu nicht antwortenden Invertern pollt, bis es Nacht wird (communication disabled).

knickohr, 2023-01-24:
OK, ich mach das mal mit meinem ESP32. Mal sehen ob ich da Waldi auch zum beißen bringe.

stefan123t, 2023-01-24:
> Auch wenn ich nicht alles verstehe:

> * wieso setzt man zunächst 'AutoAck' auf true und öffnet dann eine (merkwürdige) DUMMY_RADIO_ID Pipe, um 'AutoAck' auf false zu setzen?

Ich vermute mal man wollte das AutoAck abschalten und damit keine Pakete mit der richtigen DTU ID rausgehen hat man das mit der Dummy ID aktiviert und dann wieder deaktiviert.
Stammt entweder aus "Hubis Code" oder sogar aus dem NRF24 Send/Scan Beispielcode.
Vielleicht kann es auch weg 🤔 ?

> * braucht man mNrf24.whatHappened(tx_ok, tx_fail, rx_ready) nicht, wenn man nur sendet und nichts zurückkommt (der oben beschriebene WR offline Fall) ?

whatHappened speichert bzw exportiert doch das Ergebnis der drei Register/Flags tx_ok, tx_fail, rx_ready, damit wir die dann auswerten können und wissen wo die State Machine des NRF24 gerade rumhängt bzw welchen Zustand sie hat.
Wenn also nichts kommt, dann kommt da auch nichts. WhatHappened() eben ?

knickohr, 2023-01-26:
Ich mache mir echt Sorgen um Waldi 😲 Lebt er noch ?

Der dritte Abend Gassi gehen und er beißt nicht mehr. Braves Hundi 🦮

Meiner Meinung nach kristallisiert es sich immer mehr heraus das es ein 8266-Problem ist. Was ist anders am ESP32, außer das er 2 Core hat, schneller ist und mehr Pins hat ?  Ich schätze mal es sind die 2 Cires und deren Aufgsbenverteilung. Core 0 übernimmt ja die gesamte Hardwareaufgaben, währen im Core 1 die Applikation läuft.

beegee3, 2023-01-26:
@knickohr super, dass Waldi beim Guru bleibt.
> Was ist anders am ESP32?

Die Interruptsteuerung und deren Überwachung ist natürlich anders. Und in dem Umfeld such' ich gerade, da die Probleme vornehmlich beim Nichtantworten des/der WR entstehen. @lumapu hat in #83 ja deutliche Bilder mit der SPI 'Rushhour'.
Die kommt vom NRF24 write Befehl, der die SPI transaction startet und, da ja keine Rückmeldung kommt, 15 mal im Abstand von 750us versucht, die Payload rauszuschicken (so die aktuelle Vorgabe im Programm). Macht 15*(250us + 750us) = 15ms plus Wartezeit, bis der Fehlversuch festgestellt ist, in denen der CE PIN eingeschaltet ist.
Die Empfehlungen von Brendan und TmRh20 als Antwort auf @stefan123t's Nachfrage (https://github.com/nRF24/RF24/issues/877) gehen auch in die Richtung. Hab' diese zum Anlass genommen, verschiedenes auszuprobieren, bin aber noch nicht fertig. Folgender Zwischenstand:

- beim Senden braucht man kein auto-ack, es geht auch mit `mNrf24.setAutoAck(false); mNrf24.write(buf, len, true);`
(true = do not request ACK response). `mNrf24.openWritingPipe(DUMMY_RADIO_ID);` wird dann auch nicht gebraucht!!!
- anstelle von `mNrf24.write(...);` kann man auch `mNrf24.writeFast(buf, len, true); mNrf24.txStandBy();` nutzen. Das stellt sicher, dass CE wieder auf low geht, was bei `mNrf24.write(...);` nicht unbedingt der Fall ist. Das ändert aber nichts an der Zeit, die CE auf high steht. Genauso wenig, wie die Empfehlung `startWrite()` mit TX IRQ zu nutzen. Da hilft m.E. nur da folgende:
-  bei`mNrf24.setRetries(...)` die Anzahl der Wiederholungen reduzieren (teste ich morgen)
- `startWrite()` mit TX IRQ erfordert mehr Programmieraufwand, der m.E. das Problem nicht löst und sich daher z.Zt. nicht lohnt
- `DISABLE_IRQ` und `RESTORE_IRQ` beim `CircularBuffer` sind überflüssig. Früher wurden sie sicher mal gebraucht, aber die Interruptbehandlung erfolgt ja jetzt ausschließlich in `Radio.h`. Dort in der loop kann das 2. `RESTORE_IRQ` entfernt werden.
- dynamic payloads beim Empfang funktionieren grundsätzlich, allerdings sind die Inhalte nicht `checkPaketCrc` tauglich. Erstens ist die im Byte 0 angegebene Länge immer 32 (obwohl `getDynamicPayloadSize()` das richtige liefert und auch nur soviel gelesen wurde) und selbst mit korrigierter Länge ist CRC falsch.
Und da der Datenempfang eher unkritisch ist, kann der erstmal wie bisher bleiben.

knickohr, 2023-01-27:
2 Tage 17 Stunden, und noch immer kein Waldi 😎

ich gebe noch zu Bedenken ob vielleicht auch das Nokia uns einen Streich spielt. Hängt ja auch am SPI und teilt sich diesen mit dem NRF.

Andreas (unser Display-Guru) hat gestern so eine Anmerkung gemacht das offenbar ein spiend() bei der NRF-Rouitine fehlt, bzw. vermißt wird.

vielleicht sollte man da auch mal genauer schauen 🤔

beegee3, 2023-01-27:
spiend() ? wir wollen doch den SPI nicht abschalten. Vielleicht war SPI.endTransaction() gemeint. In der RF24 lib wird jeder SPI Transfer mit SPI.beginTransaction gestartet und mit SPI.endTransaction beendet. Das ist unproblematisch. Danach geht CE auf high und erst wieder auf low, wenn entweder TX_DS (TX data send = Daten vom WR empfangen) oder MAX_RT (maximum retries = Maximalanzahl der Sendeversuche erreicht) signalisiert wird. Da wird Zeit verbraucht. Insbesondere wenn der WR offline ist, da dann direkt wieder angefragt wird ("nothing received: Request Complete Retransmit"), was wohl die SPI 'Rushhour' erklärt. @stefan123t wollte ja noch den Timeout Calculation Code aus der Hoymiles DTU Pro raussuchen. Könnte weiterhelfen.
Keine Ahnung, ob zusätzliche Geräte am SPI Waldi eher knurren lassen, aber das Problem gab es ja auch schon bevor die Displays ins Spiel kamen.

beegee3, 2023-01-27:
hier eine Wasserstandsmeldung:

- `DISABLE_IRQ` und `RESTORE_IRQ` beim `CircularBuffer` **sind** überflüssig (s.o.), man könnte CB durch eine std::list ersetzen
- `Radio.h`: Scheint egal zu sein, ob in der `loop` das 1. oder 2. `RESTORE_IRQ` entfernt wird
- Senden geht ohne auto-ack, aber man bekommt sehr viel mehr "nothing received"
- Empfangen mit auto-ack geht gar nicht, d.h. man **muß** auto-ack zwischen Senden und Empfangen immer Umschalten
- die Reduktion der Anzahl der Wiederholungen bei `Nrf24.setRetries(...)` führt auch zu sehr viel mehr "nothing received"
- die `DUMMY_RADIO_ID` Pipe wird definitiv nicht benötigt

Ist irgendwie ernüchternd, im Fazit habe ich die Zeile `mNrf24.setRetries(3, 15); // 3*250us and 15 loops -> 11.25ms` aus `sendPacket` ins `setup` verschoben und `sendPacket` verkürzt. Dabei `write` durch `writeFast` und `txStandBy` ersetzt, aber das ist (zumindest beim ESP32) eher unwesentlich (s.o.):
```
        void sendPacket(uint64_t invId, uint8_t buf[], uint8_t len, bool isRetransmit, bool clear=false) {
            ...
            mNrf24.setChannel(mRfChLst[mTxChIdx]);
            mTxCh = getTxNxtChannel(); // switch channel for next packet
            mNrf24.openWritingPipe(invId); // TODO: deprecated
            mNrf24.setCRCLength(RF24_CRC_16);
            mNrf24.enableDynamicPayloads();
            mNrf24.setAutoAck(true);
            mNrf24.writeFast(buf, len, false); // false = request ACK response
            mNrf24.txStandBy();
            mRxChIdx = 0;
            mNrf24.setChannel(mRfChLst[mRxChIdx]);
            mNrf24.setAutoAck(false);
            mNrf24.disableDynamicPayloads();
            mNrf24.setCRCLength(RF24_CRC_DISABLED);
            mNrf24.startListening();
            ...
        }
```
In Bezug auf Waldi hilft das nicht weiter.
Aber vielleicht hilft folgender Vorschlag: für den Fall, dass gar keine Rückmeldung vom WR kommt, nicht sofort eine neue Anfrage senden, sondern das nächste Intervall abwarten. D.h. in `payload.h` den 'Request Complete Retransmit' ersetzen durch
```
            if(false == mPayload[iv->id].gotFragment) {
              DPRINTLN(DBG_WARN, F("(#") + String(iv->id) + F(") nothing received"));
              mPayload[iv->id].retransmits = mMaxRetrans;
            else {
              ...
```
Das würde den SPI deutlich entspannen. @lumapu Wäre schön, wenn du davon Nachtaufnahmen machen könntest (z.B. mit Inverter Intervall 5s) 😃

lumapu, 2023-01-27:
@beegee3 danke für deine Untersuchungen. Circular buffer durch eine `std::list` oder `std::queue` zu ersetzen klingt interessant. An der Stelle gehört auf jeden Fall was gemacht, da dies noch Urgesteine sind.

Die Aufnahme hab ich gerne machen. Ich hätte mir aus der Doku mehr versprochen aus `startWrite` als deine Analysen jetzt ergeben haben.

beegee3, 2023-01-27:
@lumapu ok, ich Versuch mal `startWrite` umzusetzen. Vielleicht geht's doch einfacher als gedacht. Hab' Bedenken, da zwischen Senden und Empfangen immer auto-ack und dynamic payloads aktiviert bzw. deaktiviert werden. Z.Zt. wird das ja in `sendPacket` gemacht. Mal sehen, ob das so bleiben kann, oder per Interrupt gesteuert werden muß.

knickohr, 2023-01-28:
Pegelstand Blaustein :  Normal 😉

Waldi ist brav, keinen Beißer, kein Knurren. ESP32-Wachhunde sind gute Hunde 😅

Uptime fast 4 Tage.

Waldo56, 2023-01-28:
Das muss jetzt einmal geschrieben werden. Ihr seid der Wahnsinn betreffend Eurem Einsatz bei der Ursachenfindung und Verbesserungen.
Eines ist mir noch aufgefallen bei einem gesteuerten Web Reboot der DTU. Es gibt öfter mal als Restart Cause eine Exception. Wieso? Werden hier nicht vor dem eigentlichen Restart die Interrupts gesperrt?

knickohr, 2023-01-28:
Hat doch Lukas schon mal ein Statement abgegeben :

https://github.com/lumapu/ahoy/issues/564#issuecomment-1378370317

lumapu, 2023-01-28:
ja ich denke die Exception kommt von den Event `Network lost` und versucht dann noch Prozesse oder was auch immer zu starten die nicht mehr existieren. Ist nicht ganz sauber aber halt auch nicht Prio 1

beegee3, 2023-01-28:
@lumapu wie vermutet ist `startWrite` problematisch. Die auto-ack und dynamic payloads Wechsel gehen noch, aber permanentes IRQ an- und ausschalten und das Channel Hopping machen Ärger. Zumindest dachte ich das, denn ich bekam immer wieder Reboots. Nach etlichem Umprogrammieren hab' ich mir erst jetzt mal genauer angesehen, was vor dem Reboot passiert. Und siehe da, die letzte Meldung ist "Alarm #2 'DTU command failed'...". Das kommt doch von `parseAlarmLog` und wird aus `payload.h` `process` aufgerufen. Darauf folgt der Aufruf `(mCbAlarm)(code, start, end);`. Da ich für die Tests Mqtt abgeklemmt habe, wird `mCbAlarm` nicht gesetzt und der Aufruf **muss** unweigerlich zum Reset führen!
**`payload.h` : bitte im `setup` `mCbAlarm = NULL;` eintragen und den Aufruf nur für `if (NULL != mCbAlarm)` machen
(bei `process` und `notify`)!**

beegee3, 2023-01-28:
@lumapu die Sonne ist weg -> keine RX Tests möglich, daher noch 2 Bemerkungen:

1. wenn oben genannte Exception von `Network lost` kommt, dann hilft in `app.h`:
```
       void tickReboot(void) {
            DPRINTLN(DBG_INFO, F("Rebooting..."));
            onWifi(false);
            ah::Scheduler::resetTicker();
            WiFi.disconnect();
            ESP.restart();
        }
```
kann man für OTA Update noch ausbauen (alle laufenden Prozesse vor dem Einspielen des Updates stoppen):
in `appInterface.h` die Zeile `virtual void setOnUpdate() = 0;` und in`app.h` das zugehörende
```
        void setOnUpdate() {
            onWifi(false);
        }
```
sowie den Aufruf in `web.h`
```
       void onUpdate(AsyncWebServerRequest *request) {
            ...
            mApp->setOnUpdate();
        }
```

2. (falls mal sonst nichts zu tun ist 😬) in `main.cpp` steht noch ein **TODO: move to HmRadio**
das lässt sich so erledigen:
in `hmRadio` letzte Zeile `volatile bool mIrqRcvd;` löschen, dafür hinter den Makros, aber **vor** `class HmRadio` die Zeile
`static volatile bool mIrqRcvd;` schreiben. Dann noch in `setup` hinter `pinMode(irq, INPUT_PULLUP);` die Zeile `attachInterrupt(digitalPinToInterrupt(irq), []()IRAM_ATTR{ mIrqRcvd = true; }, FALLING);`
Das war's. Nur noch in `main` `attachInterrupt(...)` und die Funktion `IRAM_ATTR void handleIntr(void)` löschen. Auch alle anderen `handleIntr` Funktionen in `app.h`, `app.cpp` und `hmRadio`.

knickohr, 2023-01-30:
Ich behaupte jetzt einfach mal, es ist ein 8266-Problem !

Mit einem ESP32 ist folgender Stand aktuell :
[image]
Keine Waldi‘s, keine Reboots. Die DTU läuft rund und geschmeidig. Auch keine hängenden Webseiten wie hin und wieder berichtet wird. MQTT stört den Ablauf ebenfalls nicht.

Das ist jetzt die längste Uptime die ich jeweils habe, bin schon 2. Versionen hinten an und noch auf der 76 😲

lumapu, 2023-01-30:
echt schade, dass der 8266 so zickig ist. Bei mir ist zur Zeit oft der halbe Webserver tot, die API liefert nur noch `null` aber ein `/reboot` geht noch 🤔

[7 more replies]

closed 2023-03-27 by lumapu