lumapu/ahoy #565: [ESP8266/ESP32/RaspberryPi] Problem Description / Homeassistant autodisovery failure

silviokennecke, 2023-01-08:
### Platform

ESP8266

### Model name

ESP8266 D1 Mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: <serial-no of my inverter, not pasted here>
    - Name: HM-300
    - Max Module Power (Wp): 100
    - Module Name: DG_Fenster
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP:  <local-ip, not pasted here>
    - Port: 1883
    - Username (optional): <mqtt user, not pasted here>
    - Password (optional): <mqtt user, not pasted here>
    - Topic: ahoydtu
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [ ]
    - Serial Debug: [ ]
    - Interval [s]: 5

- Reboot device after successful save: [ ]
- SAVE

### Error description

I reciently tried to connect AhoyDTU with homebridge using MQTT. The connection itself works. When submitting the autoconfig message, the manufacturer payload contains `null` instead of `Hoymiles`. Flashing the ESP8266 again didn't change a thing. The autodiscovery worked correctly when downgrading to 0.5.41.  The only relevant change I found was in [573bd3](https://github.com/lumapu/ahoy/commit/573bd31f0a4b5f9bfa032ac6eb3503ae03bc3c0b#diff-79d7d0c3f8b5696521930abf01dd8bfad0836c00a157c6c0ab475894d62ce8fcR133), although I don't see why this change caused the described error.

1) Go to http://ahoy-dtu/setup
2) open MQTT
3) Click the "send" button for "Discovery Config (homeassistant)

Expected MQTT Payload (sample for `homeassistant/sensor/HM-300_11218382/ch0_P_AC/config`):
```json
{
  "name": "HM-300 P_AC",
  "stat_t": "ahoydtu/HM-300/ch0/P_AC",
  "unit_of_meas": "W",
  "uniq_id": ""************_ch0_P_AC",
  "dev": {
    "name": "HM-300",
    "ids": "************",
    "cu": "http://************",
    "mf": "Hoymiles"
  },
  "exp_aft": 65,
  "dev_cla": "power"
}
```

Received MQTT Payload:
```json
{
  "name": "HM-300 P_AC",
  "stat_t": "ahoydtu/HM-300/ch0/P_AC",
  "unit_of_meas": "W",
  "uniq_id": ""************_ch0_P_AC",
  "dev": {
    "name": "HM-300",
    "ids": "************",
    "cu": "http://************",
    "mf": null
  },
  "exp_aft": 65,
  "dev_cla": "power"
}
```

Payload Diff:
```
     "cu": "http://************",
-    "mf": null
+    "mf": "Hoymiles"
   },
```

Also, I noticed that the property `stat_cla` is also missing, whereas it was given in 0.5.66.

stefan123t, 2023-01-08:
@lumapu could it be related to the maximum JSON Document size 
or went something wrong while moving the manufacturer into Flash Mem ?
```c
deviceDoc(128);
...
deviceDoc[F("mf")] = F("Hoymiles");
```

lumapu, 2023-01-09:
don't know. Thank you for reporting, I will have a look into that.

antibill51, 2023-01-13:
Hi, 
I encounter exactly the same problem.

I noticed that by increasing the size of the json document  `DynamicJsonDocument deviceDoc(128);` , there was better, but in this case, my 1st inverter works, while the second finds itself amputated from certain values.

Thank you for your excellent work.

`

       void sendDiscoveryConfig(void) {
            DPRINTLN(DBG_VERBOSE, F("sendMqttDiscoveryConfig"));
            char stateTopic[64], discoveryTopic[64], buffer[512], name[32], uniq_id[32];
            for (uint8_t id = 0; id < mSys->getNumInverters(); id++) {
                Inverter<> *iv = mSys->getInverterByPos(id);
                if (NULL != iv) {
                    record_t<> *rec = iv->getRecordStruct(RealTimeRunData_Debug);
                    DynamicJsonDocument deviceDoc(128);
                    deviceDoc[F("name")] = iv->config->name;
                    deviceDoc[F("ids")] = String(iv->config->serial.u64, HEX);
                    deviceDoc[F("cu")] = F("http://") + String(WiFi.localIP().toString());
                    deviceDoc[F("mf")] = F("Hoymiles");
                    deviceDoc[F("mdl")] = iv->config->name;
                    JsonObject deviceObj = deviceDoc.as<JsonObject>();
                    DynamicJsonDocument doc(384);

                    for (uint8_t i = 0; i < rec->length; i++) {
                        if (rec->assign[i].ch == CH0) {
                            snprintf(name, 32, "%s %s", iv->config->name, iv->getFieldName(i, rec));
                        } else {
                            snprintf(name, 32, "%s CH%d %s", iv->config->name, rec->assign[i].ch, iv->getFieldName(i, rec));
                        }
                        snprintf(stateTopic, 64, "/ch%d/%s", rec->assign[i].ch, iv->getFieldName(i, rec));
                        snprintf(discoveryTopic, 64, "%s/sensor/%s/ch%d_%s/config", MQTT_DISCOVERY_PREFIX, iv->config->name, rec->assign[i].ch, iv->getFieldName(i, rec));
                        snprintf(uniq_id, 32, "ch%d_%s", rec->assign[i].ch, iv->getFieldName(i, rec));
                        const char *devCls = getFieldDeviceClass(rec->assign[i].fieldId);
                        const char *stateCls = getFieldStateClass(rec->assign[i].fieldId);

                        doc[F("name")] = name;
                        doc[F("stat_t")] = String(mCfgMqtt->topic) + "/" + String(iv->config->name) + String(stateTopic);
                        doc[F("unit_of_meas")] = iv->getUnit(i, rec);
                        doc[F("uniq_id")] = String(iv->config->serial.u64, HEX) + "_" + uniq_id;
                        doc[F("dev")] = deviceObj;
                        doc[F("exp_aft")] = MQTT_INTERVAL + 5;  // add 5 sec if connection is bad or ESP too slow @TODO: stimmt das wirklich als expire!?
                        if (devCls != NULL)
                            doc[F("dev_cla")] = devCls;
                        if (stateCls != NULL)
                            doc[F("stat_cla")] = stateCls;

                        serializeJson(doc, buffer);
                        publish(discoveryTopic, buffer, true, false);
                        doc.clear();
                    }

                    yield();
                }
            }
        }
`

LustigePerson, 2023-01-14:
Is there currently a workaround for this issue?

stefan123t, 2023-01-14:
@LustigePerson as @antibill51 confirmed it can be worked around by doubling / increasing the document size:
```c
                DynamicJsonDocument doc(384);
```
@lumapu the final fix would somehow require us to allocate the document size dynamically.

lumapu, 2023-01-14:
@stefan123t or even big enough 😊

FrankHuxol, 2023-01-16:
I also ran into this issue and my workaround is, that I set up a node red flow, which listens to the autodiscovery topics of the ahoy dtu. If a message does not yet contain a "mf", it resends the mqtt message with a fixed "mf" value of "Hoymiles" (defined in the flow). This works for me currently.

```
[{"id":"8084ab981afdab69","type":"tab","label":"Ahoy DTU Autodiscovery Fix","disabled":false,"info":"Fix for https://github.com/lumapu/ahoy/issues/565","env":[]},{"id":"c1822fb0d1f0d28b","type":"mqtt in","z":"8084ab981afdab69","name":"","topic":"homeassistant/sensor/Vordach/#","qos":"2","datatype":"auto-detect","nl":false,"rap":true,"rh":0,"inputs":0,"x":130,"y":220,"wires":[["bb1cc130c847c5c8"]]},{"id":"690644d3b318a553","type":"debug","z":"8084ab981afdab69","name":"Correct","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":540,"y":280,"wires":[]},{"id":"774a4e67e4fd14b0","type":"change","z":"8084ab981afdab69","name":"","rules":[{"t":"set","p":"payload.dev.mf","pt":"msg","to":"Hoymiles","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":590,"y":140,"wires":[["69b4387aa1e72338","f6c4f83c31936afa"]]},{"id":"bb1cc130c847c5c8","type":"switch","z":"8084ab981afdab69","name":"","property":"payload.dev.mf","propertyType":"msg","rules":[{"t":"null"},{"t":"else"}],"checkall":"true","repair":false,"outputs":2,"x":370,"y":220,"wires":[["774a4e67e4fd14b0"],["690644d3b318a553"]]},{"id":"69b4387aa1e72338","type":"debug","z":"8084ab981afdab69","name":"Broken","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":880,"y":140,"wires":[]},{"id":"f6c4f83c31936afa","type":"mqtt out","z":"8084ab981afdab69","name":"","topic":"","qos":"","retain":"","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","x":890,"y":440,"wires":[]}]
```

antibill51, 2023-01-17:
@lumapu 
it works fine for inverter #0 with version 0.5.72, but I have no discovery message for inverter #1.

antibill51, 2023-01-20:
Am I the only one having this problem?

lumapu, 2023-01-20:
@antibill51 same here - I will check that further. Does not work for me either

antibill51, 2023-01-26:
@lumapu 

With 0.5.77 version, autodiscovery don't work .
One part of the message is repeated, the other is missing.

Example message : 
`
{"name":"Test1 CH1 U_DC","stat_t":"inverter2/Test1/ch1/U_DC","unit_of_meas":"V","uniq_id":"xxxxxxxxxxxx_ch1_U_DC","dev":{"name":"Test1 CH1 U_DC","stat_t":"inverter2/Test1/ch1/U_DC","unit_of_meas":"V","uniq_id":"xxxxxxxxxxxx_ch1_U_DC","dev":{"name":"Test1 CH1 U_DC","stat_t":"inverter2/Test1/ch1/U_DC","unit_of_meas":"V","uniq_id":"xxxxxxxxxxxx_ch1_U_DC","dev":{"name":"Test1 CH1 U_DC","stat_t":"inverter2/Test1/ch1/U_DC","unit_of_meas":"V","uniq_id":"xxxxxxxxxxxx_ch1_U_DC","dev":{"name":"Test1 CH1 U_DC","stat_t":"inverter2/Test1/ch1/U_DC","unit_of_meas":"V","uniq_id":"xxxxxxxxxxxx_ch1_U_DC","dev":{}}}}}}
`

schliepi, 2023-01-27:
Hi all!

Same problem here.
Code reviewed.
Looks like the removal of "DynamicJsonDocument deviceDoc(128);"  and working with only one "doc(512) led to this bug.

antibill51, 2023-01-27:
@lumapu 

This is working : 

```
        void sendDiscoveryConfig(void) {
            DPRINTLN(DBG_VERBOSE, F("sendMqttDiscoveryConfig"));

            char topic[64], name[32], uniq_id[32];
            DynamicJsonDocument doc(512);
            for (uint8_t id = 0; id < mSys->getNumInverters(); id++) {
                Inverter<> *iv = mSys->getInverterByPos(id);
                if (NULL == iv)
                    continue;

                record_t<> *rec = iv->getRecordStruct(RealTimeRunData_Debug);
                doc.clear();
                
                doc[F("name")] = iv->config->name;
                doc[F("ids")] = String(iv->config->serial.u64, HEX);
                doc[F("cu")] = F("http://") + String(WiFi.localIP().toString());
                doc[F("mf")] = F("Hoymiles");
                doc[F("mdl")] = iv->config->name;
                JsonObject deviceObj = doc.as<JsonObject>();

                for (uint8_t i = 0; i < rec->length; i++) {
                    if (rec->assign[i].ch == CH0)
                        snprintf(name, 32, "%s %s", iv->config->name, iv->getFieldName(i, rec));
                    else
                        snprintf(name, 32, "%s CH%d %s", iv->config->name, rec->assign[i].ch, iv->getFieldName(i, rec));
                    snprintf(topic, 64, "/ch%d/%s", rec->assign[i].ch, iv->getFieldName(i, rec));
                    snprintf(uniq_id, 32, "ch%d_%s", rec->assign[i].ch, iv->getFieldName(i, rec));

                    const char *devCls = getFieldDeviceClass(rec->assign[i].fieldId);
                    const char *stateCls = getFieldStateClass(rec->assign[i].fieldId);
                    DynamicJsonDocument doc2(512);

                    doc2.clear();
                    doc2[F("name")] = name;
                    doc2[F("stat_t")] = String(mCfgMqtt->topic) + "/" + String(iv->config->name) + String(topic);
                    doc2[F("unit_of_meas")] = iv->getUnit(i, rec);
                    doc2[F("uniq_id")] = String(iv->config->serial.u64, HEX) + "_" + uniq_id;
                    doc2[F("dev")] = deviceObj;
                    doc2[F("exp_aft")] = MQTT_INTERVAL + 5;  // add 5 sec if connection is bad or ESP too slow @TODO: stimmt das wirklich als expire!?
                    if (devCls != NULL)
                        doc2[F("dev_cla")] = String(devCls);
                    if (stateCls != NULL)
                        doc2[F("stat_cla")] = String(stateCls);

                    snprintf(topic, 64, "%s/sensor/%s/ch%d_%s/config", MQTT_DISCOVERY_PREFIX, iv->config->name, rec->assign[i].ch, iv->getFieldName(i, rec));
                    size_t size = measureJson(doc2) + 1;
                    char *buf = new char[size];
                    memset(buf, 0, size);
                    serializeJson(doc2, buf, size);
                    publish(topic, buf, true, false);
                    delete[] buf;
                }

                yield();
            }
        }

```

andreas-bulling, 2023-02-07:
same problem here