lumapu/ahoy #567: How to build dev branch with PlatformIO

SpaceTeddy, 2023-01-09:
## Hardware
  * [X ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: __Wmos D1 mini ____
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

## Software
* [ X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ X] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [X ] Mac OS

### Debugging:
* [X ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I Would like to clone the dev branch into Visual Code + PatformIO and build the project. Simple cloneing does not provide some informations like platformio.ini ect. 
Is there a documentation how to setup the build environment?

thx,
Chris

lumapu, 2023-01-09:
just point to the subdirectory `src`. There PIO will find the `.ini` and you're ready to compile.
The dev version is build automatically by GitHub and downloadable from https://ahoydtu.de/web_install/ if you don't want to compile by you own

SpaceTeddy, 2023-01-10:
found it in the src directory.

thx

closed 2023-01-10 by SpaceTeddy