lumapu/ahoy #570: Ahoy 70 Yield Total Correction

dtuuser, 2023-01-09:
Weiß nicht, ob das hier die richtige Stelle ist.
Wäre eine Yield Total Correction pro Solarmodul mit Addition zum Yield Total Correction pro Wechselrichter sinnvoll?
Dann wären alle Yield Total Werte "korrekt".

lumapu, 2023-01-09:
ja könnte man auch machen, braucht aber deutlich mehr Speicherplatz und damit auch RAM zur Laufzeit.
Ich war mir anfangs unsicher das Feature überhaupt einzubauen, aber pro Modul wäre natürlich sauberer.
Warten wir noch ab andere Stimmen ab und entscheiden dann wohin es geht 😊

dtuuser, 2023-01-09:
ich frage die Werte per eigener Anwendung über die ..api/live Seite ab und ziehe die Werte ab. Wenn Du es bei dir änderst, kann ich das aus meiner Anwendung raus nehmen 😊

knickohr, 2023-01-10:
Also YieldTotal sehe ich ein, wenn jemand ein (kleines) Display direkt an der DTU hat. alles andere kann doch prima über die Visualisierungstools gelöst werden.

Screambear, 2023-01-10:
Sorry fürs missbrauchen.
Gibt es die Ahoy 0.5.70 als bin-File als download zum einfachen flashen?

stefan123t, 2023-01-12:
Ja logisch, man sollte natürlich die YieldTotal und/oder YieldTotal je Solarmodul zurücksetzen können.
Dann könnte man z.B. auch einen Akku anstelle eines Solarmoduls anschließen 
und hier den KiloWattStundenzähler mal auf Null stellen wenn man etwas umbaut / konfiguriert.
@dtuuser wie soll man das aber machen, wenn nur unvollständige Ein/Angaben vorliegen ?

Ich habe bei mir z.B. aktuell:

Inverter0 YieldTotal: `502.220` kWh Correction: -`502.220` kWh `^`/`v` `[RESET]`
Module1 YieldTotal: `254.885` kWh Correction1: -`254.885` kWh `^`/`v` `[RESET]`
Module2 YieldTotal: `250.332` kWh Correction2: -`250.332` kWh `^`/`v` `[RESET]`

* alle YieldTotal Werte auf Null setzen (RESET Button)
  -> übernimmt die aktuellen YieldTotal Werte in die Correction Input Felder.
       geht auch individuell für ein einzelnes Modul dann wird nur dessen YieldTotal übernommen 
       und ggf. der Inverter YieldTotal ebenfalls um den selben Betrag angepaßt.
* maximal YieldTotal eines Moduls ins CorrectionX Input Feld eingeben / anpassen mit Pfeiltasten ?
  -> Addition dieses/aller Modul Werte zum YieldTotal Correction Wert des WR ?
       Der Betrag der Änderung zum alten CorrectionX Wert wird auch im Correction Feld des Inverters angepaßt.
* maximal YieldTotal des Inverters in das Correction Input Feld des WR eingeben / anpassen mit Pfeiltasten ?
  -> was passiert mit den Correction Werten der Module ? 
       Bleiben die auf 0, dann stimmt die Summe ja nicht mehr 
       oder werden die Correction Werte der Module um den n-ten Teil des Betrags jeweils analog dazu angepaßt ?

dtuuser, 2023-02-12:
Danke. 86 ist eingespielt.....jetzt muss nur noch der Tag beginnen

MiniOh, 2023-02-12:
Morgen, 

vielen Dank fürs umbauen.
Beim ersten Versuch scheint aber noch nicht alles zu passen.
Ich habe bei einem HM-1500 bei Modul 1 und 2 jeweils 333 kWh addiert.

Hier werden aber bei Modul 4 merkwürdige Werte ausgegeben.

[image]

[image]

dtuuser, 2023-02-12:
Bei mir auch. Panel 1 addiert die Korrektur und bei Panel 2 bleibt der Wert gleich

MiniOh, 2023-02-12:
Bei mir bleibt der obige Fehler, dass Modul 4 extrem hohe Werte anzeigt, auch dann bestehen, wenn ich die Korrektur wieder zurück auf 0 stelle.
Auch nach Reboot der DTU.

lumapu, 2023-02-12:
gefixt in `0.5.87`

dtuuser, 2023-02-12:
Klappt. 
Bei gewünschtem Abzug muss man nur einen Minuswert eingeben.
Topp. Vielen Dank

MiniOh, 2023-02-12:
Ja, hier nun auch alles ok. +XXX wird ordenltich pro Modul übernommen.

closed 2023-03-27 by lumapu