lumapu/ahoy #575: 2W at night

dsteinkopf, 2023-01-11:
### Platform

ESP8266

### Model name

Wemos D1 mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

"Normal" - as described in the readme + default config.

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

Does this matter for the described problem? I'll add this here if needed.

### Error description

Last night I had the issue that my inverter seemed to constantly produce 2.2W. I assume, this was because the night came "too quickly" and the inverter stopped responding before sending "0W" at least once. This also was an issue here: henne49/dbus-opendtu#21. But we'd consider this also a bug in Ahoy:

In `http://ahoy/api/live` the `ts_last_success` is correctly showing that the values are old. One could discuss if is acceptable that the old values (2.2W in this case) are sent out through the hole night. - In my eyes this is ok as we have the timestamp.

The problem I see is that the **latest value which is sent via MQTT is also 2.2W**. So e.g. HomeAssistant constantly shows this value - which is wrong.

lumapu, 2023-01-11:
please check the latest development firmware which includes options to zero values by different triggers. They can be set inside the settings window inside the MQTT region.

Dev build download:
https://ahoydtu.de/web_install then click the link for development version

stefan123t, 2023-01-12:
@TB-79 I consider your problem with mystrom a measurement issue of the mystrom device you are plugging between the HM-Inverter and your wall-wart. This is measuring the AC side of the Inverter and therefor it may account some power detected by small power factor changes in the grid.
Your issue is on the grid side of things and in how your mystrom Wifi switch works. We cannot do anything about that IMHO.  

The problem that we are focusing on here and that @dsteinkopf had raised is the messages via MQTT are sent with a retained flag and therefor could not be zeroed during disabled night communication (the inverter will not be able to send any new data after sunset) and/or the yield daily can now also be reset at midnight (in case you are running a battery to power your inverter).
These issues are both fixed in latest development build.

dsteinkopf, 2023-01-13:
I've now updated to dev version `d8e255d :: 0.5.70` and will see if this happens again.

BTW. I was not able to do an OTA. For debugging I connected a serial cable and saw it was an OOM exception. So I uploaded the firmware manually via serial cable. All settings were lost. Is that expected?

Also I was not able to upload the previously downloaded `ahoy_setup.json`. Console show the exception below. Is that expected?

```
[09:30:37]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[09:30:37]
[09:30:37]Exception (28):
[09:30:37]epc1=0x4000df2f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000033 depc=0x00000000
[09:30:37]
[09:30:37]>>>stack>>>
[09:30:37]
[09:30:37]ctx: sys
[09:30:37]sp: 3fffea00 end: 3fffffb0 offset: 0190
[09:30:37]3fffeb90:  40249c05 3fff3014 4026cf40 401005c8  
[09:30:38]3fffeba0:  40247909 00010083 0000010f 40260d50  
[09:30:38]3fffebb0:  00000033 00000010 402481fd 3ffee3c0  
[09:30:38]3fffebc0:  3fff3014 00000000 3ffe9b19 3fff3014  
[09:30:38]3fffebd0:  3ffe9b19 3fffc200 00000022 000000ff  
[09:30:38]3fffebe0:  3fff6025 3fff2d6c 00000007 3ffeeb04  
[09:30:38]3fffebf0:  3ffe9b19 3ffe9b19 00000033 4024a0ab  
[09:30:38]3fffec00:  3ffeeab8 40249c7c 4024c2f0 3ffeeab8  
[09:30:38]3fffec10:  402497fc 3fffed98 80808080 3fff2152  
[09:30:38]3fffec20:  00000001 00000001 3fff12a2 00000030  
[09:30:38]3fffec30:  40249cb7 3ffeebb0 3ffeebb0 40100280  
[09:30:38]3fffec40:  4024bc30 00000000 4025f50f 00000000  
[09:30:38]3fffec50:  3ffee444 4024bbf4 3ffee3c0 00000012  
[09:30:38]3fffec60:  40249c6c 40249c40 3ffee444 00000012  
[09:30:38]3fffec70:  40249c0b 3ffee3c0 00000012 40265094  
[09:30:38]3fffec80:  4024a1bc 3ffee3c0 00000000 4024c2f0  
[09:30:38]3fffec90:  40249be9 3ffee53c 00000002 00000000  
[09:30:38]3fffeca0:  4024c0a5 0000010f 00000000 0000003e  
[09:30:38]3fffecb0:  3fff12a2 0000010f 3ffee5dc 4024bef9  
[09:30:38]3fffecc0:  4024bf04 3fff2152 00000000 00000001  
[09:30:38]3fffecd0:  402656c0 3fff2152 00000000 00000000  
[09:30:38]3fffece0:  3fff2d6c 4021963e 3fff12c2 40219567  
[09:30:38]3fffecf0:  77706970 006e616c 3ffffe00 40241395  
[09:30:38]3fffed00:  402643a1 00000001 00000001 402392a0  
[09:30:38]3fffed10:  49524537 36485453 4c49584e 34345073  
[09:30:38]3fffed20:  72674c6e 76587559 5a705972 6b345536  
[09:30:38]3fffed30:  4737567a 4453556d 6b694175 6e653336  
[09:30:38]3fffed40:  67516a58 52526f67 376a5873 3f006957  
[09:30:38]3fffed50:  88da7400 00609abe 3fff2154 00000030  
[09:30:38]3fffed60:  77706970 006e616c 3ffed7f0 00000000  
[09:30:38]3fffed70:  40104f25 3ffee2a0 3ffed9d0 00040000  
[09:30:38]3fffed80:  49524537 36485453 4c49584e 34345073  
[09:30:38]3fffed90:  72674c6e 76587559 5a705972 6b345536  
[09:30:38]3fffeda0:  4737567a 4453556d 6b694175 6e653336  
[09:30:38]3fffedb0:  67516a58 52526f67 376a5873 3f006957  
[09:30:38]3fffedc0:  00000000 2c9f0300 40000581 00000002  
[09:30:38]3fffedd0:  00000000 3fff2150 00000001 00000004  
[09:30:38]3fffede0:  40218ae1 00000030 00000008 3fff2154  
[09:30:38]3fffedf0:  00000001 3fff2150 3ffefaf0 40219694  
[09:30:38]3fffee00:  3fffeef2 0000009a 00000060 40217df2  
[09:30:38]3fffee10:  3fffeee0 3fff7b4f 00000001 3fff7b44  
[09:30:38]3fffee20:  3fff5c44 00000001 3fff2154 00000030  
[09:30:38]3fffee30:  00000000 3fff2bec 3fff7f94 3fff2be8  
[09:30:38]3fffee40:  3fff7f94 00000020 3fff65a4 40100c6a  
[09:30:38]3fffee50:  3fff2bf4 00000000 00000000 401005ec  
[09:30:38]3fffee60:  3fff2d6c ffffffcc 3fff7f94 402413bc  
[09:30:38]3fffee70:  00000001 3fff5c44 3ffefaf0 40217eb4  
[09:30:38]3fffee80:  3fff7f94 3fff2c0c 3ffefaf0 4021805b  
[09:30:38]3fffee90:  00000000 4bc6a7f0 00016065 401005c8  
[09:30:38]3fffeea0:  0000012c 00000001 3fffeee0 40227ab2  
[09:30:38]3fffeeb0:  3fffeee0 3fff7b48 3fff7b44 40218098  
[09:30:38]3fffeec0:  3fffdc10 3fff2048 00000001 4022c8d4  
[09:30:38]3fffeed0:  00000001 3fff5c44 3fff7b44 40218ae4  
[09:30:38]3fffeee0:  77706970 006e616c 07000000 be88da74  
[... 247 lines cut ...]
[09:30:39]3ffffe60:  00000000 01ad8adc 3ffef764 3ffef884  
[09:30:39]3ffffe70:  3ffefabe 3ffefa64 40228974 3fffefa0  
[09:30:39]3ffffe80:  40224899 00000001 3ffffea0 4022489c  
[09:30:39]3ffffe90:  00000000 3ffef884 3ffef764 40207aa2  
[09:30:39]3ffffea0:  00000000 000c000f 00000000 3ffefa64  
[09:30:39]3ffffeb0:  3ffefabb 3ffef830 3ffef764 4022bee8  
[09:30:39]3ffffec0:  3fffdad0 00000000 3ffef764 402072bf  
[09:30:39]3ffffed0:  01010101 01010101 00010101 00000000  
[09:30:39]3ffffee0:  00000000 00000000 00000000 40220000  
[09:30:39]3ffffef0:  3ffefab4 3ffffeda 3ffef760 3fff2364  
[09:30:39]3fffff00:  3fff1160 3fff04a0 3ffef760 3fff2364  
[09:30:39]3fffff10:  3fffdad0 00000000 3ffef760 4020eb5e  
[09:30:39]3fffff20:  00000005 3fff1860 3fff1238 4022c260  
[09:30:39]3fffff30:  3ffef764 3fff1860 3fff1238 40216069  
[09:30:39]3fffff40:  3fff78f4 00000000 40205850 4022bfcc  
[09:30:39]3fffff50:  3fff6d94 00000000 402058f8 3fff2364  
[09:30:39]3fffff60:  007a1200 6d04eeb3 00000000 402296ed  
[09:30:39]3fffff70:  00000000 00000000 3fff3908 0000001b  
[09:30:39]3fffff80:  00000000 00000000 00000001 3fff2364  
[09:30:39]3fffff90:  3fffdad0 00000000 3fff2350 402176f8  
[09:30:39]3fffffa0:  feefeffe feefeffe 3fff2350 40228aa0  
[09:30:39]<<<stack<<<
[09:30:39]
[09:30:39]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

dsteinkopf, 2023-01-17:
Similar (same?) problem during the day after a reboot of Ahoy: 
**Values are sent out as 0**, although Ahoy seems to know that this is wrong as **ts_last_success is also 0**.
So in this case, the values should just not be sent out. Should I open a new issue for this?

stefan123t, 2023-01-18:
Sending out values with ts_last_success being 0 might qualify as a bug/issue.
@lumapu can you qualify that ?
Regarding the reboots / stack traces I do not know wether we can decode them without the right version / githash ?

Duesentrieb5413, 2023-01-22:
I have the same issue with Ahoy DTU not resetting the values with no updated data.
I tried the current dev-version 0.5.75 with some remarks:
- Using the option "Reset Values at Communication stop" resulted in reset values all the time (related to this issue). When I deactivated this option, updated values and timestamp were shown as expected: [image]
- installing via webinstaller did not work - I always got the error "Failed to fetch" (Chrome was started with parameter). Thus flashed via https://github.com/nodemcu/nodemcu-flasher
[image]
- The build was not stable - after some time the website of Ahoy did not show menu/values any more
[image]
Should I open additional issues for my second and third remark?

lumapu, 2023-01-22:
for second check issue #622

The third one is sometimes seen during development but was never reported to me

lumapu, 2023-02-07:
for the third point there is an individual issue now: #660

closed 2023-02-07 by lumapu