lumapu/ahoy #576: [ESP8266] AhoyDTU main page says "not procuding", but the inverter does

SpaceTeddy, 2023-01-11:
### Platform

ESP8266

### Model name

Wmos d1 mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

wire

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.70

### Github Hash

d8e255d

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

...

### Error description

-> Ahoy main page shows "inverter is available but not producing"
[image]

-> Live page shows P_AC: 2.7W P_DC:2.9W
[image]

After a while that P_AC is > 3W, the main page changed to "producing!"
 
Is there a P_xx Power threshold for the main page information`?

thx

knickohr, 2023-01-11:
Yes, if the power is below 3W, the Inverter returned „available and **not** producing“. This will stop if the inverter is not responding in 5 minutes or the Power is more than 3W.

SpaceTeddy, 2023-01-11:
alright, thanks a lot for your confirmation!