lumapu/ahoy #581: No communication with version 0.5.66

mipcom, 2023-01-13:
### Platform

ESP8266

### Model name

WeMos D1 Mini Pro V3.0

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

board

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin  GND [*]  | GND            |
| Pin  VCC    | +3.3V          |
| Pin  CE      | GPIO5  CE   D1 |
| Pin  CSN     | GPIO15 CS   D8 |
| Pin  SCK     | GPIO14 SCLK D5 |
| Pin  MOSI    | GPIO13 MOSI D7 |
| Pin  MISO    | GPIO12 MISO D6 |
| Pin  IRQ     | GPIO4  IRQ  D2 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

73b18322d7ac368fe85cdf51e06025ee

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-1500
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 405, 405, 405, 405
    - Module Name: O1,O2,W,S
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D1 (GPIO5)
    - IRQ: D2 (GPIO4)
#### Radio (NRF24L01+)
    - Amplifier Power Level: HIGH
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

1- ran 0.5.41 with same settings fine, no issues
2- upgraded with ESP8266Flasher.exe in Win10 to 0.5.66
3- no seriell communication, even no error
4- downgraded to 0.5.41 via Gui Update, DTU works fine now
5- tried one more time to upgrade now via Gui Update to 0.5.66, same
6- no seriell communication, even no error
7- changed back to 0.5.41 via Gui Update, DTU works fine

Gerri1, 2023-01-13:
Das Flashtool >flash_download_tool_3.9.3< oder den Tasmotizer 1.2 nehmen, bei denen kann man das mit "Erase before flashing" machen.

mipcom, 2023-01-13:
Hallo Gerhard,

 

ich habe es eben mit Tasmotizer und „Erase before Flash“ gemacht, aber keine Änderung.

Anbei mal alle Logs/Setup Daten.

 

Mit besten Grüßen

Dietmar Pfitzner

Gerri1, 2023-01-13:
Die PIN's Hardware - und Softwareseitig mal getautscht?

mipcom, 2023-01-13:
Wie beschrieben, habe ich für v0.5.41 am WeMos schon D1 und D2 genommen, da D3 und D4 wie im Original nicht ging. Und es auch SW seitig dementsprechend konfiguriert. Mit Besten GrüßenDietmar Pfitzner Am 13.01.23 um 14:03 schrieb Gerhard
                
            
            
                
                    Von: "Gerhard" ***@***.***>Datum: 13. Januar 2023An: "lumapu/ahoy" ***@***.***>Cc: "Author" ***@***.***>,"mipcom" ***@***.***>Betreff: Re: [lumapu/ahoy] No communication with version 0.5.66  (Issue #581)
                
                
Die PIN's Hardware - und Softwareseitig mal getautscht?

mipcom, 2023-01-13:
Leider habe ich mein Gehäuse verklebt und komme da nicht mehr so leicht ran. Aber wieso funktioniert es tadellos mit v0.5.41 ?Mit Besten GrüßenDietmar Pfitzner Am 13.01.23 um 14:15 schrieb Gerhard
                
            
            
                
                    Von: "Gerhard" ***@***.***>Datum: 13. Januar 2023An: "lumapu/ahoy" ***@***.***>Cc: "mipcom" ***@***.***>,"Author" ***@***.***>Betreff: Re: [lumapu/ahoy] No communication with version 0.5.66  (Issue #581)
                
                
Ich meinte damit D1 und D2 untereinder tauschen.

Gerri1, 2023-01-13:
> Leider habe ich mein Gehäuse verklebt

... das ist dann schlecht!
Bei meinem ESP funktionieren seit der 0.5.66 auch einige PIN's nicht mehr , was bei der 0.5.41 noch ging.

mipcom, 2023-01-13:
Ja, aber warum ändert sich das?Im ChangeLog habe ich dazu nichts gelesen. Man kann ja nicht jedesmal neu ausprobieren was wie geht. Schönes Wochenende und Beste GrüßeDietmar Pfitzner Am 13.01.23 um 15:52 schrieb Gerhard
                
            
            
                
                    Von: "Gerhard" ***@***.***>Datum: 13. Januar 2023An: "lumapu/ahoy" ***@***.***>Cc: "mipcom" ***@***.***>,"Author" ***@***.***>Betreff: Re: [lumapu/ahoy] No communication with version 0.5.66  (Issue #581)
                
                

Leider habe ich mein Gehäuse verklebt

... das ist dann schlecht!

Bei meinem ESP funktionieren seit der 0.5.66 auch einige PIN's nicht mehr , was bei der 0.5.41 noch ging.

Gerri1, 2023-01-13:
Bei meiner NodeMCU v3 Lolin gehen dafür D3 und D4 für CE und IRQ komischerweise wieder bei der 0.5.66.
Dir wird da aber eh nicht anderes übrig bleiben, als das Gehäuse zu knacken, es sei denn du willst bei der 0.5.41 bleiben.

lumapu, 2023-01-13:
an den Pins wurde nichts geändert. Leider verhalten sich die ESPs hier sehr unterschiedlich.

mipcom, 2023-01-13:
Das werde ich dann irgendwann mal machen müssen. Danke für euch Feedback. Mit Besten GrüßenDietmar Pfitzner Am 13.01.23 um 20:42 schrieb Gerhard
                
            
            
                
                    Von: "Gerhard" ***@***.***>Datum: 13. Januar 2023An: "lumapu/ahoy" ***@***.***>Cc: "mipcom" ***@***.***>,"Author" ***@***.***>Betreff: Re: [lumapu/ahoy] No communication with version 0.5.66  (Issue #581)
                
                

Bei meiner NodeMCU v3 Lolin gehen dafür D3 und D4 für CE und IRQ komischerweise wieder bei der 0.5.66.

Dir wird da aber eh nicht anderes übrig bleiben, als das Gehäuse zu knacken, es sei denn du willst bei der 0.5.41 bleiben.

stefan123t, 2023-01-13:
@mipcom schau doch bitte nochmal nach welche Firmware Du aktuell verwendet hast ?

Wir haben auch drei Firmware Binaries für I2C Displays im Programm (`_nokia5110.bin`, `_ssd1306.bin` & `sh1106.bin`).
Die Displays belegen i.d.R. D1 & D2, weil nur da beim EPS8266 der I2C Bus angeschlossen werden kann.

[image]

PS: Bitte wenn es geht in GitHub antworten und nicht immer per EMail Reply, dann haben wir nicht immer den ganzen Thread Verlauf im Issue :D Danke!

mipcom, 2023-01-14:
Ich habe die esp_8266 ohne extras genommen, allerdings die v.0.5.66.
Vielleicht sollte ich mal 0.5.70 testen, wenn die schon verfügbar ist.

jimknopf63, 2023-01-14:
0.5.69 und 70 haben aber ein Problem mit der Limitierung falls das bei Dir genutzt wird. Er schreibt immer das die Änderung nicht akzeptiert wird, ändert aber anscheinend doch den persistent Wert.

mipcom, 2023-01-14:
Ja, das Powerlimit wollte ich nutzen. Mit Besten GrüßenDietmar Pfitzner Am 14.01.23 um 11:15 schrieb Stephan
                
            
            
                
                    Von: "Stephan" ***@***.***>Datum: 14. Januar 2023An: "lumapu/ahoy" ***@***.***>Cc: "Mention" ***@***.***>,"mipcom" ***@***.***>Betreff: Re: [lumapu/ahoy] No communication with version 0.5.66  (Issue #581)
                
                

0.5.69 und 70 haben aber ein Problem mit der Limitierung falls das bei Dir genutzt wird. Er schreibt immer das die Änderung nicht akzeptiert wird, ändert aber anscheinend doch den persistent Wert.

lumapu, 2023-01-16:
@jimknopf63 @mipcom das Powerlimit geht, nur wird in der seriellen Ausgabe  versehentlich ein `not` zu viel ausgegeben. Die Version `0.5.71` wird das nicht mehr machen. Siehe auch #569

mipcom, 2023-01-16:
Danke, werde dann die Version probieren.

closed 2023-02-07 by lumapu

mipcom, 2023-04-07:
So, kurze Rückmeldung.
Ich habe meine Box mal geöffnet und die Pins D3 und D4 getauscht und dann läuft's.
Habe jetzt die v.0.6.0 drauf.
Danke für euren Support.
Beste Grüße und schöne Ostern . . 
. . mipcom