lumapu/ahoy #584: Inverter Intervall

knickohr, 2023-01-13:
Die Abfrage der Inverter, gesetzt mit Settings Inverter iInterval, verdoppelt, derdreifacht, vervierfacht … sich mit jedem weiteren Inverter.

Würde bedeuten das bei 6 Invertern bei 30s Intervall ein kompletter Abfragezyklus 6x30s, also 900s dauern würde. 

So gewollt, oder ist das ein Bug ?

Wenn es so gewollt ist, dann müßte man doch problemlos das Intervall runter setzen können ohne das es Probleme gibt ?

Mir ist das schon länger aufgefallen, nicht erst seit der DEV 70, habe dem aber noch wenig Beachtung geschenkt.

DanielR92, 2023-01-13:
Guter Punkt, da ich mich in anderen Bereichen gewendet habe... hab ich das nie gesehen. 
Das muss man mal prüfen. 

Was ich nur weiß, ich habe damals mehrere Zombie-Einträge angelegt und Ahoy versucht die nicht vorhandnen WR zu erreichen. Hierfür probiert er 5 mal und wechselt danach zum nächsten WR. 

Ich würde mir wünschen, das man eine andere Logik hinterlegt. 

Beispiel:
- Wenn ein WR nur definiert ist, soll er dauerhaft probieren den WR zu erreichen. 
- Bei mehreren WR soll 2-3 probiert werden diese zu erreichen, danach soll er zum nächsten hopsen. 

Somit würde man nicht ewig warten, bis dann der 6te WR die Daten zu sehen.

knickohr, 2023-01-13:
Ganz zu Beginn, ich glaube es war so um die 17 rum, war das mal „gefühlt“ anders. Ich bilde mir ein, man konnte es an der LED am ESP sehen wie er gepollt hat und wie er zum nächsten Inverter ging. Habe dem da auch nicht allzu große Aufmerksam geschenkt.

Ich finde, die DTU soll nur versuchen alle eingetragenen Inverter genau x-mal (Einstellung) zu pollen, danach sofort zum nächsten gehen und da das gleiche Spiel wieder machen. das Intervall soll aber für alle Inverter sein, nicht hintereinander jeweils.

Es ist mir klar, das irgendwann die Intervallzeit nicht mehr ausreichen wird. So geschätzt braucht er pro Inverter 3 bis 5 Sekunden. Wenn also einer nicht antwortet kann das schon mal 15 Sekunden sein bis er die 5 Versuche durch hat. Würde bedeuten das die Intervallzeit schon nach 2 Invertern abgelaufen sein kann.

Mit der jetzigen Konfiguration warten er halt einfach die Intervallzeit ab (abzüglich der „verpollten“ Zeit) bis er zum nächsten geht. Auch nicht schlecht, man sollte es halt im Hinterkopf behalten. So könnte er nicht in Zeitdruck geraten. Dann würde ich es halt nicht mehrIntervall nennen, sondern Pause.

knickohr, 2023-01-17:
Siehe auch #564

knickohr, 2023-08-10:
Mit den bereits sehr kurzen möglichen Intervallzeiten ist dieser Issue hinfällig.

closed 2023-08-10 by knickohr