lumapu/ahoy #586: Power Limit per Befehl ohne MQTT

dtuuser, 2023-01-14:
Ahoy läuft jetzt seit 6 Tagen ohne Probleme. Aktuelle Werte per eigener Anwendung vom Stromzähler auslesen läuft auch.
Jetzt würde ich gerne eine Nulleinspeisung realisieren. Meine Anwendung lässt bei Einspeisung aktuell einen kleinen Leuchtturm im Regal leuchten. Jetzt meine Idee.
Sobald längere Zeit eingespeisst wird, würde ich gerne ein Power Limit am WR einstellen.
So nach dem Motto....
http://ahoyip/api/????/powerlimit(wrid)/Wattangabe
Leider habe ich im Netz nichts dazu gefunden, ob das geht. Daher meine Frage, gibt es die Möglichkeit das über einen Rest Api Aufruf  zu realisieren?
Wäre klasse, ansonsten würde ich irgendwie versuchen einen mqtt Befehl per Visual Basic zu schicken.

knickohr, 2023-01-14:
https://github.com/lumapu/ahoy/blob/main/User_Manual.md

dtuuser, 2023-01-14:
Hatte ich gesehen. Aber wie schickt man eine Json Datei an das Webapi?

stefan123t, 2023-01-14:
Du mußt einen POST Befehl mit Content-Type application/json absetzen.
Z.B. so mit CURL auf der Kommando Zeile:
```bash
curl --location --request POST 'http://ahoy-dtu.local/api/ctrl' \
--header 'Content-Type: application/json' \
--data-raw '{
    "id": <INVERTER_ID>,
    "cmd": "limit_persistent_absolute",
    "val": <VALUE>
}'
```

dtuuser, 2023-01-14:
Hallo stefan123t,
danke für das Script.  Bekomme unter Windows in cmd immer Fehlermeldungen. 

Im Script habe ich aus '  " gemacht 

curl --location --request POST "http://192.168.2.4/api/ctrl" \
--header "Content-Type: application/json" \
--data-raw "{
    "id": 1,
    "cmd": "limit_nonpersistent_relative",
    "val": 50
}"

C:\>curl --location --request POST "http://192.168.2.4/api/ctrl" \
curl: (52) Empty reply from server
curl: (6) Could not resolve host: \

C:\>--header "Content-Type: application/json" \
Der Befehl "--header" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

C:\>--data-raw "{
Der Befehl "--data-raw" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

C:\>    "id": 1,
Der Befehl ""id":" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

C:\>    "cmd": "limit_nonpersistent_relative",
Der Befehl ""cmd":" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

C:\>    "val": 50
Der Befehl ""val":" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

C:\>}"

;-(

dtuuser, 2023-01-14:
Läuft

curl -i -X POST -H "Content-Type: application/json" -d "{""id"":""1"",""cmd"":"""limit_nonpersistent_relative""",""val"":""50""}" http://192.168.2.4/api/ctrl

Kann es sein, dass der WR nach einer Limitänderung erst einmal für einige Minuten nicht erreichbar ist?

lumapu, 2023-01-14:
normalerweise nicht. Evtl. kannst du parallel in die webserial console schauen und sehen was passiert

dtuuser, 2023-01-15:
so, endlich hell......

WR ist nicht weg, aber er aktualisiert erst 3 Minuten nach dem Limit Set. Erkläre ich mit dem eingestellten Intervall von 60 Sekunden. Daten werden abgefragt, Limit wird beim nächsten Mal geschickt, beim nächsten Intervall kommt die Rückmeldung, dass das Limit akzeptiert wurde und danach kommen die Daten wieder normal rein. 

9:15:04 Last received data request 
dann sende 33%. Laut serial/console um 9:16:04 vom WR akzepriert
9:16:04 Anzeige im Live Fenster auf 33% geändert, aber received time noch bei 9:15:04
9:18:04 nächsten Datensatz empfangen

9:18:04 Last received data request 
9:19:04 100% gesetzt laut serial control vom wr akzeptiert
9:20:04 Anzeige im Live Fenster auf 100% geändert, aber received time noch bei 9:18:04
9:21:04 nächsten Datensatz erhalten

Nulleinspeisung ist damit per Batchdateien realisierbar

Danke nochmals an stefan123t

closed 2023-02-07 by lumapu