lumapu/ahoy #588: Gibt es die Möglichkeit eine "Scan" Funktion zu implementieren?

Meckiilino, 2023-01-15:
Hallo zusammen,
ich meine so eine Funktion, die man aus dem WLAN-Bereich kennt.
Einmal auf Scan drücken und alle kompatiblen Inverter in Reichweite werden aufgelistet.

Juergen2453, 2023-01-15:
Wozu, willst du dann andere Inverter manipulieren? (So in der Art, der Nachbar hat mich geärgert dann manipuliere ich sein Inverter)

Meckiilino, 2023-01-15:
Autsch - daran habe ich gar nicht gedacht.
Es war eigentlich als Erleichterung gedacht - wenn man die Seriennummer nicht zur Hand hat und aufs Dach müsste.

Issue kann somit geschlossen werden...

stefan123t, 2023-01-15:
Dazu brauchst Du das BROADCAST 0x02 Command laut der Hoymiles DTU Software.
Wir haben bisher unzureichende Informationen zum Ablauf und brauchen ordentliche Traces von einer echten DTU und bevorzugt mehreren WR die discovered / gescannt werden.
Siehe https://github.com/lumapu/ahoy/wiki/Protocol#broadcast-0x02

closed 2023-01-15 by stefan123t

Thomas-PV-Tryer, 2023-04-01:
Schade - ich hatte das gleiche "Urproblem nicht lesbare Seriennummer"! Habe mich dann an die Hoymiles Hotline gewendet - dort gibt es ja die Scanfunktion und die Hotline verweist auch drauf.
Wäre toll, wenn sie doch jemand tragen und dann implementieren könnte ...
Mir sind die Hoymiles DTU's zu teuer ....