lumapu/ahoy #589: ESP8266

Loetnase, 2023-01-15:
### Platform

ESP8266

### Model name

Wemos D1 mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.70

### Github Hash

d8e255d

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

hallo zusammen

230108_ahoy_0.5.70_d8e255d_esp8266_sh1106.bin

In der neuen 0.5.70 Yield Total Correktion sind die Werte beim Berechnen vertauscht.
30,61kWh Total minus 20 kWh sollten nicht -9,4 sondern 10,6 kWh rauskommen.
[image]

Ich wünsche noch ein schönes Wochenende

### Error description

[image]

lumapu, 2023-01-16:
ich denke es war eine fehldene Klammer 😄

Loetnase, 2023-01-21:
@lumapu 
ich denke an der Klammer lag es nicht, ich sehe immer noch negative Werte. 
in der 5.73 und der 5.75 ist das noch gleich :-(

Ich denke es wird das Doppelte (40) vom Yield Total Correction (bei mir 20 eingetragen) der  Yield Total (aktuell 36,65 kWh) abgezogen ergibt die angezeigten -3,35 kWh. 
Der Faktor 2 könnte mit der Modulanzahl (2 bei mir) zusmmenhängen

Yield Total Correction (bei mir 20 eingetragen)
[image]
[image]
[image]

Yield Total Correction (bei mir 0 eingetragen)
[image]

Eventuell noch ein paar Gedanken zum Offseteintrag.
Ich denke der Eintrag wird durch die Anzahl der Module geteilt und dann auch bei den einzelnen Modulen abgezogen. Aber sobald die Module unterschiedlich produzieren (Verschattung usw.) oder gar bei einem 4-fach Inverter nur 3 Module angeschlossen sind, passt das Verhältnis so nicht mehr.
Ist es ein großer Aufwand für jedes Modul einen eigenen Offseteintrag machen zu können?
Die Summe daraus würde ja dann die Yield Total Correction ergeben.
Ich denke nur so lassen sich die einzelnen Module bewerten.

Heute bei leichtem Schneefall wurden bis jetzt 112 Wh produziert.
Ich wünsche allen ein schönes Wochenende. :-)

dtuuser, 2023-01-22:
Gerade auf 0.5.76 geupdatet.
Habe jetzt mal den halben Korrekturwert bei der Correction eingegeben. Jetzt passt es. 
Kann es sein, dass der Correction Wert jeweils vom Panel abgezogen wird?

Loetnase, 2023-01-22:
@dtuuser Ja, du hats recht, mit dem halben Korrekturwert passt das bei mir auch in der 0.5.76 an meinem HM600.

Da ist noch was zu tun.

Kann das noch jemand an einem 4 fach Wechselrichter HM1200 oder HM1500 ausprobieren, vielleich muss da nur ein Viertel eingetragen werden, oder bei einem HM300 der ganze Wert. 

In der  Version 0.5.76 sind jetzt auch schön die Displays mit eingebaut worden.
Ich habe das SH1106 ausgewählt aber die parallel geschalteten SSD1306 interresiert das nicht, die funzen auch in meinem Testaufbau.
[image]

Das gefällt mir. Vielen Dank an den oder die "Macher".

Loetnase, 2023-02-02:
@lumapu, @stefan123t, @dtuuser   

in der neuen 0.5.78 ist der Bug immer noch drin.
Es funktioniert aber wenn man den halben Offsetwert einträgt.

Eventuell noch ein paar Gedanken zum Offseteintrag. Siehe auch Post oben.
Ich denke der Eintrag wird durch die Anzahl der Module geteilt und dann auch bei den einzelnen Modulen abgezogen. Aber sobald die Module unterschiedlich produzieren (Verschattung usw.) oder gar bei einem 4-fach Inverter nur 3 Module angeschlossen sind, passt das Verhältnis so nicht mehr.
Ist es ein großer Aufwand für jedes Modul einen eigenen Offseteintrag machen zu können?
Die Summe daraus würde ja dann die Yield Total Correction ergeben.
Ich denke nur so lassen sich die einzelnen Module bewerten.

lumapu, 2023-02-02:
klingt sinnvoll und wäre konsistent. Denke auch wir müssen für jeden Eingang einen Wert vorsehen.

MiniOh, 2023-02-11:
Hallo, 

wurde hier schon etwas angepasst?
In der 0.5.78 ist es ja noch so, dass die eingetragene "Yield Total Correction (will be subtracted) [kWh]" im einem Inverter, pro Modul abgezogen wird.

Also wenn ich meine Yield Total um 1000kWh anpassen möchte, mütte ich bei einem HM-1500 "250" eintragen, richtig?
Zudem ist es nur möglich, die Werte zu verringern, da die Werte ja subtrahiert werden.
Ich möchte eigentlich +1000kWh in die Yield Total dazurechnen. Geht das auch?

Danke schon mal.

lumapu, 2023-02-11:
ja das ist noch offen, Plan ist es Modulweise also pro Eingang anzubieten

MiniOh, 2023-02-11:
Ist es evtl möglich kurzfristig positive Korrekturen einzubauen mit einer Auswahl ob + oder -?
Pro Modul kann man sich ja helfen indem man ganze, halbe oder viertel Werte einträgt?
Evtl hast du ja schnell eine 0.5.86 gebaut?
Würde mich freuen, da ich einen Inverter tauschen musste.

lumapu, 2023-02-11:
eher gegen Abend heute 😃

MiniOh, 2023-02-11:
Klingt fast genauso gut ;)

MiniOh, 2023-02-12:
Morgen, 

vielen Dank fürs umbauen.
Beim ersten Versuch scheint aber noch nicht alles zu passen.
Ich habe bei einem HM-1500 bei Modul 1 und 2 jeweils 333 kWh addiert.

Hier werden aber bei Modul 4 merkwürdige Werte ausgegeben.

[image]

[image]

closed 2023-03-27 by lumapu