lumapu/ahoy #594: Feature request: Display-Helligkeit einstellen per Weboberfläche.

TB-79, 2023-01-16:
Die Displays mit weißer Schrift sollen schnell altern/einbrennen.
Von daher wäre die ein/aus und auch die Sunrise-Kopplung anzuraten (Feature request bereits eingereicht) und zusätzlich auch die Möglichkeit die Helligkeit einzustellen in den Web settings.

knickohr, 2023-01-16:
Sehr gut, wenn wir dann schon dran sind, auch gleich eine Option zu. drehen des Displays um 180 Grad. Wird bei den blau/gelben OLED interessant, da diese einen hardwaretechnisch bedingte abgedeckte Zeile haben. Prinzipiell könnte man das OLED auch dauerhaft um 180 Grad drehen.

kloppy1984, 2023-01-17:
Ich denke am ESP8266 wird kein Pin frei sein, für die Steuerung der Helligkeit

knickohr, 2023-01-17:
Das wird über die Software eingestellt 😉

kloppy1984, 2023-01-17:
@knickohr 
Dann hätte ich gern mal ein Anschlussplan wo der Light angeschlossen wird... :-) 
Bei sind alle alle Pins belegt...

knickohr, 2023-01-17:
Es geht hier um OLED, wie @TB-79 im ersten Beitrag geschrieben hat. Klar, das Nokia kann man beleuchten, kannst ja über einen Schalter machen. Wenn Du einen PWM möchtest, dann wird’s in der Tat knapp mit dem 8266.

closed 2023-03-27 by lumapu