lumapu/ahoy #597: Sonderzeichen im Modul Namen

Ollipop030, 2023-01-17:
Nur eine Kleinigkeit:

Sonderzeichen im Modulnamen werden zwar übernommen, aber falsch dargestellt. Aus "Süd" wird "S�d". Bekommt man das irgendwie hin?

knickohr, 2023-01-17:
Ich würde ehrlich gesagt von Sonderzeichen, Umlauten und all dem Zeuchs Abstand nehmen. Bringt immer wieder Probleme, auch wenn es augenscheinlich funktionieren mag.

QuickStopGroceries, 2023-01-17:
Du kannst "Süd" als "S&uuml;d" eintragen, dann funktioniert es.

stefan123t, 2023-01-18:
Kannst Du mal schauen ob die Seite als UTF-8 encoded ausgeliefert wird ?

Ollipop030, 2023-01-18:
> Du kannst "Süd" als `S&uuml;d` eintragen, dann funktioniert es.

Das ist ein guter Tipp, danke.

stefan123t, 2023-01-18:
@lumapu der Workaround mit `&uuml;` (6 Bytes) statt dem Umlaut `ü` 1 Byte will mir nicht recht gefallen.
Haben wir den Content-Type Header auch überall richtig gesetzt ?

Hier wird er nur im Request-(?) Header gesetzt:
https://github.com/lumapu/ahoy/blob/main/src/web/html/api.js#L99
```c
            xhr.setRequestHeader("Content-Type", "application/json;charset=UTF-8");
```

Und hier zum Download:
https://github.com/lumapu/ahoy/blob/main/src/web/RestApi.h#L144
```c
            response->addHeader("Content-Type", "application/octet-stream");
```

Müssten wir den Content-Type Header mit `charset=UTF-8` nicht auch generell im Response Header der API setzen oder macht das der ESPAsyncWebServer im Default ?

lumapu, 2023-02-27:
ich brauche den issue noch als Gedankenstütze 😊

Ollipop030, 2023-02-28:
Sorry dafür. Habe gestern mal ein paar erledigt Issues geschlossen. War hier wohl ein bisschen zu schnell. ;-)

lumapu, 2023-03-08:
sollte mit aktuellen Dev Versionen gehen, ich denke ab `0.5.93`

closed 2023-03-27 by lumapu