lumapu/ahoy #599: Verhaltensweise Powerlimit persistent/non persistent

MubiTec, 2023-01-18:
Hallo zusammen,

Kurze Frage zur Logik von persist bzw. Non persistent.
Grundfunktion ist mir eigentlich klar. Persistent bleibt auch bei WR aus erhalten, non persistent nicht.
Was passiert, in folgenden Szenario? Ich setze einen persistenten Wert, z. B. 50%. Anschließend setze ich dann einen non persistenten Wert auf z. B. 80%. Anschließend ist Nacht und der WR geht aus, der non persistenten Wert ist also gelöscht. Ist der persistenten 50% Wert dann noch da? Oder gibt's dann gar kein Limit mehr?
2. Frage: ich meine mal irgendwo gelesen zu haben, dass persistente Werte auf die Lebenserwartung des WR, weil die ins EEPROM geschrieben werden. Ist das wirklich so? Sollte man die Werte also nicht täglich neu setzen?

Danke und Gruß!

stefan123t, 2023-01-18:
@MubiTec ja hast Du beides richtig verstanden. Das temporäre Limit von 80% wird befolgt.
1. WR war aus, also wird beim Start das persistent Limit aus dem Flash/EEPROM des WR wieder aktiv. Das temporäre Limit des WR von 80% hat er dabei vergessen.
2. der WR muss sich das persistent Limit irgendwo in Flash/EEPROM speichern. Dieses wird eine begrenzte Lebensdauer (ca 10.000-1.000.000 Schreibzyklen ?) haben. Also tägliches neu Setzen des persistent Limit dürfte zwar eine ganze Weile gut gehen aber empfohlen wird es aus den bekannten Gründen nicht. Warum auch ? Es gibt ja das temporäre Limit das nur im RAM gespeichert wird und keiner Alterung unterliegt.

MubiTec, 2023-01-18:
ok, danke für die schnelle Info! 
Bei dem Verhalten brauch ich tatsächlich das persistente Limit nur ein einziges Mal zu setzen.
Ich will nachts per Batterie einspeisen und da will ich dann direkt ein Limit aktiv haben, sodass er da nicht gleich unter Volllast reinknallt. Und das geht ja nur mit dem persistenten, aber wenn das immer aktiv ist, dann passt das ja. Dann muss ich nur Tagsüber nach dem Starten das non persistente auf 100% setzen bzw. Nachts dann anpassen, um genau den derzeitigen Verbrauch zu treffen.

closed 2023-01-18 by MubiTec