lumapu/ahoy #600: Feature request: MQTT -> Alarm Nachricht als string und nicht nur ID

MadMacMad, 2023-01-18:
hallo,
erstmal Danke für die tolle arbeit !
würde mich freuen wenn man die letzte Alarm Nachricht wie im Webinterface zu sehen über MQTT mit senden könnte.
Das würde sehr helfen bei der Überwachung...
Danke

knickohr, 2023-01-19:
Hmmm,

wir haben erst kürzlich alle Textmeldungen in numerische Werte umgestellt. Nun gut, die Alarm-Message war immer schon numerisch, aber würde bedeuten das wir jetzt wieder genau das Gegenteil machen 🤔

Wie denkt ihr darüber ?

MadMacMad, 2023-01-19:
> warum hinterlegst Du nicht einfach eine Tabelle mit dem entsprechendem Text in Deinem MQTT-Client ?
könnte ich auch machen... aber wo finde ich denn die codes mit den entsprechenden Meldungen dazu?
ausserdem kommen die teils unterschiedlich an... 
Ahoi sagt Alarm: 
HM-1200 -> Inverter Start
HM-1200 -> Inverter Start
MQTT sagt: 
ALARM_MES_ID = 80 -> HM-1200
ALARM_MES_ID = 89 -> HM-1500
seit meinen Aufzeichnungen werden die Nummern immer größer... also macht das gar keinen sinn daraus irgendwas abzuleiten... 
deswegen aus meiner Sicht sollte man die Nachricht "last Alarm: Inverter start" und nicht eine aufsteigende Nummer ohne Bedeutung senden... 
oder zumindest eine eindeutige ID die für einen Status steht...

MadMacMad, 2023-01-19:
hab jetzt die MQTT Nachricht für LastAlarmCode gefunden....
dort steht bei MQTT:
LastAlarmCode = 1 -> denke bedeutet Inverter start ... laut Manual ...
gibt es eine Auflösung was die anderen ausser 1 ... eg Fehlfunktion ... zu heiß .... Strom Problem... haben könnte?
hab da leider nix zu gefunden....
von meiner Hoymiles DTU weiß ich das es noch andere Fehlermeldungen gibt...
Danke

stefan123t, 2023-01-20:
Also das Parsen des kompletten AlarmData bzw AlarmUpdate Payloads ist mW in Ahoy noch nicht vollständig implementiert. So wurde auf dem Discord auch schon bemängelt, dass unter LastAlarmCode in Wirklichkeit der FirstAlarmCode steckt und der ist idR. immer 1 weil das eben das erste ist was der Inverter früh morgens in seine AlarmLog schreibt.
Die Dokumentation des Protokolls ist hier https://github.com/lumapu/ahoy/wiki/Protocol#wie-sieht-das-device-info-kommando-alarm-data-0x11--alarm-update-0x12-aus-
Und dort findet sich mit dem Link zur Bedienungsanleitung auch die offizielle Liste der Fehlercodes https://www.alpha-solar.info/media/Dokumente/Wechselrichter/Hoymiles%20HM/Deutsch/Bedienungsanleitungen/HM-600%20%20HM-800%20Bedienungsanleitung.pdf
Als Richtschnur kann man auch die Implementierung von OpenDTU zu Rate ziehen dort sind mW alle Fehlercodes hinterlegt sogar Einige (wie 2 DTU command error) für die wir keine Dokumentation gefunden haben.

https://github.com/tbnobody/OpenDTU/blob/master/lib/Hoymiles/src/parser/AlarmLogParser.cpp

stefan123t, 2023-01-20:
Siehe #608 und #177

lumapu, 2023-01-21:
@stefan123t in den neusten development Versionen ist es drin, allerdings nur auf die Serielle Console.

@MadMacMad ich habe vor das zu implementieren

closed 2023-08-17 by lumapu