lumapu/ahoy #605: Nulleinspeisung all-in-one mit ESP32 und CT-Klemmen

Adminius, 2023-01-19:
Erst mal danke für das tolle Projekt!

ich habe keine Möglichkeit an den Stromzähler zukommen (Mehrfamilienhaus), deswegen muss ich mit CT-Klemmen (z.B. günstige 100A SCT-013-000) Strom auf allen 3 Phasen direkt in der Unterverteilung in der Wohnung ermitteln. Spannung wird als fester Wert angenommen, z.B. 230V und so hat man aktuellen Verbrauch (genau genug für Nulleinspeisung).
Die Berechnungen laufen über emon-Library.
Nun, der selbe ESP32 kann gleich auch Ahoy-Part abdecken, damit man nicht noch mit 2 ESP arbeiten muss, die über etwas (z.B. MQTT) mit einander reden müssen.

Habt ihr Interesse das mit in die FW aufzunehmen? Wenn ja, ich könnte den Code zumindest teilweise beisteuern (bin gerade dabei zu schreiben) und ich bräuchte Hilfe wo ich am besten ansetzen kann.

Die Idee geht leider nur mit ESP32, da ESP8266 nur einen und sehr schlechten Analogeingang hat. 
Und externe i2c ADC würde ich gerne vermeiden (noch mehr Hardware, Verkabelung, SW...)

Adminius, 2023-01-20:
das könnte dafür ein guter Start sein:
https://github.com/lumapu/ahoy/issues/410

Ollipop030, 2023-01-20:
Wäre eine schöne Lösung, gerade für Mieter. Die Spannung muss aber nicht fest auf 230V stehen, die kann man doch von dem Wechselrichter abfragen, somit wäre die Leistung genauer berechnet.

stefan123t, 2023-01-20:
@Adminius das klingt nach einem interessanten Ansatz. Willst Du das evtl. im Discord einmal vorstellen oder durchsprechen ?
Ich weiß dass einige an einer ModBus485 Schnittstelle oa. für genau so einen Anwendungszweck arbeiten.
Es gäbe also ein / mehrere Input Plugins für die aktuelle Leistung per ModBus485 oder mit den von Dir geplanten ADCs und CT Klemmen.
Und dann müsste man eine allgemeine Logik / Steuerung anbieten die entweder einen festen Betrag (idR 600W bzw 800W) über der verbrauchten Leistung oder eben 0W Einspeisung sicherstellt und bei Bedarf das temporäre Limit des / der Inverter aktualisiert.
Die Original Hoymiles DTU Pro mach das in AntiReflux.c bzw AntiBackflow.c
Dort kommen 30W In-/Dekremente je Inverter zum Einsatz um die Regelung nicht zu häufig und zu granular zu machen.

Adminius, 2023-01-20:
@Ollipop030 leider nein. Die Idee hatte ich auch schon. der WR hebt die Spannung ab damit er einspeisen kann und schon ist diese höher als die wirklich ist und dann nur auf einer Phase. Ich habe erst heute wieder beobachtet, bei keiner sehr kleiner Einspeisung: 234V, bei vollen 600W waren es schon 237V. Aber, mein SmartPI in der UV hat immer noch nur um 234 angezeigt. Andere Phasen waren wieder bei den anderen Werten. D.h. da wäre mir lieber, wenn ich einen fixen Wert eintrage. Ja, die Spannung springt hin und her je nach Tageszeit, aber Mittelwert könnte man schon annehmen. Und mal ehrlich:
Bei z.B. 600W WR sind es um 2,5A AC. Ob die Spannung 230, 234 oder 237V ist, ist fast schon egal, da wir gerade mal von 10W bei 234V  bzw. 17W bei 237V Unterschied im Vergleich zu 230V reden. Wir reden hier über 2-3% Abweichung. So genau sind geeichte Zähler wenn überhaupt. Also ist ein fixer Wert z.B. 234V (zumindest bei mir) wäre genau genug.
[image]

@stefan123t ja klar, ich kann es auch im Discord vorstellen, muss nur noch mich dort anmelden :)
Deine Idee mit InputPlugins ist sehr cool, dann kann jeder sein eigenen Weg auswählen. Bitte noch MPPT subscription an einen bestimmten Topic der die Aktuelle Leistung liefert ;)

Warum ich eine Nulleinspeisung brauche: mein Hoymiles hängt am Akku, Akku wird von Victron MPPT geladen und Victron wird von 2x PV Modulen gespeist. Balkonkraftwerk mit Akku.

Ideensammlung:
Es muss eine Möglichkeit geben minimale Leistung einzutragen (Jeder Haushalt hat Standby, dank Router und Kühlschrank und anderen Geräten in Standby). Es kann z.B. 50W sein. Unter diese Leistung wird nicht geregelt. Muss noch überlegen ob es wirklich sinnvoll ist... hm.

Wenn es automatisch nicht bekannt ist, muss man maximale Leistung des WRs eingeben können (z.B. 600W). D.h. wenn der Hausverbrauch höher als 600W ist: wird max. 600W eingespeist.
Man muss letzten Watt-Limit merken (nicht %!): Beispiel: Verbrauch ist 100W, wir stellen WR auf 100W. Jetzt geht der Verbrauch auf 150W, sehen wir nur 50W. D.h. der WR muss auf die gemerkten 100W + zusätzliche 50W eingestellt werden, aber nicht höher als MAX-Limit (600W aus dem Beispiel).

Für Akkubetreiber: Spannungslimits am PV Eingang festlegen. Beispiel: 16S LiFePo4 Akku wäre Spannungsbereich 40-58,4V
Man könnte folgendes Vergeben: unter 48V auf einen Limit von XY Watt einstellen. Dann wird der Akku nicht unnötig mit voller Leistung belastet, wenn er schon im unteren SoC Bereich ist. Heute erst getestet, 50V Leerlaufspannung, bei 600W sind es nur noch 48V). D.h. man müsste Spannungslimit für die Freigabe der voller Leistung auch mit angeben, da bei kleineren Leistung die Spannung wieder nach oben geht, sofort die volle Leistung gezogen wird, dann wieder Limit greift weil die Spannung wieder fällt usw. Da könnte man z.B. +2V für Freigabe angeben. 

Zusätzlich/Optional/NiceToHave könnte man sagen: ab 58,4V (100% SoC) auf volle Leistung gehen -> wenn Akku voll, das Haus "zu wenig" verbraucht, und PV viel Produziert, kann man die überschüssige Energie doch ins Netz schenken. Auch hier, sobald die Akkuspannung fallen wird, nur auf das Hausverbrauch reduzieren (in dem Moment sind wir bei negativer Leistung am Messpunkt, z.B. -450W, da müsste man dazu 600W von WR addieren. So haben wir den Hausverbrauch von 150W und auf die 150W setzen wir den WR wieder. Ich werde das bei mir aus dem Victron auslesen (und nur diese Leistung ins Netz senden, dann ist der Akku weniger belastet und man verliert wirklich keine Energie)

Zur HW: ich werde so anfangen: https://savjee.be/blog/Home-Energy-Monitor-ESP32-CT-Sensor-Emonlib/
für ESP8266 Nutzer (aber auch ESP32 falls jemand möchte) könnte man auf ADS1115 über i2c setzen. Da muss man aber suchen wie es genau geht.

Ollipop030, 2023-01-21:
Die Berechnung des Powerlimits dürfte gar nicht so schwer sein, wir müssen nur ein paar Werte kennen: Derzeitiger Netzbezug, derzeitige Eispeisung, derzeitiges Powerlimit.

Das zu setzende Powerlimit berechnet sich dann wie folgt:
WENN [Netzbezug] größer als 1 DANN [neues Powerlimit] = [Netzbezug], SONST [derzeitiges Powerlimit] - [derzeitige Einspeisung]

Wenn also Netzbezug 300W ist, wird das Limit auf 300W gesetzt. Wenn Netzbezug 0 ist, aber Einspeisung bei 100W werden vom Powerlimit genau die 100W abgezogen.

Allerdings ist bei der nächsten Prüfung der Netzbezug bei 0W, somit wäre dann das Powerlimit bei 0W, dass müsste man noch korrigieren.

Die Prüfung würde dann alle paar Sekunden gemacht werden. Ist nur ein grobe Formel, hier ist kein Puffer berücksichtigt. Was man auch noch kennen müsste ist die maximale Leistung des Wechselrichters. Es macht nämlich keinen Sinn einem 600W Wechselrichter ein Powerlimit von 2000W zu senden.

Adminius, 2023-01-21:
Klar, das ist kein rocket science.
Mal eine andere Idee: wenn wir keine Bezugsleistung kennen, aber den Strom (mein ADC Fall mit CT Sensoren), dann können wir das selbe mit Strom rechnen und müssen nicht eine Fixe Spannung annehmen:
Strom vom WR kennen wir ja, WR Spannung auch und die Leistung. D.h. wir müssen ausrechnen wieviel Strom man haben müsste und dann die Aktuell WR AC Volatge nehmen, mit dem Wunschstrom multiplizieren und so hat man gewünschte Limit-Leistung.
Ich glaube das ist noch genauer als mit Leistungen und einer Fixen Spannung zu arbeiten.
Weil im Grunde, ziehen wir aus dem Netz ja Strom und nicht die Leistung. Leistung ist nur eine berechnte Größe

Kann mir jemand sagen an welcher Stelle ich im Ahoycode die nicht persistente Wunschleistung setzten kann (ohne Webui oder MQTT)

lumapu, 2023-01-22:
hier:
https://github.com/lumapu/ahoy/blob/main/src/app.cpp#L218

stefan123t, 2023-01-22:
Power Limit Regelung per MQTT auf einem separaten ESP in https://github.com/lumapu/ahoy/pull/618

Adminius, 2023-01-24:
ich habe gestern ein paar Tests mit dem Akku durchgeführt:
mit 100% Limit Akku entladen bis BMS abschaltet.
Und jetzt das Problem: Zellen erholen sich ohne Last -> Spannung steigt wieder -> BMS schaltet frei -> Hoymiles geht für paar Sekunden an und versucht sofort die volle Leistung zu ziehen. Jegliche Versuche das Limit über Ahoy zu setzen schlagen fehl, da WR nur wenige Sekunden an ist bis sich BMS wieder abschaltet... 
D.h. man muss unbedingt ab einer bestimmter DC-Spannung automatisch die Leistung reduzieren, so dass der WR bei einem Neustart der Batterie diese nur mit ganz wenig Leistung belastet.

stefan123t, 2023-01-24:
@Adminius dafür gibt es Softstart Schaltungen damit die Kondensatoren im WR langsam vorgeladen werden. Siehe die gepinnten Nachrichten ua. im Kanal #ess-speichersysteme im Discord.
Ausserdem kann man ein permanentes PowerLimit setzen das bleibt auch "am nächsten Tag" aktiv. Nur eben nicht ständig dieses Limit anpassen sonst ist der WR geflasht. Das temporäre Limit merkt er sich nur im RAM das ist für eine Steuerung besser geeignet!

Adminius, 2023-01-25:
es ging nicht um Kondensator Problem sondern wirklich darum, dass der WR versucht mit vollen 600W (nach dem Start) einzuspeisen, da bricht die Spannung im Akku wieder ein und BMS schaltet ab. 
Softstart brauche ich eher wegen Victron MPPT... der zieht wirklich viel beim anschließen. Da gibt es aber einfache Lösung: Victron muss zuerst PV Spannung haben und dann Akku anschließen

stefan123t, 2023-01-25:
@Adminius setz doch das permanente Limit auf 100W oder gleich 0W dann speist er morgens erstmal nichts ein.
Und die Regelung machst du dann mit dem temporären Limit das Du dann einfach per API / MQTT steuerst.
So machen es alle bisher :smile_cat:

Adminius, 2023-01-28:
Ich bin einen Schritt weiter: so richtig empfehlen kann ich die ADC Lösung doch nicht :/
Die Stromklemmen haben im unterem Messbereich, wo wir uns eigentlich befinden nicht linearen Verlauf... d.h. bei kleineren Strömen (z.B in der Nacht) sind Abweichungen nicht zu vermeiden.
Wenn ich mit 70W Halogenbirne meine ADC Ergebnisse kalibriere, bin ich bei 20W Birne nur bei 15W ADC bzw 585W Wasserkocher bei ca. 547W ADC.
So richtig 0 Einspeisung ist es nicht, mein Akku-Fall kann es aber schon gut abdecken.

Dazu kommt noch, dass emon Lib alles blockt bis die ADC Ergebnisse da sind.
Ich bekomme pro Phase ca. Alle 300ms einen Wert (wegen AC werden 1600+ Messungen gemacht). In diesen 300ms geht nichts mehr. Ob das die Kommunikation stört  habe ich nicht getestet.
Somit werde ich wohl doch auf 2x getrennte ESPs setzten und über WLAN/http beide verbinden.

Für eine Richtige abulleinspeisung empfehle ich aber einen Zähler den man auslesen kann.

Soll och issue schließen? Oder erstmal weiter berichten? (um HTTP zu vermeiden hätte ich gerne eine physikalische UART Schnittstelle von SteuerESP zum Ahoy 🤣 )

Adminius, 2023-01-30:
ich "blogge" mal weiter: 
ich habe die erste Version eines Scripts, der die Leistung des WRs setzt je nach Hauslast. (aktuell alles über MQTT und die Hauslast nehme ich von einem SmartPi der die Daten über MQTT zur Verfügung stellt. SmartPI wird aber am Einsatzort nicht verfügbar sein, deswegen wollte ich ja ADC Lösung)

Was ich bis jetzt herausgefunden habe:
- WR lese ich alle 5 Sekunden aus. Das ist schnell genug.
- Nach dem Setzen neuer Leistung sendet der WR ca. 15-20s keine Daten
- Leistungsänderung merkt man ca. 3-4 Sekunden nach dem Setzen.
- Wenn man eine "negative" (=WR liefert zu viel) Leistung zur Verfügung hat, kann man viel einfacher neuen WR-Limit berechnen. 
=> ich bin kurz davon mir einen SDM72DM/DRT428M-2 zu kaufen und auf die ADC Spielereien zu verzichten... Die 50-60 Euro sind mir mehr Wert als der Ärger mit CT+ADC (CT kosten auch schon 15 Euro für 3 Stück...) #431

lumapu, 2023-01-30:
ja die SDM Zähler sind schön, ich habe den SDM630, der liefert über Modbus die Daten bisher nur zu meinem ioBroker 😊

Adminius, 2023-01-30:
das Problem ist nicht der Zähler selbst, sondern deren Einbau in die Verteilung... deswegen wollte ich ja auf die CT-Sensoren setzen, die man einfach um die Leitung klemmen kann. Das schafft auch der Leie... in meinem Fall muss ich nicht nur einen Stromzähler einbauen, sondern auch noch Abzweigklemmen schieben, zusätzliche dazu fügen, Verkabelung ändern usw., und ich habe noch glück dafür genug Platz zu haben... und ich weiß schon was ich tue... 
D.h. eigentlich kommen zum 57 Euro Zähler noch 10 Euro Abzweigklemmen und ein paar 10mm² Drähte dazu, und schon wird es "nicht wirtschaftlich" und viel weiter weg von 15 Euro CT-Klemmen entfernt. Aber mit weniger Experimenten=Zeiteinsatz verbunden... ich bin hin und her gerissen :/

Adminius, 2023-01-31:
ich erkläre ADC Versuch als gescheitert.
Nach mehreren SW/HW Tunings bin ich nicht mal zu einem sauberen 0 bei den Messungen gekommen. Da hat schon bei "CT-Sensoren auf dem Tisch" zu "Phantommessungen" von bis zu 28W geführt. 
SDM72 ist bestellt... ich mache dann doch mit RS485 weiter.
Issue mache ich zu. schade...

closed 2023-01-31 by Adminius

stefan123t, 2023-01-31:
@Adminius danke fürs mitbloggen und -spielen. Es geht dann also in #431 weiter.