lumapu/ahoy #606: Tasmota Stromzähler Support in Ahoy

gandalfred, 2023-01-20:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen,
Ich besitze einen Hichi Stromzähler im Stromkasten der über Tasmota den aktuellen Stand der Stromzähler 1.8.0 und 2.8.0 per WLAN auf einer separaten IP darstellen und ausgeben kann. Es wäre toll, wenn man im Ahoy die IP Adresse des Hichi Tasmota Stromzählers eintragen könnte und die Zählerstände 1.8.0 und 2.8.0 sowie der Momentanverbrauch würden im ahoy Web IF angezeigt ohne dass man sich mühsam einen ioBroker aufsetzen muss. So hätte man auf einen Blick alle Strom Verbrauchs-und Erzeugungswerte auf einen Blick in einer Anzeige.
In einem weiteren Schritt könnte man so dann auch eine einfache Nulleinspeisung im Ahoy realisieren indem man den Momentanverbrauch minus momentan  Erzeugter Strom rechnet und wenn das negativ ist man den WR entsprechend drosselt um Nulleinspeisung zu realisieren. Der erste Punkt wäre mir aber wichtiger alle Werte an einer Stelle auf einen Blick zu haben. Was meint ihr?
Gruss Steve

dtuuser, 2023-01-20:
Hallo Steve,
habe ich auch so am Laufen. Ich frage die Werte über ein eigenes Tool per Webaufruf und Analyse der Webseite ab.
Klappt auch vom Ahoy. Aktuell baue ich an der Logik, wie man das Limit berechnet. 
Es ist ja nicht damit getan, wenn  der  aktuelle Verbrauch ein Minuswert ist, dann setze das Limit am WR auf WRLimitWert minus Verbrauchswert. Sobald der aktuelle Verbrauch wieder im Plus ist, dann muss der WRLimitWert um den Pluswert erhöht werden. Und das WR Limit kann ja über einen gewissen Wert nicht hinaus.
Sobald die Berechnung bei mir plausibel läuft, kann ich das hier posten.

tqma1, 2023-01-20:
2.8.0 ist die gesamte Einspeisung
2.7.0 ist die aktuelle Einspeisung - wäre cool hierfür ein Limit setzen zu können z.Bsp. max 150W

stefan123t, 2023-01-20:
Klingt nach einem sehr guten Vorschlag. Welche Informationen braucht es denn um den Hichi ESP direkt dirch die AhoyDTU abfragen zu können ?
Da müsste doch im Prinzip der Endpoint als URL genügen bzw. vielleicht noch der Name des gesuchten Wertes den kann man ja im Hichi/ Tasmota Script beliebig benennen.
Wegen der Null oder Maximal-Einspeisung siehe auch #605

dtuuser, 2023-01-20:
Aufruf    http://<IPvomHichi>/cm?cmnd=status%208
bringt das als Rückmeldung
{"Verbrauch_Summe":1234.56,"Einspeisung_Summe":123.45678,"Watt_Summe":620,"Watt_L1":373.66,"Watt_L2":141.66,"Watt_L3":104.23,"Volt_L1":237.8,"Volt_L2":237.2,"Volt_L3":236.2}}}

Watt_Summe  ist der aktuelle Verbrauch. Bei Einspeisung als Minuswert

MubiTec, 2023-01-20:
Echt coole Idee.
Mit http://192.168.178.94/cm?cmnd=status%2010
Bekommt man alle Daten als json:
{"StatusSNS":{"Time":"2023-01-20T18:14:40","Haus":{"Total_in":9935.7223,"Total_out":11022.9129,"Power_curr":444,"Volt_p1":231.4,"Volt_p2":230.6,"Volt_p3":232.4,"Amperage_p1":0.9,"Amperage_p2":2.4,"Amperage_p3":0.5,"phase_angle_p1":-63.0,"phase_angle_p2":-51.0,"phase_angle_p3":-71.0,"frequency":51}}}

dtuuser, 2023-01-20:
Unter  Windows habe ich ein eigenes Tool geschrieben, was die Daten vom Hichi und von Ahoy abrufen.
Nur beim Limit setzen, da muss die genaue Logik noch her. Aber das Setzen vom Limit klappt schon über Curl  #586

tqma1, 2023-01-20:
Bitte bei der Abfrage flexibel bleiben da Tasmota nicht die alleinige Lösung sein soll, da es die Entschlüsselung der Daten nicht kann ...
Bei mir sieht es so aus (RestAPI):
http://192.168.1.8/getLastData?user=xxxxxxx&password=yyyyy
Ergebnis:
{
"1.8.0":"17708776",
"2.8.0":"2",
"3.8.1":"59644",
"4.8.1":"4552041",
"1.7.0":"237",
"2.7.0":"0",
"3.7.0":"0",
"4.7.0":"158",
"1.128.0":"0",
"uptime":"0000:00:09:13"
}

Ollipop030, 2023-01-20:
Ich habe einen emlog Lesekopf, wie in #604 schon beschrieben. Hier kommt beim Zugriff auf http://192.168.0.18/pages/getinformation.php?heute&meterindex=1" das hier raus:

{"Leistung170":475,"Leistung171":0,"Leistung172":0,"Leistung173":0,"Leistung270":0,"Leistung271":0,"Leistung272":0,"Leistung273":0,"Waehrung":"EUR","Stand300":null,"M300":0,"Stand180":0,"Stand181":0,"Stand182":0,"Stand280":0,"Stand281":0,"Stand282":0,"Kwh180":0,"Kwh181":0,"Kwh182":0,"Kwh280":0,"Kwh281":0,"Kwh282":0,"Kwh300":0,"Betrag180":0,"Betrag181":0,"Betrag182":0,"Betrag280":0,"Betrag281":0,"Betrag282":0,"Betrag300":0,"Metertype":"Strom"}

Wäre es dann bei der Anzahl an unterschiedlicher Hardware nicht am einfachsten, in der DTU die Adresse mit den json Daten anzugeben, und jeweils den relevanten Wert für 2.7.0 und 2.8.0?

Adminius, 2023-01-20:
Hi @dtuuser mag du schon deinen zwischenstand teilen. Dann muss ich das Rad nicht neu entwickelten.
Ich habe ähnliches vor nur mit anderem "Zähler" #605

dtuuser, 2023-01-21:
Hallo Adminius,
Werte von beiden Geräten kann ich abrufen und gehen in die Berechnung.
Aktuell teste ich mit dem Grenzwert von 400Watt, da ich seit November keine Einspeisung bekomme.
Gestern habe ich laut meiner Abfrage keine Werte unterschritten, was eigentlich nicht sein kann.
Ich kann meine Abfrage später hier posten. Vielleicht habe ich etwas übersehen.

gandalfred, 2023-01-21:
Servus Jungs, cool dass es hier soviele Rückmeldungen gibt und ihr schon an entsprechenden Lösungen bastelt. Da es verschiedene Zähler gibt die unterschiedliche Werte ausgeben können wäre es super wenn das generisch berücksichtigt werden könnte. Mein Stromzähler kann zb nur 1.8.0 aktueller Zählerstand Bezug , 2.8.0 bisheriger eingespeiste Leistung und den Momantanverbrauch aufsaldiert anzeigen Er gibt bei mir aber nicht die einzelnen Phasenleistungen an den hichi mit aus. Zähler ist ein easymeter q3b.

dtuuser, 2023-01-21:
Hallo Adminius,
bitte nicht erschlagen..... aber ich bin Oldschool..daher VB Code
Die 700 sind für einen HM-700

z = aktuellerverbrauch   'kommt vom Zähler über Hichi
wr2 = einspeisungwr2   'kommt vom Ahoy 
If z < 0 Then                  'wenn aktueller Verbrauch < 0...also EInspeisung
   If wr2 + z > 10 Then   'wenn erzeugte Leiszung + (negativen Zählerwert) >10   die 10 habe ich als Minimum Limit 
   limit_wr2 = wr2 + z     ' dann Limit vom WR = aktuelle Erzeugung + (negativem Zählerwert) 
   Else                              ' falls WR2 + (negativem Zählerwert) <10 
   limit_wr2 = 10             ' dann Limit WR auf 10
   End If
End If
                                                                     ' ab hier, falls ein Limit am WR eingestellt wurde
  If wr2 > z And limit_wr2_set < 700 Then     ' aktuelle Erzeugung > aktuellen Verbrauch und das Limit am WR <700        
     limit_wr2 = wr2 + z                                 ' Limit WR = aktuelle Erzeugung + aktuellen Verbrauch
     If limit_wr2 > 700 Then limit_wr2 = 700   ' falls Limit WR > 700 dann Limit WR auf 700 setzen
  End If

If limit_wr2_set <> limit_wr2 And limit_wr2_set < 700 Then        ' Prüfung, ob Limit <700 und es einen Unterschied zur vorherigen Prüfund gibt
limit_wr2_set = Round(limit_wr2, 0)    ' falls ja, dann neues Limit auf den berechneten Wert

danach die Ausgabezeile mit dem neuen Limit_Wr2_set Wert erzeigen und in der Shell starten

stefan123t, 2023-01-22:
Power Limit Regelung per MQTT auf einem separaten ESP in #618

tqma1, 2023-01-22:
Na ja, da bin ich persönlich nicht der größte Fan von …

Cool wäre doch folgendes Szenario:
Im HM-xxxx ein persistentes Limit hinterlegt (z.BSP. HM 1500 / Limit 600 Watt)
=> I h habe mein Balkonkraftwerk gemeldet und Speise in schlimmsten Fall 600 Watt ein.

Funktioniert ohne irgend welche Abhängigkeiten immer!

Jetzt die coole Ahoy Nummer:
Wir passenden Richtung Nulleinspeisung das Limit an.
Muss all3s über einen Controller laufen, da wir ansonsten wieder falsch ausgeregelte Zuständehaben ….

stefan123t, 2023-01-23:
Also wir brauchen die Endpoint URL (evtl. mit Basic Authentication Username + Password):
* http://ipvomhichi/cm?cmnd=status%208
* http://192.168.178.94/cm?cmnd=status%2010
* http://192.168.1.8/getLastData?user=xxxxxxx&password=yyyyy
* http://192.168.0.18/pages/getinformation.php?heute&meterindex=1

Und dann sollen wir zwei Werte auslesen:
 1.7.0    aktuelle Wirkleistung P+ in Watt (Bezug)

Ollipop030, 2023-01-23:
Was aber auch berücksichtigt werden sollte sind Anlagen bestehend aus mehreren Wechselrichtern. Hier wird die Berechnung des Limits komplizierter. Vielleicht kann man dann auf Basis der Modulleistung ein Limit berechnen, wobei dann die Verlustleisung nicht einberechnet wird. Oder man verteilt die Leistung über die Anzahl der Module. Beispiel:
1 Modul am HM300, und vier am HM1500. Jetzt werden 250 Watt benötigt, der HM300 lässt 50 Watt durch, und der HM1500 lässt 200 Watt durch.

Ollipop030, 2023-01-23:
Ja klingt gut. Und auf diesem Wert wird dann noch der Wert für die Einspeisung gerechnet, wenn denn jmd einspeisen möchte. Ich selbst würde vielleicht immer 100Watt einspeisen, nur als Puffer. 600Watt sind angemeldet, also sagt auch der Netzbetreiber nichts wenn im Jahr bei 2.8.0 ein paar kWh dazukommen. Mit Modulausrichtungen würde das ganze gigantische Ausmaße annehmen. Dann müsste mach auch Aufstellwinkel, Modulart und Verschattungen einbeziehen.

gandalfred, 2023-01-23:
Bräuchten wir nicht auch noch 1.7.0 aktueller Momentanverbrauch power consumption ?
1.8.0 wäre der aktuelle Zählerstand Bezug bzw Verbrauch
2.8.0 wäre der aktuelle Zählerstand Einspeisung

Ollipop030, 2023-01-23:
> Bräuchten wir nicht auch noch 1.7.0 aktueller Momentanverbrauch power consumption ?

Ja das stimmt. Hatte stefan123t ja geschrieben.

dtuuser, 2023-01-23:
> Bräuchten wir nicht auch noch 1.7.0 aktueller Momentanverbrauch power consumption ? 1.8.0 wäre der aktuelle Zählerstand Bezug bzw Verbrauch 2.8.0 wäre der aktuelle Zählerstand Einspeisung

Eigentlich reicht doch der momentane Verbrauch, da dieser Wert bei Einspeisung doch negativ wird. Oder ist das zählerabhängig?

Adminius, 2023-01-23:
Mein Zähler sagt dann 0 Verbrauch und wird nicht negativ.
D.h. man braucht am besten Einspeiseleistung und Verbrauchsleistung oder muss sich anhand von Verbrauchsleistung an die 0 rantasten...

Ollipop030, 2023-01-23:
Das ist tatsächlich Zählerabhängig. Bei mir geht der Bezug (1.7.0) auf Null, gleichzeitig steigt aber die Einspeisung (2.7.0). Es gibt Zähler, die zeigen dann minus. Einige Zähler zeigen sogar Leistungen auf den einzelnen Phasen an. 1.7.0 und 2.7.0 ist einfach der saldierte Wert der drei Phasen.

Adminius, 2023-01-23:
ich meine es ist sogar eindeutig definiert:
1.7.0 aktueller Verbrauch in kW (Wirkleistung) aus dem Netz
2.7.0 aktuelle Einspeisung in kW (Wirkleistung) in das Netz
d.h. die o.g. Werte sind immer positiv ("Rücklaufsperre"), aber nie gleichzeitig.

Genauso wie der Zählerstand immer positiv ist:
1.8.0 Verbrauch Summe in kWh (Wirkarbeit)
2.8.0 Einspeisung Summe in kWh (Wirkarbeit) in das Netz

Omega13x, 2023-01-23:
Hallo, ich habe mal ein Anliegen an die Entwickler. Ich habe erfolgreich drei funktionierende DTU´s zusammengebaut (Den Elko habe ich gehuldigt 😉 ). An meinem modernen Stromzähler habe ich ein Powerfox Poweropti-LED der mir die Zählerdaten auf das Smartphone schickt. Auf der Homepage von Powerfox habe ich gesehen das man die Daten über eine Kunden API abrufen kann. Ist es möglich diese Daten irgendwie für eine Einspeiseregelung mit der Ahoy-DTU zu nutzen? Ich selber habe von so was null Ahnung. Deswegen wäre eine Lösung ohne zusätzliche Hardware optimal.

Die aktuellen Daten lassen sich mit diesem Link anzeigen: "https://backend.powerfox.energy/api/2.0/my/main/current" in Watt oder mit diesem: "https://backend.powerfox.energy/api/2.0/my/main/current?unit=kwh" in KW auslesen.
Hier die Ausgabe:  {"Outdated":false,"Watt":242,"Timestamp":1674510379,"A_Plus":641566,"A_Minus":68938}

[image]

stefan123t, 2023-01-23:
@Omega13x die Anleitung für die API findet sich hier: https://www.powerfox.energy/wp-content/uploads/2020/05/powerfox-Kunden-API.pdf

Eventuell lässt es sich ja mit der selben Strategie wie die Tasmota Geräte einbinden ? Andererseits sind es closed source und hardware Geräte die hier folglich nicht an erster Stelle stehen.

Adminius, 2023-01-24:
@Omega13x größtes Problem bei online Schnittstellen sind die Abfrageintervalle.
Beispiel: SmartMeter sollen nur alle 15 Minuten die Zählerstände übermitteln -> Sinnlos für Nulleinspeisung, wo wir in 1-10S Bereich sind...

@stefan123t Eine Idee: man kann einen Scriptinterpreter (z.B. für Lua) integrieren. Dann kann man eigene Abfragen von eigenen APIs in eigenen Formaten auseinander nehmen und an die Bestimmte "Empfangspunkte" des Ahoy übergeben. 
Ich stelle mir es so vor:
Ahoy hat eine Funktion die aktuelle Bezugs- und Einspeiseleistung empfängt oder sogar nur eine Leistung positiv wie negativ. Nulleinspeiseberechnungen sind auch mit drin. D.h. der User soll nur diese Funktion aus eigenem Script oder aus eigenem Code in der Loopschleife aufrufen und mit Daten füttern.

Omega13x, 2023-01-24:
@Adminius 
Moin,
beim Powerfox werden die aktuellen Verbrauchs- und Einspeisedaten alle 5 Sekunden aktualisiert.

Adminius, 2023-01-24:
oh cool! D.h. man kann die auch alle 5 Sekunden abfragen, oder? Das würde dann passen.

Omega13x, 2023-01-24:
> oh cool! D.h. man kann die auch alle 5 Sekunden abfragen, oder? Das würde dann passen.

Ja, genau.

dtuuser, 2023-01-24:
Mein Hichi fragt jede Sekunde ab und stellt diese per Weboberfläche dar

Verbrauch_Summe	12345.25 kWh
Solareinspeisung_Summe	12.5502177 kWh
Aktueller_Verbrauch	515 W
--
Leistung_L1	348.65 W
Leistung_L2	119.57 W
Leistung_L3	46.53 W
--
Spannung_L1	238.1 V
Spannung_L2	238.4 V
Spannung_L3	237.8 V

pvstrom, 2023-01-24:
> es könnten auch HM300 = 250W und HM1500 = 0W sein. Evtl. ist es vom Wirkungsgrad sogar besser. Wie die Module gedreht sind ist aber eine andere (komplexe) Sache...
> 
> Einfachere Alternative wäre: Prozentual von der Leistung: 300+1500 = 1800W sind 100% Leistung. 250W wären dann 14%. => HM300 42W HM1500 210W

Wäre eine Priorisierung nicht einfacher?
Z.b. Nutzer wählt aus, dass HM300 zuerst gedrosselt wird, dann der HM-15000 und der HM-600 gar nicht.
Erst wenn wenn die Drosselung des HM300 nicht ausreicht wird auch der HM-1500 gedrosselt.

Im Optimalfall dann sogar noch mit einem Zeitplan zu Minimal- und Maximalwert der Drosselung und evtl. auch noch in Abhängigkeit der Eingangspannungen (DC-Seite).
Wenn z.b. der HM-300 an direkt an einer Batterie betrieben wird, macht es Sinn den zuerst den zu drosseln oder zeitlich über einen Zeitplan z.b. unter tags grundsätzlich auf 0% zu stellen und evtl. für Batteriebetrieb andere Limits zu setzen (PV max. +600 Batterie max -50W). Bei zu geringer Batteriespannung ware eine automatische Abschaltung (Limitierung auf 0%) und Anschaltung (wenn Akku voll) zudem auch nicht schlecht.

Ich habe jetzt schon mehrere gesehen, die ihre Hoymiles mit Pylontechs verwenden wollen, wäre daher evtl. nicht ganz unintressant.

Nur die Umsetzung mit dem Hichi Lesekopf dürfte etwas schwierig werden.
Hier liefert jeder Zähler andere Werte, die dann vom verwenden Script auch noch unterschiedlich benannt werden.
Bei mir (MT175 mit unverändertem Tasmota Script) ist die aktuelle saldierte Leistung "P:" und geht bei Einspeisung ins Negative.
Alternativ zusätzlich zum NRF2401L+ Modul noch ein UART to RS485 Modul für einen SDM630 zu verwenden dürfte hier evtl. einsteigerfreundlicher sein.

Adminius, 2023-01-25:
Statt SDM630 könnte man SDM72 nehmen. Export Power hat er auch, kostet aber nur die Hälfte. Bei der Lösung ist man aber aber an einen Elektriker gebunden. Sogar wenn man CT-Variante von dem Zähler nimmt, braucht man alle 3 Phasen um die Spannung zu kennen.

Das ist ja der Grund warum ich mit ESP32+ADC+CT-Sensoren experimentiere #605 . Die Klemmen um die Zuleitung wickeln, 5V Netzteil in die Steckdose stecken und fertig. Kein Elektriker, aber halt nicht so genau.

> Wäre eine Priorisierung nicht einfacher?

aus meiner Sicht nein. In meiner Lösung muss man nur % aus der Gesamleistung (diese muss man eingeben oder automatisch ermitteln) ausrechnen und diese % an die WRs direkt senden.
In deiner Lösung (auch wenn ich die bevorzugen würde) muss man viel einstellen und viel Logik schreiben. Man muss klein anfangen und dann ausbauen ;) Auch ich habe HM600 am Akku ;)

pvstrom, 2023-01-26:
SDM72 wäre natürlich auch eine Alternative.
Wäre jetzt mal davon ausgegangen dass entweder bereits wie oftmals bereits ein SDM630 vorhanden ist oder sich ein SDM630/SDM72 schnell montieren lässt ( WR-Leistung >600W sollte auch jemand mit Ahnung anschließen).
Damit wäre das Problem der unterschiedlichen Ausgabe mit den Leseköpfen umgangen.
Zumal rücklaufgesperrte Ferraris oftmals auch gar nicht getauscht werden.

> In deiner Lösung (auch wenn ich die bevorzugen würde) muss man viel einstellen und viel Logik schreiben

Viel Aufwand aber auch viel Gewinn.
Grob mal als erstes Hirngespinst ...
>Überschuss#0: Power Einspeisung – Power Einsspeiselimit#0 ( > Einspeiselimit BatterieWR z.b. -10W)
>Überschuss#1: Power Einspeisung – Reduzierung#0 – Power Einspeiselimit#1 (> Einspeiselimit PVWR z.b. +600W)
>Überschuss#2: Power Einspeisung – Reduzierung#0 – Reduzierung #1 – Power Einspeiselimit#1
>
>
>Wenn Überschuss#0 > 10 dann
>Reduzierung#0 WR#0 um (Überschuss#0) oder wenn (Überschuss#0 >WR#0Pac dann Reduzierung um WR#0Pac)
>Wenn Überschuss#1 > 10 dann 
>Reduzierung WR#1 um (Überschuss#1) oder wenn (Überschuss#1 > WR#1Pac dann Reduzierung um WR#1Pac
>Wenn Überschuss#2 > 10 dann
>Reduzierung WR#2 um (Überschuss#2) oder wenn (Überschuss#2 > WR#2Pac dann Reduzierung um WR#2Pac
>
>Unterschuss#PV: Netzbezug + Einspeiselimit#1
>Unterschuss#Batt: Netzbezug + Einspeiselimit#0 
>
>Wenn Unterschuss#PV > 1 dann
>LeistungserhöhungWR#2 um (Unterschuss#PV) oder wenn (Unterschuss#PV > WR#2maxP dann Leistungserhöhung auf >WR#2maxP
>Wenn Unterschuss#PVaktualisiert > 1 dann
>LeistungserhöhungWR#1 um (Unterschuss#PVaktualisiert) oder wenn (Unterschuss#PVaktualisiert > WR#1maxP dann >Leistungserhöhung auf WR#1maxP)
>
>Wenn Unterschuss#Batt > 1 über Hysterese X
>LeistungserhöhungWR#0 um (Unterschuss#Batt) oder wenn (Unterschuss#Batt> WR#0Limit dann Leistungserhöhung auf >WR#0Limit

Mal schauen, da lässt sich evtl. #618 anpassen.
Oder ich probiers mal logosoftmäßig auf NodeRed.
Oder vll. wirds einfach ein faules fixes Limit :)

> Auch ich habe HM600 am Akku ;)
Soweit bin ich noch gar nicht.
Meine hängen derzeit noch an den Modulen, da ich eig. mit geregeltem Ladegerät>Akku>HM-300 arbeiten wollte.

Adminius, 2023-01-26:
so oder so brauchen wir offizielle Stelle im Code wo man eigene Zeller und die Abfragen dazu implementieren kann.
Die Stelle sollte aktuelle Daten vom Wechselrichter liefern, aber auch Limit-Wert und Limit-Typ empfangen und setzen können.
alles andere dürfen die Nutzer selbst implementieren, egal ob SDM72/630, Shelly oder was auch immer. 
@stefan123t so wie es ausschaut, bekommt das Thema langsam höhere Prio. was denkst?

stefan123t, 2023-01-27:
Die Abfrage des Tasmota ist eigentlich relativ einfach. Man braucht nur die richtigen Topics aus der JSON Antwort selektieren. Dafür kann man neben der URL noch zwei Topics für Einspeiseleistung und Bezugsleistung vorsehen. Analog dazu könnte man auch die selben Daten per MQTT vorerst beim Mosquitto Broker subscribe und so indirekt beziehen.
Die Logik / Berechnung für das zu setzende PowerLimit sollte man sich evtl nochmal überlegen. Dass man manche WR (die ihren Strom von einem Akku beziehen) in eine eigene "Prio"-Gruppe packt die erst dann eingeschaltet / hochgefahren wird wenn der Bedarf durch die andere WR-Gruppe nicht mehr gedeckt wird (zB Nachts) wäre ja einfach und logisch.
Das andere ist die Entscheidung wieviel jeder WR einer Gruppe einspeisen darf. Hier sollte jeder WR das Maximum einspeisen dürfen bis das Gesamtlimit (0W/600W/800W/Unlimited) erreicht wird. Wie anderweitig schon besprochen regelt die Hoymiles Software das in 30W Schritten jeweils rauf / runter bis das Optimum erreicht wird. Dabei sind zB auch Ost/West Anlagen o.a. durch Beschattung, Bewölkung / Ausrichtung zeitweise indirekt limitierte WR zu berücksichtigen.

Der Vorschlag das in NodeRed zu machen wurde mW bereits umgesetzt dazu bitte mal im Discord unter #ess-speichersysteme nachlesen.

Hier geht es also vorrangig um den einfachen Fall die Einspeise- und Verbrauchsdaten per REST API URL und zwei JSON Elementen bzw deren JPath zu ermitteln und ggf so das Limit für einen (oder auch für mehrere) WR zu limitieren.

Das Thema der Lilitierung von mehreren WR würde ich sogar erstmal zurückstellen und hier nur den einfachsten Fall angehen.

Für die Modbus RS485 Schnittstelle existiert ebenso ein anderes Issue.

zanycam, 2023-02-06:
Hi, hier hat jemand gute Ideen für die Umsetzung der Steuerung für eine Nulleinspeisung über mehrere WR (mittels MQTT) dargelegt, ist davon was vielleicht hier mit brauchbar als Umsetzungsgrundlage?

VG, Sandy

zanycam, 2023-02-06:
> 

Edit: Link vergessen >•<

https://github.com/tbnobody/OpenDTU/discussions/406#discussion-4625886

reserve85, 2023-02-08:
Da ich hier gerade drübergestolpert bin: ich habe die Nulleinspeisung mit Tasmota und Ahoy relativ einfach über die WebAPI realisiert. 
Funktioniert bei mir wunderbar, mein aktueller Stromverbrauchszähler wird allerdings negativ wenn ich einspeise (P=-123W). Evtl kann jemand damit was anfangen: https://github.com/reserve85/HoymilesZeroExport
Ich habe das Script auf einem Raspi laufen.

Ollipop030, 2023-02-08:
Schön zu sehen, das man es mit einem so einfach Script hinbekommt. Wenn es jetzt noch in Ahoy eingebaut wird, und ein wenig variabel gehalten wird (Eingabemaske mit URL, Variablenamen) sind wir ja einen großen Schritt weiter. 

Die Maximalleistung des WR könnte man noch über die Seriennummer des WR ermitteln.

dtuuser, 2023-02-09:
Kurze Rückmeldung. Meinen WR steuere ich seit ein paar Tagen über ein eigenes Programm.  Zwei weitere Programme "sammeln" die Daten von der AhoyDTU bzw. vom Tasmota Zähler und speichern die in meiner Registry zwischen.
Mit der folgenden Abfrage klappt das recht gut. Heute nur 20W "verschenkt", da ich noch 5 Minuten zwischen den Curl Aufrufen warte.

'Z = Aktueller Zählerverbrauch. Bei meinem Zähler bei Einspeisung -123
'WR = aktuelle Erzeugung
'limit_wr_set = zu letzt gesetztes Limit
'limit_wr = berechnet Limit

If z < 0 And limit_wr_set > 10 Then
limit_wr = wr + z
If limit_wr < 10 Then limit_wr = 10
End If

If z > 0 And limit_wr_set < 600 And wr + 20 > limit_wr_set Then
limit_wr = wr + z
If limit_wr > 600 Then limit_wr = 600
End If

'30 Minuten vor Sonnenuntergang das Limit auf 600W setzen
If aktuelletagesminuten > Val((sonnenunterganginminuten) - 30) Then
limit_wr = 600
End If

If limit_wr_set <> limit_wr And limit_wr_set <= (600) Then
Limit mit "curl -i -X POST -H .....limit_nonpersistent_absolute.... & limit_wr2_set" an ahoyDTU schicken

Ollipop030, 2023-02-12:
Mal eine Frage zu deinem Programm: Worauf laufen die Programme bei dir? Raspi? Windows Maschine? Vielleicht könntest du uns ja daran Teil haben lassen.

dtuuser, 2023-02-12:
Laufen auf einem 24/7 Windows Rechner und sind in Oldschool VB6 programmiert

Ollipop030, 2023-02-12:
Würdest du mir/uns das zur Verfügung stellen? Falls nicht versteh ich das, ist ja deins.

reserve85, 2023-02-12:
Wäre auch froh wenn mein Script noch jemand testen könnte. Benötigt halt installiertes Python

dtuuser, 2023-02-12:
Ein Programm liest vom Tasmota per Webseitenaufruf den aktuellen Stromverbrauch (bei mir bei Einspeisung Minuswert) und speichert den in der Registry.
Ein weiteres Programm liest vom Ahoy per Webseitenaufruf die Einspeisung aus und speichert den aktuellen Wert in der Registry.
Auf beide Registry-Werte greift das 3.Programm zu und berechnet über einen Timer den obigen Quellcode.
Z = Registry Wert für den aktuellen Verbrauch, WR = Registry Wert die aktuelle Erzeugung
Daraus berechnet sich das zu setzende Limit, welches vor dem CURL Aufruf noch auf eine Ungleichheit zum letzten gesetzten Limit Wert geprüft wird. Der Timer wird nach einem CURL Aufruf 5 Minuten pausiert, da die Umstellung vom Limit auf dem WR bei mir meistens erst nach 3 Minuten umgesetzt ist.

Laut Herrn Habeck, sollen wir ja angeblich zukünftig jedes eingespeiste KW vom Verbrauch abgezogen bekommen, demnach entfällt irgendwann die Nulleinspeisung. Und einen Akku braucht man auch nicht mehr ;-)

reserve85, 2023-02-12:
Aber der Hoymiles reagiert doch direkt auf das Limit-Kommando. Das dauert bei meinen Beobachtungen gefühlt keine 3 Sekunden bis er das neue Limit eingestellt hat. Mit einem 5-Minuten Timer klappt das doch nicht, was passiert wenn du ein Verbraucher anschaltest? Dann ist im schlechtesten Fall der Hoymiles noch weitere 5 Minuten limitiert obwohl du vom Netzbetreiber beziehst?

dtuuser, 2023-02-12:
Ich frage alle 60 Sekunden die WR Werte mit Ahoy ab. Im ungünstigsten Fall wird der neue Limit Wert 59 Sekunden vor der nächsten Ahoy Kommunikation an die AhoyDtu geschickt. Dann warte ich auf das Feedback vom WR, ob das Limit gesetzt wurde. 
So komme ich auf bis zu 3 Minuten. Wenn ich das Ahoy Intervall auf 15 Sekunden setze, dann könnte ich auf eine Minute kommen.
Mir ist aktuell wichtig, dass ich nicht zu viel einspeise. Habe im letzten Jahr seit Ende Juni 270KW produziert und 20 davon eingespeist.

reserve85, 2023-02-12:
Interessanter Gedanke, mir wäre es allerdings wichtiger das Maximum an der Eigennutzung rauszuholen anstelle möglichst nix einzuspeisen. Die 20kwh sind doch Peanuts?

dtuuser, 2023-02-12:
Problem sind meine Nachbarn und eine mögliche Verbindung zum Netzbetreiber. Da habe ich lieber so wenig Einspeisung wie möglich.
Im Sommer werden anstatt der Limitierung vom WR dann sowieso die Poolpumpe oder die Klimaanlage gesteuert.

Ollipop030, 2023-02-20:
Ich lasse bei mir im Moment das Script von reserve85 laufen: 
https://github.com/reserve85/HoymilesZeroExport

Mit einigen Anpassungen läuft das super. Danke auch nochmal an @reserve85 für deine Hilfe. Im Moment läuft das auf einem Raspi, ist mir aber ehrlich gesagt zuviel an Geräten (Lesekopf, DTU. Raspi), für so eine einfach Aufgabe.
@lumapu Könnte man das so oder ähnlich nicht in die DTU integegrieren? Oder ein Fenster, in dem man einfach sein eigenes python Script einfügen kann?

Ollipop030, 2023-02-20:
Danke, erfordert aber MQTT, und Open DTU geht nur auf dem ESP32. Es soll ja möglichst einfach sein, Tasmota an den Zähler kleben, Ahoy einrichten und fertig.

Adminius, 2023-02-20:
na gut, ob ESP32 oder ESP8266 ist relativ egal, kosten ähnlich um 5 Euro in china.
ich habe SDM72/630 Support eingebaut (per RS485), wird bald in dieses Branch gemerged (kannst schon mal den PR anschauen).
Danach kommt HTTP/JSON Schnittstelle für Zähler (da habe ich halt nichts daheim, deswegen müsste ich "blind" entwickeln)

Adminius, 2023-02-21:
Hi @Ollipop030 nutzt du Shelly's original FW oder Tasmota?
Hast du für mich Infos wie man an die Daten kommt und einen Response? ;)

Ollipop030, 2023-02-21:
Habe kein Shelly, ich habe einfach einen optischen Lesekopf (ist aber auch kein Tasmota). Aber lt. den Posts weiter oben scheinen die verschiedenen Leseköpfe alle json Daten auszuwerfen. Die Domäne und die Topics unterscheiden sich allerdings immer etwas.

reserve85, 2023-02-21:
bei Shelly kommst du auch mit einer WebAPI ganz easy drauf.
siehe https://shelly-api-docs.shelly.cloud/gen1/#shelly-3em-emeter-index

Adminius, 2023-02-21:
@reserve85 danke. also für 3 Phasen 3 abfragen... hm.

reserve85, 2023-02-21:
ggf. geht auch das hier: https://shelly-api-docs.shelly.cloud/gen1/#shelly-3em-status
da steht `total_power | number | Sum of the power of the three channels, Watts`

Adminius, 2023-02-21:
ich habe keine Shelly, sonst müsste ich nicht fragen ;)

dtuuser, 2023-02-21:
Ich hatte einen normalen Shelly vor dem WR. Original Firmware drauf...und alle zwei bis drei Wochen Spannungswerte jenseits von 255 Volt. Mit Reset.
Diesen hatte ich per API Aufruf abgefragt. 
Sollte bei den wirklichen Verbrauchs-Shellys auch gehen. Mir war damals die Anschaffung von einem 3-em zu teuer. Daher 33 Eueo Tasmota WiFi Kopf am Zähler und immer die richtigen Werte.

Ollipop030, 2023-02-21:
> Daher 33 Eueo Tasmota WiFi Kopf am Zähler und immer die richtigen Werte.

Und viel einfacher für Laien.

[19 more replies]