lumapu/ahoy #607: 0.5.74 lässt kein Power Limit zu

Ollipop030, 2023-01-20:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.74

### Github Hash

76B5818

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1++++++++
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Hi,

0.5.74 lässt keine Änderung am Power Limit zu, es kommt folgende Meldung:

Ctrl result: Error: inverter does not accept dev control request at this moment

Auch wird in der Live Ansicht das aktuelle Limit nicht angezeigt. Wechselrichter ist aber verfügbar und produziert.

Juergen2453, 2023-01-20:
Ja kann ich bestätigen, bei mir das selbe.

beegee3, 2023-01-20:
@lumapu wenn man in `hmInverter.h`, `getQueuedCmd` die Zeile `if ((actPowerLimit == 0xffff) && isConnected)` durch 
`if (actPowerLimit == 0xffff)` ersetzt (wie bei v0.5.72), funktioniert alles wieder.
Das Problem ist, dass `isConnected` erst dann auf `true` gesetzt wird, wenn dass Powerlimit (erstmalig) vom WR gemeldet wird und die Anfrage dafür soll ja genau hier in `getQueuedCmd` gemacht werden. Da beißt sich die Katze in den Schwanz 😸

lumapu, 2023-01-20:
@beegee3 danke, habe es eingebaut da beim start direkt die Subscription vom MQTT `ctrl` aktiv wird. Daher versucht Ahoy gleich zu beginn das Limit abzufragen und macht nichts anderes mehr.
Ich überlege nochmal wie man die Katze nicht verstümmelt 🤣

beegee3, 2023-01-21:
@lumapu 👍 funktioniert wieder, die Katze hat's überlebt 🐈

closed 2023-02-27 by Ollipop030