lumapu/ahoy #608: Anzeige last Alarm:

Juergen2453, 2023-01-20:
Auf der Webseite von AhoyDTU gibt es dir Info **"last Alarm: Inverter start"**
Allerdings ändert sich die Anzeige nicht auch wenn es andere Alarme gibt.
Besser wäre es wenn es einen Punkt geben würde z.B. letzte Alarme oder Fehlerbericht...
Dann öffnet sich ein Fenster und die Meldungen werden angezeigt.
Der jetzige Anzeige "last Alarm: Inverter start" macht so kein sinn.

jimknopf63, 2023-01-20:
> Auf der Webseite von AhoyDTU gibt es dir Info **"last Alarm: Inverter start"**
> Allerdings ändert sich die Anzeige nicht auch wenn es andere Alarme gibt.
> Besser wäre es wenn es einen Punkt geben [...]

Also im Prinzip wie das bei OpenDTU gehandelt wird. Da werden 15 Alarme im Verlauf gezeigt

Juergen2453, 2023-01-20:
Genau so in der Art, oder überhaupt keine Anzeige.

stefan123t, 2023-01-20:
Siehe auch #600 und #177

lumapu, 2023-01-21:
@stefan123t kannst du erklären warum es auch ein "AlarmUpdate" Command gibt und wann ich es nutzen sollte?

stefan123t, 2023-01-22:
@lumapu ich weiß es nicht, aber soweit ich es vermute kann man mit dem AlarmUpdate Command die alten bereits abgerufenen AlarmData Zeilen aus dem WR RAM löschen. Wir haben das Kommando m.W. bisher noch nicht bei einer Original DTU mit getraced daher ist die Anwendung hier nicht zweifelsfrei klar.

lumapu, 2023-01-23:
@stefan123t danke für deine Antwort. Ins Wiki hatte ich schon geschaut, aber nichts brauchbares finden können

stefan123t, 2023-01-23:
@lumapu eventuell hilft hier nochmal der Blick in die Hoymiles DTU Pro sources oder ein neues Trace mit einem vollen Event/AlarmLog. Dann sollte die Hoymiles DTU ja das AlarmData Log aufräumen und dazu braucht sie mE auch das AlarmUpdate.
Laut Thomas kann man mit schneller Abfrage des SystemConfig Commands nach einem PowerLimit Fehler im WR Log erzeugen.

knickohr, 2023-08-15:
Endlich !  😎

[image]

lumapu, 2023-08-15:
Inverter start sollte schon drin stehen, allerdings fehlt noch das Zeitzonen Offset, das kommt heute noch dazu 😇

knickohr, 2023-08-15:
Ein kleiner Schritt für einen Programmierer, aber ein großer Schritt für die Benutzer 😅

Damit wird Ahoy nochmals um einen großen Schritt aufgewertet 😎

closed 2023-08-17 by lumapu