lumapu/ahoy #613: ESP-32  serial output:  W: while retrieving data: last frame missing: Request Retransmit  --> es werden keine Daten angezeigt

franzgyoeroeg, 2023-01-21:
### Platform

ESP32

### Model name

ESP 32 D0WDQ5

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
 
| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

 

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

release 0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: [114181815193]
    - Name: HM-800
    - Active Power Limit: 450
    - Active Power Limit Control Type: no powerlimit
[... 276 lines cut ...]
I: (#0) sendTimePacket
I: sendTimePacket 1
I: TX 27B Ch23 | 15 81 81 51 93 85 38 77 44 80 01 00 63 CB F5 62 00 00 00 00 00 00 00 00 D2 52 67
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: sendTimePacket 1
I: TX 27B Ch40 | 15 81 81 51 93 85 38 77 44 80 01 00 63 CB F5 62 00 00 00 00 00 00 00 00 D2 52 67
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: sendTimePacket 1
I: TX 27B Ch61 | 15 81 81 51 93 85 38 77 44 80 01 00 63 CB F5 62 00 00 00 00 00 00 00 00 D2 52 67
I: resetPayload: id: 1
I: (#1) sendTimePacket
I: sendTimePacket b
I: TX 27B Ch75 | 15 81 81 53 89 85 38 77 44 80 0B 00 63 CB F5 80 00 00 00 00 00 00 00 00 7C 88 E3
W: while retrieving data: last frame missing: Request Retransmit
I: (#1) sendTimePacket
I: sendTimePacket b
I: TX 27B Ch3 | 15 81 81 53 89 85 38 77 44 80 0B 00 63 CB F5 80 00 00 00 00 00 00 00 00 7C 88 E3
W: while retrieving data: last frame missing: Request Retransmit
```

### Error description

1) mit Version 0.5.72 kein geändertes Verhalten

lumapu, 2023-01-22:
die üblichen Fragen:

- Kondensator verbaut (hier nicht so relevant, aber evtl. zu späterem Zeitpunkt)
- Pinout schon mal verdreht getestet (IRQ und CE vertauschen). Gilt für Hardware als auch in den Einstellungen
- Abstand zum Wechselrichter, NRF24 mit oder ohne Antenne

franzgyoeroeg, 2023-01-22:
zu 1) kein Kondensator verbaut
zu 2) CE und IRQ getauscht --> keine Änderung
zu 3) NRF24 mit Antenne, Abstand zum Wechselrichter etwa 5 m, die Standard DTU light von Hoymiles funktionert im selben Abstand tadellos

lumapu, 2023-01-22:
ok, das sind schon gute Voraussetzungen.
Hat du zufällig noch einen zweiten ESP um auch diesen ausschließen zu können. Es gab auch schon Defekte nrf24 Module.

Sind die Verbindungen gelötet? Wir hatten leider auch schon mit Steckverbindungen Probleme.

franzgyoeroeg, 2023-01-23:
Verbindungen sind gesteckt, nicht gelötet
Habe dann nacheinander 
1) 2. esp32 neu  --> keine Änderung
2) 2. nrf24l01+ neu --> keine Änderung
3) alle Steckverbindungen neu (neue Drähte)  -> keine Änderung

Folgende Software habe ich eingesetzt (geflasht mit flash_download_tool_3.9.3)
Directory ahoy_0.5.66
Bootloader.bin		                        0x1000
partition.bin 		                                0x8000
221230_ahoy_0.5.66_esp32_f8fe044.bin	0x10000
von  download flash_download_tool_3.9.3
boot_app0.bin		                               0xe000

wird ota.bin benötig ?

welche Bedeutung haben die wemos pinouts ?
- CS: D8 (GPIO15)
- CE: D4 (GPIO2)
- IRQ: D3 (GPIO0)

recht herzlichen Dank für die Unterstützung

lumapu, 2023-01-24:
am einfachsten über das Online-Flash-Tool von Ahoydtu.de flashen. Dann muss man sich keine Gedanken über die richtigen Dateien machen.
Das gepostete Pinout gehört zum ESP8266 und nicht zu dem hier verwendeten ESP32. Evtl. liegt hier das Problem.

stefan123t, 2023-01-24:
@franzgyoeroeg bitte sowohl Elko an das NRF24 Modul anlöten als auch die Steckerleisten. Wir hatten auch schon Nutzer die die Steckerleisten einfach so zusammengesteckt hatten ohne ordentliche Kontakt/Lötstellen.
Am Besten mal ein Foto vom Aufbau posten.

franzgyoeroeg, 2023-01-25:
zuersteinmal die Fotos
[image]
[image]
[image]
Muß noch einen Elko organisieren und löten

stefan123t, 2023-01-25:
Manche haben auch berichtet dass die Dupont Jumper Wire's zu dünn / lang sind um genügend Spannung am NRF ab zu liefern, daher ja der Elko. Also Elko schadet nicht und die Lötstellen sehen ganz gut aus. 
Bis auf rot (3.3V) und schwarz (GND) habe ich bei mir andere Farben, daher kann / will ich das jetzt nicht kontrollieren :stuck_out_tongue_winking_eye:

franzgyoeroeg, 2023-01-30:
Mit Elko, 100uF gelötet,  plaudern nun die Komponenten (5 Inverter HM800). Zeitweise kommt noch immer zu Fehlermeldungen. frame1, frame2 oder last frame missing, auf unterschiedlichen Kanälen. Daten sind aber vorhanden und fürs erste plausibel! Steckverbindungen noch nicht gelötet! Test über längere Zeit nun aktiv mit Version 0.5.66.  Nächster Schritt Drähte verstärken und danach löten.

beegee3, 2023-01-30:
missing frames sind nicht wirklich Fehlermeldungen. Die WR übermitteln ihre Daten auf immer wechselnden Kanälen. Ein Schema für diese Wechsel ist nicht erkennbar, daher wird zyklisch auf allen in Frage kommenden Kanälen kurze Zeit gelauscht. Da kann es passieren, das man eine Nachricht nicht (vollständig) mitbekommt. Solche 'missing frames' werden aber sofort nochmal abgefragt. Man verpasst also nichts!

closed 2023-03-08 by lumapu