lumapu/ahoy #619: Feature Request :  Displayorientation um 180 Grad drehbar machen

knickohr, 2023-01-22:
Was für ein OLED belanglos ist, da man es in jedem Winkel gleich gut ablesen kann, fällt bei dem LCD von Nokia schon ins Gewicht. das Display läßt sich in der jetzigen Orientation schlechter ablesen als wenn man es um 180 Grad dreht. der Grund ist Hardwaretechnisch bedingt, LCDs haben eine Vorzugsblickrichtung.

Deshalb bitte die Orientation des Nokia um 180 Grad drehen oder es über ein Setting auf der Webseite einstellbar machen.

tqma1, 2023-01-22:
Würde auch bei den OLED´s nicht schlecht sein aufgrund der Position der Anschlüsse und Einbausituation im Gehäuse ....

lumapu, 2023-01-23:
hoffentlich kann das die Lib, dann ist es schnell eingebaut 😊

knickohr, 2023-01-23:
Ich bin mir sicher das die Lib das hat :
https://github.com/olikraus/u8g2/wiki/u8g2reference#setdisplayrotation

Bei der Gelegenheit bitte auch einen „clean Screen“ machen nach dem Bootlogo und Reboots. Hin und wieder Bleiben einzelne Pixel im Soeicher der Displays hängen.

https://github.com/olikraus/u8g2/wiki/u8g2reference#cleardisplay

knickohr, 2023-02-28:
Ich glaube das kann man zu machen, ist ja umgesetzt.

closed 2023-02-28 by knickohr