lumapu/ahoy #623: [ESP8266/ESP32/RaspberryPi] Problem Description / Beschreibung

dknubben, 2023-01-23:
### Platform

ESP32

### Model name

wemos mini D32

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.74

### Github Hash

84334fb

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Hello, thanks a lot to the dev/devs for this great work!

Since version 0.5.75 I'am not able to compile the source code (branch development03) for ESP32.
I have also tried a fresh "git clone" from this branch with no success.
What could be the problem?

platformio run -e esp32-wroom32-release
errors:

Indexing .pio/build/esp32-wroom32-release/libf13/libRF24.a
In file included from hm/hmSystem.h:10,
                 from appInterface.h:10,
                 from app.h:16,
                 from main.cpp:7:
hm/hmRadio.h: In member function 'void HmRadio<BUFFER, IRQ_PIN, CE_PIN, CS_PIN, AMP_PWR>::dumpBuf(const char*, uint8_t*, uint8_t)':
hm/hmRadio.h:288:17: error: there are no arguments to 'DHEX' that depend on a template parameter, so a declaration of 'DHEX' must
be available [-fpermissive]
                 DHEX(buf[i]);
                 ^~~~
hm/hmRadio.h:288:17: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
In file included from hm/hmSystem.h:10,
                 from appInterface.h:10,
                 from app.h:16,
                 from app.cpp:6:
hm/hmRadio.h: In member function 'void HmRadio<BUFFER, IRQ_PIN, CE_PIN, CS_PIN, AMP_PWR>::dumpBuf(const char*, uint8_t*, uint8_t)':
hm/hmRadio.h:288:17: error: there are no arguments to 'DHEX' that depend on a template parameter, so a declaration of 'DHEX' must
be available [-fpermissive]
                 DHEX(buf[i]);
                 ^~~~
hm/hmRadio.h:288:17: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
Archiving .pio/build/esp32-wroom32-release/libb49/libTime.a
Indexing .pio/build/esp32-wroom32-release/libb49/libTime.a
Compiling .pio/build/esp32-wroom32-release/lib919/WiFiClientSecure/WiFiClientSecure.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib919/WiFiClientSecure/esp_crt_bundle.c.o
Compiling .pio/build/esp32-wroom32-release/lib919/WiFiClientSecure/ssl_client.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/MqttClient.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Packets/Packet.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Packets/Parser.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Packets/RemainingLength.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Packets/String.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Transport/ClientAsync.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Transport/ClientPosix.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Transport/ClientSecureSync.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Transport/ClientSync.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/Transport/IPAddress.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/TypeDefs.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/espMqttClient.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib862/espMqttClient/espMqttClientAsync.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib219/Timezone/Timezone.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib767/Wire/Wire.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib8a7/U8g2/U8g2lib.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib8a7/U8g2/U8x8lib.cpp.o
Compiling .pio/build/esp32-wroom32-release/lib8a7/U8g2/clib/mui.c.o
Compiling .pio/build/esp32-wroom32-release/lib8a7/U8g2/clib/mui_u8g2.c.o
*** [.pio/build/esp32-wroom32-release/src/utils/dbg.cpp.o] Error 1
In file included from app.h:28,
                 from app.cpp:6:
web/web.h: In instantiation of 'void Web<HMSYSTEM>::setup(IApp*, HMSYSTEM*, settings_t*) [with HMSYSTEM = HmSystem<10>]':
app.cpp:62:35:   required from here
web/web.h:97:28: error: 'registerDebugCb' was not declared in this scope
             registerDebugCb(std::bind(&Web::serialCb, this, std::placeholders::_1)); // dbg.h
             ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
web/web.h:97:28: note: suggested alternative: 'register_t'
             registerDebugCb(std::bind(&Web::serialCb, this, std::placeholders::_1)); // dbg.h
             ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             register_t
In file included from hm/hmSystem.h:10,
                 from appInterface.h:10,
                 from app.h:16,
                 from app.cpp:6:
hm/hmRadio.h: In instantiation of 'void HmRadio<BUFFER, IRQ_PIN, CE_PIN, CS_PIN, AMP_PWR>::dumpBuf(const char*, uint8_t*, uint8_t) [with BUFFER = CircularBuffer<packet_t, 30>; unsigned char IRQ_PIN = 16; unsigned char CE_PIN = 4; unsigned char CS_PIN = 5; unsigned char AMP_PWR = 1; uint8_t = unsigned char]':
app.cpp:100:61:   required from here
hm/hmRadio.h:288:21: error: 'DHEX' was not declared in this scope
                 DHEX(buf[i]);
                 ~~~~^~~~~~~~
hm/hmRadio.h:288:21: note: suggested alternative: 'HEX'
                 DHEX(buf[i]);
                 ~~~~^~~~~~~~
                 HEX
*** [.pio/build/esp32-wroom32-release/src/main.cpp.o] Error 1
*** [.pio/build/esp32-wroom32-release/src/app.cpp.o] Error 1
================================================== [FAILED] Took 10.06 seconds

stefan123t, 2023-01-23:
Dear @dknubben please take a look at the build actions on github. We do automatic builds for a variety of targets and I am pretty sure this works for ESP32 too.
Could it be that something with the ESP-IDF / ESP Arduino platform within your PlatformIO has been changed. I believe I have seen such errors in the past when either the IDF version was wrong and/or the ESP8266 & ESP32 platforms needed to be installed both as the ESP32 Referenced some files from the ESP8266 and/or vice versa.

dknubben, 2023-01-30:
Hello @stefan123t

thanks for your answer. I have tried it on three platforms - same problem.
But I found the reason:
I had to delete the CLI platformIO and all the caches from it and installed it completely new.
Now the ESP32 build is working again (ESP8266 build was not "broken").
I guess it's one wrong included third party package (perhaps version etc.)...

closed 2023-01-31 by dknubben