lumapu/ahoy #624: AHOY-DTU: Bitte in den Settings einen Schalter einbauen, der den Homeassistant ausschaltet

bgoldsch, 2023-01-23:
### Platform

ESP8266

### Model name

ESP8266 + nRF24L01+

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.17

### Github Hash

1

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

1

### Debug Serial Log output

```shell
-
```

### Error description

Bitte in den Settings einen Schalter einbauen, der die Homeassistant-Kommunikation ausschaltet.
Hintergrund: Bei mir läuft ein MQTT-Server im IOBroker. Von dort bekomme ich alle notwendigen Daten. Die Homeassistant-Daten kommen immer automatisch mit, werden aber nicht gebraucht, da der IOBroker alle Daten sauber verarbeitet.

stefan123t, 2023-01-23:
Bitte einmal klären ob die Home Assistant Discovery Daten evtl als retained im MQTT Broker gesetzt wurden.

Sollte das der Fall sein, dann bezieht sich dieser Feature Request auf ein "Löschen" / Überschreiben der bestehenden Discovery Daten mit leeren / Null Werten.

knickohr, 2023-01-24:
Ja, die Daten sind retained drin. Wenn Du einmal auf Discovery Config drückst, sind die Daten für die Ewigkeit im Broker. Ich habe da irgendwann mal meine mosquitto DB gelöscht und lasse jetzt tunlichst die Finger von dem Button.

Ach Ja, available_text kommt seit neuem auch wieder. Ist das so gewollt ?  Haben wir doch der Rationalisierung zum Fraß vorgeworfen 😉

lumapu, 2023-01-24:
Achtung sehr alte Version. Hier wird Version 0.5.17 bemängelt, die noch automatisch bei jedem Bootvorgang die Homeassistantdaten sendet.
Für alte Versionen gibt es keinen Support mehr. Bitte auf 0.5.66 (aktuelles Release) aktualisierten.

closed 2023-01-28 by lumapu