lumapu/ahoy #625: ESP8266/ESP32 - MQTT, Reset YieldDay at Midnight

knickohr, 2023-01-25:
### Platform

ESP8266

### Model name

AZ

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.76

### Github Hash

2610f1a

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

Inverter 0

Communication Enable
Serial Number (12 digits)*Name*Max Module Power (Wp)Module Name
Yield Total Correction (will be subtracted) [kWh]Delete

Inverter 1

Communication Enable
Serial Number (12 digits)*Name*Max Module Power (Wp)Module Name
Yield Total Correction (will be subtracted) [kWh]Delete

### Error description

Reset YieldDay at Midnight Funktioniert nicht, oder wird nicht über MQTT ausgegeben.

knickohr, 2023-01-28:
Ich erweitere diesen Issue um den folgenden Punkt :

Mir ist seit DEV 7x aufgefallen das am Morgen immer komische Werte (eben nicht 0) auf einigen Uhren meines Dashboards standen. Habe deshalb gestern einen Ticker mitlaufen lassen wann diese Werte gesendet werden. Hier das Ergebnis von eben.

[image]

Um Mitternacht (UTC) donnert die DTU nochmal alte Werte über MQTT raus. Vermutlich die letzten noch im Speicher hängen gebliebenen. Betrifft wohl Power DC und Irradiation (vermutlich aber noch mehr). Das sind meine „Nullsettings“.

[image]

Das Nullen des YieldDay funktioniert aber wie bei Eröffnung des Incidents nicht. Ich vermute mal stark, es wird im gleichen Moment eben nochmal durch die alten Werte überdonnert.

beegee3, 2023-01-28:
bei Reset YieldDay at Midnight wird z.Zt. nur der YieldDay jedes Inverters genullt, nicht aber YieldDay der einzelnen Panel. Danach werden alle Werte ausgegeben, also auch die alten Werte, die der Inverter zuletzt übermittelt hatte. Die YieldDay Werte der Panel sollten m.E. auch gelöscht werden und danach nur alle YieldDay gepublished werden.
@lumapu: alles was man dafür braucht ist schon vorhanden, also kein `mSendList.push(...); sendIvData();` sondern topic zusammenbauen und direkt pushen.

lumapu, 2023-01-29:
@beegee3 danke, das ist gleich nocht in `0.5.78` eingeflossen. 
Ich weiß nicht, ob @knickohr es so verstanden hat, dass alles auch bei Mitternacht genullt wird. Evtl. könnte man hier auch einfach wie bei den anderen `zerovalues()` aufrufen.

knickohr, 2023-01-29:
@lumapu @beegee3 

Ahhh, die Yield Day der Inverter, nicht der gesamte YieldDay im Topic total. Alles klar, kann auch so bleiben, außer jemand wünscht sich das. Ich kann mir das über Node-Red selbst zusammen stricken. Was aber nicht OK ist, das er um Mitternacht UTC nochmal alte Werte ausspuckt. Da muß entweder eine Null rein, oder aber nur das ausgegeben werden was genullt wird, aber eben nicht alles.

dtuuser, 2023-01-29:
Wird der YieldDay Wert nur genullt, wenn man einen mqtt Server nutzt?

lumapu, 2023-01-29:
> Wird der YieldDay Wert nur genullt, wenn man einen mqtt Server nutzt?

ja das ist mir auch schon aufgefallen, dass es ungünstig ist.

@knickohr durch @beegee3 Änderungsvorschlag sollte das jetzt Geschichte sein

beegee3, 2023-01-30:
bleibt noch die Frage, ob YieldDay nur für die Inverter gelöscht wird (ch0), oder auch für die einzelnen Panel (ch1, ch2, ...) und für die Summenwerte (YieldDay im Topic total)? Halte letzteres für die logisch sinnvollere Variante. Nicht jeder kann sich das wie @knickohr selbst zusammen basteln.

dtuuser, 2023-01-30:
berechnen sich die Werte nicht aufbauend auf die Panel Werte?

knickohr, 2023-01-30:
@dtuuser 

Die Total-Werte ja, aber nicht die Tageswerte. Die werden immer wieder zu jedem Tag neu genullt.

dtuuser, 2023-01-30:
dachte der WR liefert nur die einzelnen Panel Werte und alles andere berechnet sich

beegee3, 2023-01-30:
Der einzelne WR liefert auch die Summe der Panel Werte. Nur die Summe der WR Werte werden berechnet (über alle WR, für die im Setup 'Communication Enable' angehakt ist.

lumapu, 2023-02-24:
denke das Problem ist gefixt - und spätestens durch das Update von @beegee3 mit der Central Europe Time #697

knickohr, 2023-02-25:
Jepp, zu das Ding 👍

closed 2023-02-25 by knickohr