lumapu/ahoy #631: ESP32 zeigt kein Wlan nach flashen

raspihomegit, 2023-01-29:
## Hardware
  * [ ] ESP8266
  * [X ] ESP32
  * [ ] Raspberry Pi

Modelname: ESP32-DOWD-V3______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [X ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [X ] Windows
* [ ] Mac OS

### Debugging:
* [X ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Guten Tag,

ich versuche die Version 5.66 von AhoyDTU zu flashen. Das Programm zeigt 100% an, aber nach dem Neustart des ESP32 erscheint WLan. Ich komme an dieser Stelle nicht weiter. Die Datei zum Flashen heißt: 221230_ahoy_0.5.66_esp32_f8fe044.bin

Kann mir jemand bitte einen Rat geben?

knickohr, 2023-01-29:
Wie hast Du geflashed und was alles ?

raspihomegit, 2023-01-29:
Ich habe den nodemcu Pyflasher verwendet. Die genannte Datei angegeben und geflasht. 
Der flacher gab auch an, welcher esp am com 3 ist, wurde gelöscht und neu geschrieben. Aber irgendwie passiert nun nichts weiter.

knickohr, 2023-01-29:
Nimm das Flash Download Tool und geh nach den Anleitungen von OpenDTU vor :

https://github.com/tbnobody/OpenDTU#flash-with-espressif-flash-download-tool-windows

Natürlich Didier Dateien von Ahoy verwenden. Es müssen alle 4 geflasht werden, an die richtige Adresse. Und man beachte die Häkchen und Optionen im Tool !

knickohr, 2023-01-29:
Hier der Screenshot für Ahoy :

[image]

raspihomegit, 2023-01-29:
Mit diesem Tool hatte ich es auch probiert. Allerdings hatten mir die Adressen gefehlt. 🤦🏻‍♂️😅 nun habe ich alles eingetragen und es funktioniert. Ich danke dir vielmals.

lumapu, 2023-01-29:
oder noch einfacher mit dem online Flashtool auf unserer Website:
https://ahoydtu.de/web_install/

DanielR92, 2023-02-03:
Kann geclosed werden oder?

raspihomegit, 2023-02-05:
ja, kann geschlossen werden. Danke

closed 2023-02-05 by raspihomegit