lumapu/ahoy #632: [ESP8266] MQTT discovery sets Manufactor file to value null

lubeda, 2023-01-29:
### Platform

ESP8266

### Model name

HM-300

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

release

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
does not matter
```

### Error description

A look in the source states that there should be "Hoymiles"

{
  "name": "rechts YieldTotal",
  "stat_t": "solar/rechts/ch0/YieldTotal",
  "unit_of_meas": "kWh",
  "uniq_id": "112171909775_ch0_YieldTotal",
  "dev": {
    "name": "rechts",
    "ids": "112171909775",
    "cu": "http://192.168.178.80",
    **"mf": null**
  },
  "exp_aft": 65,
  "dev_cla": "energy",
  "stat_cla": "total_increasing"
}

sker65, 2023-01-29:
Same here, firmware version 0.5.78, on wmos D1 mini pro

lubeda, 2023-01-30:
Well I think the allocated size of the jsondocument is too small for the data. So the json isn't big enough and some values are dropped. 

After shortening some data, like the inverter-name I get Homilies.

sker65, 2023-01-30:
Well an example discover msg looks like:
`{
  "name": "balkonoben CH1 U_DC",
  "stat_t": "inverter/balkonoben/ch1/U_DC",
  "unit_of_meas": "V",
  "uniq_id": "114183080207_ch1_U_DC",
  "dev": {
    "name": "balkonoben",
    "ids": "012345678912",
    "cu": "http://192.168.178.241",
    "mf": "Hoymiles"
  },
  "exp_aft": 65,
  "dev_cla": "voltage",
  "stat_cla": "measurement"
}
`
So the dev object has 105 byte, the whole msg is 336 byte in my example. Really too big? What about using the String class from the arduino runtime / stdlib and allocate strings more dynamically?

lubeda, 2023-01-30:
In Zeile  149 von pubMqtt.h ist die Größe auf 128 begrenzt
`149: DynamicJsonDocument deviceDoc(128);`

Ich kann das aber nicht compilieren und testen. In dem Beispiel von [sker65](https://github.com/sker65) fehlt z.B. auch der Wert mdl, der im Quelltext in Zeile 154 hinzugefügt wird.

stefan123t, 2023-01-31:
See PR #630 for the current work in progress in the development03 branch.

lumapu, 2023-02-07:
I think it was already solved with the most recent development versions, please verify that.

sker65, 2023-02-07:
I've tried the 0.5.81 and it does resolve the issue with mf set to null

closed 2023-03-27 by lumapu