lumapu/ahoy #633: fehlende Anzeige der Leistung einzelner Inverter im Display

soelli22, 2023-01-30:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ESP32-D0WDQ5

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.78

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [x] Mac OS

Hallo zusammen,

seit einigen DEV-Versionen ist für die Displays (SH1106/SSD1306) die Anzeige der Leistung der einzelnen Inverter verschwunden.
Mache ich einen Fehler bei den Einstellungen, oder ist dies aktuell so. Wenn zweiteres, würde ich mich freuen, wenn diese Möglichkeit wieder verfügbar wäre.

Vielen Dank!

lumapu, 2023-01-30:
ja das wurde rausoptimiert 😉. Ich würde hier ein bisschen nachhören, was die Gemeinschaft so will, kann mir aber gut vorstellen dass Feature wieder anzubieten.

tqma1, 2023-02-03:
Wäre ich auch dafür ….

sh-florian, 2023-02-06:
Mir hat das auch vorher besser gefallen.

knickohr, 2023-02-07:
Vielleicht sollte man sich auch in die Diskussion in der Discord-Gruppe einbringen. Nachher rummäkeln finde ich nicht so pralle.

https://discord.com/channels/984173303147155506/1065266691463974972

knickohr, 2023-02-07:
War nicht persönlich gemeint. Aber bei 5 verschiedenen Leuten geht ich in der Annahme das mindestens einer auch im Discord liest 😉

tqma1, 2023-02-07:
Discord ist mir fast zu steil. Da wird ja über alles mögliche fast gleichzeitig diskutiert.
E-Paper Display, diese Thema hier, …
Ich lese gerne mal was nach bzw. suche nach näheren Informationen usw.
Aber den einzelnen Themen zu folgen - da fehlt mir offensichtlich der Durchblick

abloop046, 2023-02-07:
Frag mich auch , was ich dann hier soll, wenn die wesentlichen Sachen woanders passieren.

stefan123t, 2023-02-08:
@knickohr also das Feature die einzelnen Channels der WR auf dem Display zu sehen scheint doch einige Freunde zu haben.
Die Gesamtleistung aller WR alleine bzw. ein wechseln der einzelnen WR genügt wohl auch nicht, da manche z.B. bei Ost/West Ausrichtung beide/alle Kanäle mal einzeln sehen wollen.
Wenn es für einzelne Einsatzzwecke zu viel Detailinformationen sind, kann man die Details ja über ein Setting optional machen:
* Nur Total aller WR der DTU
* Total der DTU plus Total aller WR im Wechsel
* Total der DTU plus Total aller WR plus Werte der Kanäle des jeweiligen WR im Wechsel

Bei vielen WR mit zwei/vier Kanälen muß man halt warten bis die Anzeigen durchgelaufen sind oder das Setting anpassen.

knickohr, 2023-02-08:
Ich habe nicht gesagt das ich dagegen bin 😉 Ich meine halt, man kann es eben nicht allen Recht machen. Wo fangen wir an, wo hören wir auf ?

Damals, als das Thema Display zum ersten Mal aufkam schreite auch jeder das brauchen wir nicht. Und wo sind wir jetzt ?  😅

Das Problem ist halt der recht beschränkte Platz auf dem Display. Was bei einem ePaper noch einigermaßen machbar ist, fällt bei einem OLED schon mal flach. Hier hilft also wirklich nur auswählen und sich die Rosinen, ähhh Werte rauszupicken.

Bei meinen 6 bis 8 Invertern wird das spannend. Da hilft wirklich nur noch Auswahl über Settings @lumapu wird uns zerfleischen 😵

lumapu, 2023-02-08:
> Frag mich auch , was ich dann hier soll, wenn die wesentlichen Sachen woanders passieren.

hier ist der Hauptteil des Projekts, Discord ist eher der Stammtisch 😃

In Discord geht's ab und zu richtig ab, ich sage dort immer wieder, das alle Ungereimtheiten hier als issue erfasst werden müssen, das kann kein Normalsterblicher in seiner Freizeit alles lesen 😂

lumapu, 2023-07-23:
ist meines Wissens wieder in der Dev-Verison drin

abloop046, 2023-07-24:
Danke für die Info!

stefan123t, 2023-08-18:
@lumapu dann kann hier ein [fixeddev] label dran ?

closed 2023-08-18 by lumapu