lumapu/ahoy #636: PCB Pinout Einstellungen.

Banabas, 2023-01-31:
Tag zusammen.
Habe eine ahoy Platine hier aus dem git bekommen.
Wemos D1 ESP8266 v3 und das Funkmodul auf die Platine gelötet aber bekomme mit dem mit der Standart config keine Verbindung zum hm-1500.
Muss das Pinout in den Einstellungen geändert werden?

Grüße
Banabas

Ollipop030, 2023-01-31:
Was steht denn bei dir im Webinterface unter System/Radio?

stefan123t, 2023-01-31:
@Banabas welcher Link zur Platine und schick mal ein Foto hier oder im Discord.

Banabas, 2023-02-01:
Tach zusammen.
[https://github.com/lumapu/ahoy/tree/main/tools/pcb](url)
Das ist die Platine.
Wenn ich die Bauteile per Kabel verbinde geht´s.
Aber die Platine funktioniert bei mir irgendwie nicht.
Auf Durchgang geprüft... ok
Verschiedene NRF mit und ohne Kondensator versucht, andere Wemos aber keine Verbindung zum Wechselrichter.
[image]
[image]

stefan123t, 2023-02-02:
@Banabas frag mal @pascaltippelt den Autor der Platine persönlich. Es gab da auch Schwierigkeiten mit den Pins D3/D4 bei einigen Wemos D1 Boards. Das liess sich idR durch tauschen von D3&D4 in der Verkabelung **und** in den Settings beheben.

pascaltippelt, 2023-02-02:
Moin @Banabas !

Also wenn ich deinen Log so ansehe, hast du kein Problem mit der physikalischen / elektrischen Verbindung zwischen Platine und Funkmodul. Da steht nämlich groß "nrf24l01+ is connected". Allerdings scheinst du auch die Sendeleistung am Minimal gestellt zu haben (Power Level MIN). Eventuell als erstes mal auf MAX stellen.

Im Log steht ja auch "last frame missing", sprich da kommt irgendwas zurück, aber er kann damit nichts anfangen. Klingt für mich nach schlechtem Empfang.

pascaltippelt, 2023-02-02:
[image]

So sieht meine Konfiguration auf dem Board aus.

stefan123t, 2023-02-02:
Ja wenn der Elko dran ist kann man auch mal vorsichtig die Sendeleistung erhöhen. Je nach NRF mit/ohne externer Antenne klappt das früher oder später.
@pascaltippelt das Last frame missing ist in dem Zusammenhang etwas irreführend, die AhoyDTU will komischerweise zuerst das letzte Paket empfangen und fragt dann die fehlenden Pakete 1..N nach. Die Original Hoymiles DTU und OpenDTU fragen in so einem Fall die ganze Nachricht aus N Paketen erneut an. Der WR hat ja vermutlich das Kommando gar nicht bekommen.
Sollte also der WR nicht wissen dass wir den Status mit N Paketen von ihm wollen sondern als letztes nur ein Ack auf Start Inverter im Kommandospeicher haben dann gibt es da auch nur ein Paket, folglich kann er gar nicht mit dem Paket N antworten und bleibt stumm.
@lumapu das wollten wir doch mal korrigieren, oder ?

beegee3, 2023-02-02:
@stefan123t 
> die AhoyDTU will komischerweise zuerst das letzte Paket empfangen

das stimmt so nicht, es werden alle Pakete angefragt und in der richtigen Reihenfolge 1..N empfangen.  Erst wenn aber kein einziges Paket empfangen wird, kommt das irreführende 'Last frame missing' und das letzte Paket wird angefordert. Das ist in der development03 geändert und es werden alle Pakete neu angefragt (wie bei Original Hoymiles DTU und OpenDTU).

Das Protokoll oben lässt vermuten, dass der WR gar nicht antwortet. Aufgrund der Uhrzeit ist anzunehmen, dass er noch nicht Offline war (es sei denn, die Kollektoren sind extrem verschattet). Die üblichen Probleme sind schon oben angesprochen, daher mal die banale Frage (bitte nicht schlagen): schon geprüft, ob die Seriennummer richtig eingetragen ist?

Ollipop030, 2023-02-03:
> Auf Durchgang geprüft... ok

Hast du evtl. auch mal geprüft, ob irgendwo ein Beinchen gebrückt ist, also irgenwohin Durchgang hat, wo er nicht hin sollte?

Banabas, 2023-02-04:
> Hast du evtl. auch mal geprüft, ob irgendwo ein Beinchen gebrückt ist, also irgenwohin Durchgang hat, wo er nicht hin sollte?

Nee das hab ich nicht gemacht. aber die Lötpunte sehen gut aus. Ich bekomme heute noch andere Wemos und werde es damit nochmal testen. Kann ja irgenwie nicht sein... ma schaun. Ich melde mich.

Banabas, 2023-02-04:
So der Test mit nem anderen Wemos war positiv.
Also muss irgendwas mit den Wemos nicht stimmen. 
ESP8266EX

Loetnase, 2023-02-05:
Sobald sich so ein ESP8266 (wemos) komlett komisch verhält, versuche ich den mit der Arduino IDE mit einem kleinen Beispielprogramm zu flashen. Z.B- Blink oder einen WLAN Scanner (ScanNetworks). Dabei in Werkzeuge, Erase Flash, All Flash Contents auswählen.
Ich denke das löscht alles überflüssige im Speicher.
Dann den ESP mit der Ahoy DTU wie gewohnt komplett neu aufsetzen und hoffenm das es dann geht.
Ich hatte schon mehrmals das Gefühl der ESP ist kaputt uns so gieng er dann wieder.
Ich habe so das Gefühl das der ESP Web Installer nicht immer den ganzen Speicher platt macht.

stefan123t, 2023-02-08:
@Banabas hast Du bereits das Gefühl dass es jetzt soweit rund läuft und wir das issue schließen können ?

closed 2023-03-27 by lumapu