lumapu/ahoy #637: MQTT reports zero values for inverters not yet online

Argafal, 2023-02-01:
With **0.5.78** an old bug is back. I have three inverters. The inverters come online at different times. 

Expected behaviour: whenever an inverter comes online it should report on MQTT. The other inverters should not report anything.
Actual behaviour: As soon as the earliest inverter reports, all the other (not yet online) inverters also report on MQTT. The latter report zero values. 

Example 1: inverter _Central_ makes first contact and reports its firmware version. The two inverters _North_ and _Pilot_ have not send anything to the dtu yet, but (falsely) report zero values at the same moment.

> inverter/Pilot/ch0/FWVersion 0
> inverter/Pilot/ch0/FWBuildYear 0
> inverter/Pilot/ch0/FWBuildMonthDay 0
> inverter/Pilot/ch0/FWBuildHourMinute 0
> inverter/Pilot/ch0/HWPartId 0
> inverter/Central/ch0/FWVersion 10016
> inverter/Central/ch0/FWBuildYear 2021
> inverter/Central/ch0/FWBuildMonthDay 1012
> inverter/Central/ch0/FWBuildHourMinute 1706
> inverter/Central/ch0/HWPartId 100
> inverter/North/ch0/FWVersion 0
> inverter/North/ch0/FWBuildYear 0
> inverter/North/ch0/FWBuildMonthDay 0
> inverter/North/ch0/FWBuildHourMinute 0
> inverter/North/ch0/HWPartId 0

Example 2: once _Central_ is fully online and reports production, _North_ and _Pilot_ are still not online. They report zero values that shouldn't be there:

> inverter/Pilot/ch1/U_DC 0
> inverter/Pilot/ch1/I_DC 0
> inverter/Pilot/ch1/P_DC 0
> inverter/Pilot/ch1/YieldDay 0
> inverter/Pilot/ch1/YieldTotal 0
> inverter/Pilot/ch1/Irradiation 0
> inverter/Pilot/ch2/U_DC 0
> inverter/Pilot/ch2/I_DC 0
> inverter/Pilot/ch2/P_DC 0
> inverter/Pilot/ch2/YieldDay 0
> inverter/Pilot/ch2/YieldTotal 0
> inverter/Pilot/ch2/Irradiation 0
> inverter/Pilot/ch0/U_AC 0
> inverter/Pilot/ch0/I_AC 0
> inverter/Pilot/ch0/P_AC 0
> inverter/Pilot/ch0/Q_AC 0
> inverter/Pilot/ch0/F_AC 0
> inverter/Pilot/ch0/PF_AC 0
> inverter/Pilot/ch0/Temp 0
> inverter/Pilot/ch0/ALARM_MES_ID 0
> inverter/Pilot/ch0/YieldDay 0
> inverter/Pilot/ch0/YieldTotal 0
> inverter/Pilot/ch0/P_DC 0
> inverter/Pilot/ch0/Efficiency 0
> inverter/Central/ch1/U_DC 34.4
> inverter/Central/ch1/I_DC 0.64
> inverter/Central/ch1/P_DC 22
> inverter/Central/ch1/YieldDay 38
> inverter/Central/ch1/YieldTotal 286.329
> inverter/Central/ch1/Irradiation 0
> inverter/Central/ch2/U_DC 34.4
> inverter/Central/ch2/I_DC 0.64
> inverter/Central/ch2/P_DC 21.8
> inverter/Central/ch2/YieldDay 37
> inverter/Central/ch2/YieldTotal 258.346
> inverter/Central/ch2/Irradiation 0
> inverter/Central/ch3/U_DC 34.9
> inverter/Central/ch3/I_DC 0.67
> inverter/Central/ch3/P_DC 23.5
> inverter/Central/ch3/YieldDay 40
> inverter/Central/ch3/YieldTotal 289.98
> inverter/Central/ch3/Irradiation 0
> inverter/Central/ch4/U_DC 34.9
> inverter/Central/ch4/I_DC 0.81
> inverter/Central/ch4/P_DC 28.1
> inverter/Central/ch4/YieldDay 46
> inverter/Central/ch4/YieldTotal 291.493
> inverter/Central/ch4/Irradiation 0
> inverter/Central/ch0/U_AC 235.8
> inverter/Central/ch0/I_AC 0.38
> inverter/Central/ch0/P_AC 90.7
> inverter/Central/ch0/Q_AC 21.7
> inverter/Central/ch0/F_AC 49.99
> inverter/Central/ch0/PF_AC 0.973
> inverter/Central/ch0/Temp 12.2
> inverter/Central/ch0/ALARM_MES_ID 1
> inverter/Central/ch0/YieldDay 161
> inverter/Central/ch0/YieldTotal 1126.15
> inverter/Central/ch0/P_DC 95.4
> inverter/Central/ch0/Efficiency 95.073
> inverter/North/ch1/U_DC 0
> inverter/North/ch1/I_DC 0
> inverter/North/ch1/P_DC 0
> inverter/North/ch1/YieldDay 0
> inverter/North/ch1/YieldTotal 0
> inverter/North/ch1/Irradiation 0
> inverter/North/ch2/U_DC 0
> inverter/North/ch2/I_DC 0
> inverter/North/ch2/P_DC 0
> inverter/North/ch2/YieldDay 0
> inverter/North/ch2/YieldTotal 0
> inverter/North/ch2/Irradiation 0
> inverter/North/ch3/U_DC 0
> inverter/North/ch3/I_DC 0
> inverter/North/ch3/P_DC 0
> inverter/North/ch3/YieldDay 0
> inverter/North/ch3/YieldTotal 0
> inverter/North/ch3/Irradiation 0
> inverter/North/ch4/U_DC 0
> inverter/North/ch4/I_DC 0
> inverter/North/ch4/P_DC 0
> inverter/North/ch4/YieldDay 0
> inverter/North/ch4/YieldTotal 0
> inverter/North/ch4/Irradiation 0
> inverter/North/ch0/U_AC 0
> inverter/North/ch0/I_AC 0
> inverter/North/ch0/P_AC 0
> inverter/North/ch0/Q_AC 0
> inverter/North/ch0/F_AC 0
> inverter/North/ch0/PF_AC 0
> inverter/North/ch0/Temp 0
> inverter/North/ch0/ALARM_MES_ID 0
> inverter/North/ch0/YieldDay 0
> inverter/North/ch0/YieldTotal 0
> inverter/North/ch0/P_DC 0
> inverter/North/ch0/Efficiency 0

lumapu, 2023-02-01:
understand why this happens, I must think about that and can't imagine that this behavior was different in past.
As background information: once one inverter receives data a list will be appended and a loop will publish all data.

Thank you for reporting, I'm sure we can solve that 😊

Argafal, 2023-02-01:
I think it used to work just fine, see related issues #49 #55 #397. A lot of development has happened since then, of course, so thanks to anyone who can help sort this out one more time :)

stefan123t, 2023-02-02:
@beegee3 hier wieder mal ein issue mit run time conditions je nach Zustand der Inverter nach dem Booten der DTU. Du bist doch so geschickt beim Finden der Fehler ;)

lumapu, 2023-02-02:
@stefan123t ich denke man muss einfach das available beim loopen über alle inverter auswerten und nur wenn es true ist übertragen.

beegee3, 2023-02-02:
die Probleme der beiden Beispiele kommen daher, dass letzte Status jeden Inverters gespeichert wird. Bei der Initialisierung wird ein ungültiger Status gesetzt. Ändert sich später ein Status, wird ein Publish gestartet. Jeder Inverter bekommt bei der ersten Prüfung (also nach Systemstart) mindestens den Status "nicht verfügbar", was sofort ein Publish auslöst.

Mein Vorschlag: Bei der Initialisierung den Status "nicht verfügbar" vorgeben und wie @lumapu schon sagt, nur bei "verfügbar" ein Publish starten.

@lumapu hier der Vorschlag für die Anpassungen, Hinweise zu `processIvStatus`:
- habe `mIvAvail` durch lokale Variable `anyAvail` ersetzt, d.h. **`mIvAvail` bitte aus der Variablenliste löschen!**
- `allAvail` und `totalComplete` hatten dieselbe Funktion, `allAvail` reicht also
- habe die Statusermittlung sozusagen umgedreht. Schau bitte genau drüber, ob das deiner Logik entspricht!

```
        PubMqtt() {
            mRxCnt = 0;
            mTxCnt = 0;
            mSubscriptionCb = NULL;
            memset(mLastIvState, MQTT_STATUS_NOT_AVAIL_NOT_PROD, MAX_NUM_INVERTERS);
        }

        bool processIvStatus() {
            // returns true if all inverters are available
            bool allAvail = true;   // shows if all enabled inverters are available
            bool anyAvail = false;  // shows if at least one enabled inverter is available
            bool changed = false;
            char topic[7 + MQTT_TOPIC_LEN], val[40];
            Inverter<> *iv;
            record_t<> *rec;

            for (uint8_t id = 0; id < mSys->getNumInverters(); id++) {
                iv = mSys->getInverterByPos(id);
                if (NULL == iv)
                    continue; // skip to next inverter

                rec = iv->getRecordStruct(RealTimeRunData_Debug);

                // inverter status
                uint8_t status = MQTT_STATUS_NOT_AVAIL_NOT_PROD;
                if (iv->config->enabled) {
                    if (iv->isAvailable(*mUtcTimestamp))
                        status = (iv->isProducing(*mUtcTimestamp)) ? MQTT_STATUS_AVAIL_PROD : MQTT_STATUS_AVAIL_NOT_PROD;
                    else // inverter is enabled but not available
                        allAvail = false;
                }

                if(mLastIvState[id] != status) {
                    mLastIvState[id] = status;
                    changed = true;

                    if((MQTT_STATUS_NOT_AVAIL_NOT_PROD == status) && (mCfgMqtt->rstValsNotAvail))
                        zeroValues(iv);

                    snprintf(topic, 32 + MAX_NAME_LENGTH, "%s/available", iv->config->name);
                    snprintf(val, 40, "%d", status);
                    publish(topic, val, true);

                    snprintf(topic, 32 + MAX_NAME_LENGTH, "%s/last_success", iv->config->name);
                    snprintf(val, 40, "%d", iv->getLastTs(rec));
                    publish(topic, val, true);
                }
            }

            if(changed) {
                snprintf(val, 32, "%d", ((allAvail) ? MQTT_STATUS_ONLINE : ((anyAvail) ? MQTT_STATUS_PARTIAL : MQTT_STATUS_OFFLINE)));
                publish("status", val, true);
                sendIvData(false); // false prevents loop of same function
            }

            return allAvail;
        }

        void sendIvData(bool sendTotals = true) {
            ...

                for (uint8_t id = 0; id < mSys->getNumInverters(); id++) {
                    Inverter<> *iv = mSys->getInverterByPos(id);
                    if ((NULL == iv) || (MQTT_STATUS_NOT_AVAIL_NOT_PROD == mLastIvState[id]))
                        continue; // skip to next inverter
                        ...
```

lumapu, 2023-02-07:
was fixed with PR from @beegee3 #652

closed 2023-03-27 by lumapu