lumapu/ahoy #638: MQTT configuration settings unclear - rephrase?

Argafal, 2023-02-01:
Version 0.5.78

There are new MQTT settings but I find them hard to understand. 

> Reset YieldDay at Midnight
> Reset Values at Communication stop
> Reset Values on 'not available'

What's the difference between a "communication stop" and "not available"? 
What does reset mean (zero, not/available, empty string, etc)? 

I propose we re-phrase the text for these options to make it clear what they actually do.

knickohr, 2023-02-01:
Communication Stop bezieht sich auf die Setting „Sunrise/Sunset“. Hier wird also resettet wenn Du entweder zum Sonnenuntergang aktiviert hast oder aber sogar noch einen Offset eingegeben hast. Natürlich nur wenn der Haken darunter gesetzt ist.

„Not Available“ bedeutet wenn der Inverter den betreffenden Status ausgibt, also offline geht.

Reset =. Wert 0

Ob das nur auf MQTT beschränkt ist, kann ich Dir jetzt nicht beantworten. Sollte aber meiner Meinung so sein, das es im Setting MQTT ist.

Argafal, 2023-02-01:
Thanks, @knickohr 

Proposal:
> Reset YieldDay at Midnight
Set YieldDay to zero at midnight

> Reset Values at Communication stop
Set production values to zero at (sunset+offset).

> Reset Values on 'not available'
Set production values to zero when an inverter changes status to 'not available'.

Argafal, 2023-02-01:
Next question: How does "Reset Values at Communication stop" interact with the sunset/sunrise setting "disable night communication"? What does the latter do?

knickohr, 2023-02-01:
Zu diesen Zeitpunkt hört die DTU auf die Wechselrichter zu pollen/abzufragen. es findet also kein Verbindungsaufbau zu den Invertern mehr statt („Elektrosmog“)

Argafal, 2023-02-01:
Thanks, that makes a lot of sense. Proposal: "Stop polling inverters during night. "

Argafal, 2023-02-01:
And one more item that's connect to this general theme. On the front page I find a message "Polling inverter(s), will stop at 1.2.2023, 17:27:38". I propose to also mention here why the polling should stop. Proposal: Polling inverter(s), will pause at 1.2.2023, 17:27:38 due to night-time polling being disable in the settings (see sunrise/sunset). Polling will continue at TIMESTAMP"
Maybe a little long? :-/

Argafal, 2023-02-08:
Patch attached. This updates the strings discussed above. Bonus: I found that the labels and their checkboxes were not well aligned, they had slightly different margins set in css. Fixed in the patch. 

`diff --git a/src/web/html/index.html b/src/web/html/index.html
index cb16ddc..21bfb5f 100644
--- a/src/web/html/index.html
+++ b/src/web/html/index.html
@@ -145,12 +145,12 @@
                 if(obj["ts_sunrise"] > 0) {
                     if(((obj["ts_sunrise"] - obj["ts_offset"]) < obj["ts_now"])
                         && ((obj["ts_sunset"] + obj["ts_offset"]) > obj["ts_now"])) {
-                        commInfo = "Polling inverter(s), will stop at " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleString('de-D
E'));
+                        commInfo = "Polling inverter(s), will stop at sunset " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleStrin
g('de-DE'));
                     }
                     else if(obj["disNightComm"]) {
-                        commInfo = "Night time, no Communication to Inverter, ";
+                        commInfo = "Night time, inverter polling disabled, ";
                         if(obj["ts_now"] > (obj["ts_sunrise"] - obj["ts_offset"])) {
-                            commInfo += "stopped polling at " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
+                            commInfo += "stopped at " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
                         }
                         else {
                             commInfo += "will start polling at " + (new Date((obj["ts_sunrise"] - obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
diff --git a/src/web/html/setup.html b/src/web/html/setup.html
index 58c3153..7676094 100644
--- a/src/web/html/setup.html
+++ b/src/web/html/setup.html
@@ -96,11 +96,12 @@
                         <label for="invRetry">Max retries per Payload</label>
                         <input type="text" class="text" name="invRetry"/>
 
-                        <label for="invRstMid">Reset Values and YieldDay at Midnight</label>
+
+                        <label for="invRstMid">Reset values and YieldDay at midnight</label>
                         <input type="checkbox" class="cb" name="invRstMid"/><br/>
-                        <label for="invRstComStop">Reset Values at Communication stop</label>
+                        <label for="invRstComStop">Reset values when inverter polling stops at sunset</label>
                         <input type="checkbox" class="cb" name="invRstComStop"/><br/>
-                        <label for="invRstNotAvail">Reset Values on 'not available'</label>
+                        <label for="invRstNotAvail">Reset values when inverter status is 'not available'</label>
                         <input type="checkbox" class="cb" name="invRstNotAvail"/><br/>
                     </fieldset>
                     </div>
@@ -134,7 +135,7 @@
                         <label for="sunOffs">Offset (pre sunrise, post sunset)</label>
                         <select name="sunOffs"></select>
                         <br>
-                        <label for="sunDisNightCom">disable night communication</label>
+                        <label for="sunDisNightCom">Stop polling inverters during night</label>
                         <input type="checkbox" class="cb" name="sunDisNightCom"/><br/>
                     </fieldset>
                     </div>
diff --git a/src/web/html/style.css b/src/web/html/style.css
index 8f4b473..51f9a9b 100644
--- a/src/web/html/style.css
+++ b/src/web/html/style.css
@@ -282,7 +282,8 @@ input.btn:hover {
 }
 
 input.cb {
-    margin-bottom: 20px;
+    margin-bottom: 15px;
+    margin-top: 10px;
 }
 
 label {
(END)
`

Argafal, 2023-02-08:
Noch ein Satz der mir aufgefallen ist: "Latitude and Longitude must be set to be stored!" 
Ich hab jetzt ne ganze Zeit lang gegrübelt, und komm nicht drauf: Was will mir das sagen?

stefan123t, 2023-02-08:
Dass sie gesetzt werden müssen um gespeichert zu werden ist irgendwie logisch und unnötig.
Ich glaube umgekehrt wenn die beiden Angaben Lat/Lon nicht gesetzt sind, 
dann klappt die Berechnung von Sunrise / Sunset nicht.

closed 2023-03-27 by lumapu